Spark-Concepts/xPro-V5 #1: Howto : dynamically assign motor drivers to axes.

mathojojo, 2020-12-13:
Hello,

In the Hardware descrition, we can find the following :
"Motor drivers can be dynamically assigned to axes, so a 4 motor XYZA controller could be converted to a XYYZ (dual motor Y axis) without any hardware changes."

It would be nice to explain how we can do that, it's an essential function, and I can't find it myself (sorry, I might be noob).

Thank you very much

Spark-Concepts, 2020-12-13:
Mike,

Can you answer this question?  Can we do this dynamically - or - does this
require a firmware mod?

samwilt, 2020-12-31:
Any update on this? Thanks

Spark-Concepts, 2021-01-03:
At the present time this does require a firmware mod and will be available within the next couple of days (XYZA). We are presently finishing up test and evaluation; once complete we will post the firmware mod here: (https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware)

...updating firmware: https://github.com/Spark-Concepts/xPro-V5/wiki/Updating-Firmware

In the near future we will be posting a separate firmware update in which you will be able to dynamically configure axes through G-code instead of recompiling.

We have been working on this for a few months and working to make (nearly) everything reconfigurable at run-time without recompiling. Stay tuned.

Cheers!
~Bill

kmjonsson, 2021-03-01:
Will there be XYZA-firmware available?
The only reason I got this was for the real XYZA-feature.
If it's not available I can as well use it as a book stand.

synackbar, 2021-03-26:
I'll second kmjonsson comment. Any update on this?

closed 2021-04-07 by Mboss32

mathojojo, 2021-04-07:
Hello, is it closed because you posted some moded firmwares for each case of use ? Or is it closed because you added the ability to dynamically configure axes through G-code instead of recompiling ??