Spark-Concepts/xPro-V5 #4: E-Stop issue

samwilt, 2021-01-17:
When using the E-Stop port and having changed the pins inside the unit, I cannot get any axis to move even when the E-Stop is not activated. When pressed, a red light comes up on the unit indicating that the E-Stop is activated and when releasing the E-Stop the unit then goes green.

Other issues such as not being able to activate spindles etc

Any comments?

Gezzunder, 2021-01-22:
Hi Sam,
I had the same issue.  Mike from Spark Concepts told me;
"there are some older pictures on the E-stop operation that had a mistake.  The Estop jumper needs to be in the Estop Disabled position, and the physical button needs to be wire N/O"

So try reverting to default jumper settings and, if you have a normally closed switch then push it in and it should work.  When you press the button (you'll need to change it to a normally open switch) it comes up as a door alarm, but everything stops

TRUDIM, 2021-03-30:
Hey,

I got a NC E-stop button and for the life of me cannot get it to work with the xPro V5. Anyone know a way of connecting it from the side of the button to the side of the xPro. Where should the jumper be? 
I send an e-mail to Spark design, asking them these questions but they seem to be unavailable (or have no idea how to do it either...)

Anyone?

Spark-Concepts, 2021-03-30:
All - I'm getting ready to push a big update to the wiki - Please see the attached (you'll also need to update the binary also in the .zip)

[NC_e-stop-hooup&bin.zip](https://github.com/Spark-Concepts/xPro-V5/files/6232421/NC_e-stop-hooup.bin.zip)

TRUDIM, 2021-03-31:
Nice! Will give it a try later on today!

Spark-Concepts, 2021-03-31:
Absolutely not, please see the attached for NO E-stops:

[NO_e-stop-hookup&bin.zip](https://github.com/Spark-Concepts/xPro-V5/files/6236517/NO_e-stop-hookup.bin.zip)

philshel1, 2021-03-31:
Is there anything new in the firmware posted here that the rest of us should upgrade for?

Spark-Concepts, 2021-03-31:
Gezzunder,

The precompiled firmware is named to clearly identify specific compile time options - you can find the most ***up-to-date*** firmware here: https://github.com/Spark-Concepts/xPro-V5

CNC_xPRO_V5_  | MotoAssignment_ | SpindleType_ | DoorSwitchType.bin
------------- | -------------|------------- | -------------
" | XYYZ | PWM | NO
" | XYAZ | PWM | NO
" | XYYZ | PWM | NC
" | XYAZ | PWM | NC
" | XYYZ | 485 | NO
" | XYAZ | 485 | NO
" | XYYZ | 485 | NC
" | XYAZ | 485 | NC

**MotorAssignment:** *the options are XYYZ (3 axis, ganged Y) and XYAZ (4 axis)*

**SpindleType:** *the options are PWM which allows for 0-5V or 0-10V spindle (or laser) and 485 allows for the use of an RS485 enabled VFD* (only available for HY series VFDs)   

 **DoorSwitchType** *controls whether or not the door/Estop switch is Normally Open (NO), or Normally Closed (NC)*

**NOTE:**
***selecting NC requires a Normally Closed switch to be attached at all times or the system will not run***
***selecting NO allows a Normal Open switch to be installed or not***

**Default firmware is:**  ***CNC_xPRO_V5_XYYZ_PWM_NO.bin*** *(this denoted a 3 axis dual Y motor machine, with a pwm spindle(0-5v, or 0-10v), and a normally open door switch)*

Mboss32, 2021-04-07:
> Is there anything new in the firmware posted here that the rest of us should upgrade for?

This release was primarily to push the precompiled options for XYZA, RS485, and NC door switches.  A few minor tweaks to the code but nothing major.

closed 2021-04-07 by Mboss32