Spark-Concepts/xPro-V5 #6: motor burned up

IdahoCreator, 2021-02-07:
I have burned up a nema 23 2.8a stepper. All othe steppers are fine but were extremely hot. I cant find any info on wiki on how to adjust stepper voltage. I assume its in software. This occurred after I upgraded to the newest firmware as my controllercwas stopping in the middle of cut programs. That issue was resolved with firmware upgrade. Thanks for your help

Spark-Concepts, 2021-02-10:
Definitely sounds like your motor current is set high.  By default, we use a setting for 2.8A Nema23s - the current can be adjusted using these settings ( https://github.com/Spark-Concepts/xPro-V5/wiki/Changing-settings#xcurrentrun-or-ccurrentrun-140-thru-145--xyzabc--stepper-run-current-extended-settings).  Note the Trinamics work with RMS not MAX current.

IdahoCreator, 2021-02-13:
Ok I will adjust when I get my new motor. You may check the newest firmware as I didnt have this problem prior to the update.just verifing $140-143 are the adjustmemta I need to.make. i will set them at 2.4amps as I am running 2.8a nema 23.

Spark-Concepts, 2021-02-16:
We will definitely double check the firmware.  For a 2.8A motor, you will need to set the value to 1.9A or below.  The setting controls RMS current, so 0.707 * Max.  Would recommend a setting of around 1.8A for your motors to give a little overhead in the max current.

closed 2021-04-07 by Mboss32