Spark-Concepts/xPro-V5 #7: question about setting up

Papapayana, 2021-02-15:
so im looking to ad the ability to use a laser engraver along with my cnc mill and was wondering if i wanted to use tinyurl.com/3s1ns3yc  would i be able to have it hooked up along with my VFD or would i have to find a differant way to do it?

wolfkabal, 2021-02-24:
I'm not an expert, but I think through the use of a splitter adapter from the "tool" plug, with an inline SPDT switch, you could then have both VFD and Laser both connected, and via the switch, control which is active.

https://docs.openbuilds.com/doku.php?id=docs:blackbox:iot-and-laser - here's a reference to what's used for a basic spindle (via IoT Relay) - but the same could be done for laser and vfd I'd presume.

Mboss32, 2021-04-07:
This should work perfectly fine.  I would argue that grabbing a second 4pin header plug then wire the PWM spindle to one plug,  the PWM Laser to the other.  Clearly mark the two and just swap them in the controller when needed.  Greatly simplifies things and avoids some potentially painful oops.

closed 2021-04-07 by Mboss32