Spark-Concepts/xPro-V5 #8: Firmware Source?

wolfkabal, 2021-02-22:
Where is the firmware source located? 

Documentation indicates in multiple locations "additional configuration in config.h". Is the xPro-V5 source available for custom compilation?

Spark-Concepts, 2021-03-31:
The precompiled firmware is named to clearly identify specific compile time options - you can find the most ***up-to-date*** firmware here: https://github.com/Spark-Concepts/xPro-V5

CNC_xPRO_V5_  | MotoAssignment_ | SpindleType_ | DoorSwitchType.bin
------------- | -------------|------------- | -------------
" | XYYZ | PWM | NO
" | XYAZ | PWM | NO
" | XYYZ | PWM | NC
" | XYAZ | PWM | NC
" | XYYZ | 485 | NO
" | XYAZ | 485 | NO
" | XYYZ | 485 | NC
" | XYAZ | 485 | NC

**MotorAssignment:** *the options are XYYZ (3 axis, ganged Y) and XYAZ (4 axis)*

**SpindleType:** *the options are PWM which allows for 0-5V or 0-10V spindle (or laser) and 485 allows for the use of an RS485 enabled VFD* (only available for HY series VFDs)   

 **DoorSwitchType** *controls whether or not the door/Estop switch is Normally Open (NO), or Normally Closed (NC)*

**NOTE:**
***selecting NC requires a Normally Closed switch to be attached at all times or the system will not run***
***selecting NO allows a Normal Open switch to be installed or not***

**Default firmware is:**  ***CNC_xPRO_V5_XYYZ_PWM_NO.bin*** *(this denoted a 3 axis dual Y motor machine, with a pwm spindle(0-5v, or 0-10v), and a normally open door switch)*

Mboss32, 2021-04-13:
Hi Arthur,  I fully understand and appreciate your concern.  This repo "xPRO V5" is set up to push specific hardware/firmware updates and troubleshooting to the xPRO V5 hardware end-user, not firmware developers.  For those interested in firmware development, source code is freely available on our website as a download (with the appropriate licensing, and cautions).  With limited man-hours this allows us to keep this repo focus on hardware-specific information.

arthurwolf, 2021-04-13:
The source code *for those exact bins* is available on your website?
How do I know exactly what was compiled into each bin?
If I can't map a specific bin to an exact ensemble of source code, that's
obfuscation (even if it's not malicious).

It's generally good pratice to keep the source code packages/available
along with binary files.
If that's inconvenient, there are alternative solutions:
* You should at least have the bins accompanied by enough information (a
Readme?) to map specific bins to specific source code (or is it build
options in this case? That still needs to be explained/mapped )
* Similarly, the bins should be accompanied by their license file, or at
least a link/enough information to find said licensing information.

Essentially: It's an issue that coming here I can not find the source code
or any way to find the source code, I can not find a license or even know
what the license is, and I can not know what version/build option/diff of
the code was compiled into those bins.
It's not proper practice for Open-Source code publication.

Hope this helps.
Cheers.

nxperera, 2021-04-13:
Hi MBoss32, 
I am interested in the source code for the firmware for the xPro V5 I bought last month, would you be able to let me know where I can down load it? I was looking at the website https://www.spark-concepts.com/ I can find anywhere I can download the source code.

Mboss32, 2021-04-14:
Hi nxperera, if you search the site for "firmware" it will popup as a free downloadable product. If you are still having trouble let me know and I will grab the link when I get back to a computer.

nxperera, 2021-04-14:
Mboss32, Thanks I was able to get it. Next is how to build this, hope platform io or Arduino environment would do it. Will post my update on the build here. Planing to load to a ESP32 controller first and see if I can get the WebUI running so that I can reload old firmware again in case :).

nxperera, 2021-04-15:
I was not able to build from CNC_xPRO_V5_20210318 code base as is yet but I compared the code to Bart Dring's Grbl_ESP32 code base which I am able to build and run on another ESP32, the both source bases are pretty similar very minimal customizations. Also looks like Spark didn't use platformIO so trying Arduino IDE to build but it keep failing on required libraries. 
MBoss32 please let me know if you already know what I need to do to get this compiling, I know you guys are busy and the hardware is superb but the firmware code base (code organization at least) looks hackish, if you load it to GitHub people can help to clean it up.

Mboss32, 2021-04-16:
nxperera, the code is very similar to Bart's with customizations made for the specific hardware on the controller.   We do use Arduino IDE to program the controllers (directions here: https://github.com/bdring/Grbl_Esp32/wiki/Compiling-with-Arduino-IDE).  The code definitely needs a clean, what is posted is the in dev portion of it, the github repo for firmware is in progress.

nxperera, 2021-04-16:
Mboss32 thanks for the info I'll try again, It's wonderful to hear you are building the git repo for this.

nxperera, 2021-04-16:
Ok MBoss32, Thanks I was able to finally build it and load to an ESP32 lying around on my desk and yes it runs !!!

      ⸮ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4

[MSG:Grbl_ESP32 Ver 1.3a Date 20210311]
[MSG:Compiled with ESP32 SDK:v3.3.5-1-g85c43024c]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_PWM_NO]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test failed. Check motor power]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,300.000)]
[MSG:Y  Axis Trinamic driver test failed. Check motor power]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,300.000)]
[MSG:Y2 Axis Trinamic driver test failed. Check motor power]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test failed. Check motor power]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:BT Started with CNC_xPRO_V5]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]

Grbl 1.3a ['$' for help]

brianredbeard, 2021-07-03:
As folks can see from the automated messages above I spent a moment unwinding the changes made to https://github.com/bdring/Grbl_Esp32.

I'll leave my commentary out of this issue.  For users who want to inspect the changes as a set of four unified diff patches, I've posted them as a gist: https://gist.github.com/brianredbeard/d682717a7fc4f4b3bb98a06bfdf44cc7.  

These would be the patches needed to convert _from_ the zip file provided by Spark Concepts _to_ the sources provided in Grbl_Esp32... thus most users will want to watch https://github.com/bdring/Grbl_Esp32/pull/934 or simply retrieve the file `src/Machines/CNC_xPRO_V5_XYYZ_PWM_NO.h` from the pull-request.

brianredbeard, 2021-07-08:
Also, as an FYI, I've discovered that there are (seemingly) _some_ changes which were made for the RS-485 builds which are not reflected in the source bundle.  That assessment stems from the following:

In https://github.com/Spark-Concepts/xPro-V5/issues/18#issuecomment-815652827 we see the following console output:

```
...
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485 Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
...
```

This was a bit confusing, because even when building from both the supplied source bundle and the upstream code I was seeing the following output:

```
...
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
An unexpected error was detected: [MSG:RS485 VFD spindle errors]
...
``` 

Based on the code at: [bdring/Grbl_Esp32#eaffb964/Grbl_Esp32/src/Spindles/VFDSpindle.cpp#L369](https://github.com/bdring/Grbl_Esp32/blame/eaffb964f7f1585336b3e354ea59dd544599b815/Grbl_Esp32/src/Spindles/VFDSpindle.cpp#L369) we see:

```c++
//Lines 366 - 371
#ifdef VFD_RS485_RTS_PIN
        _rts_pin = VFD_RS485_RTS_PIN;
#else
        grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "Undefined VFD_RS485_RTS_PIN");
        pins_settings_ok = false;
#endif
```

We notice that on the next line that `pins_settings_ok` is set to `false`. This triggers an error state (which it _does_ in the upstream) when `VFD::init()` is called (checking the pin state on line 276) returns before it could then call `config_message()` on line 343.

```c++
        if (!get_pins_and_settings()) {  // line 276
            vfd_ok = false;
            grbl_msg_sendf(CLIENT_SERIAL, MsgLevel::Info, "RS485 VFD spindle errors");
            return;
        }
```

The sleight of hand here is that `config_message()` is clearly called in the first console output stanza above because we see `[MSG:VFD RS485 Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]` and **_not_** the error on line 278 which appears in the second console output stanza.

As a hail mary (noting the diagnostic output `RTS:GPIO(0)`) I tried setting the following in the machine definition header:

```c++
#define VFD_RS485_RTS_PIN               GPIO_NUM_0
```

Which then reported the following console output:

```
...
[MSG:Initializing RS485 VFD spindle]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
...
```

After making this change I was able to start the spindle with `M3 S<RPM>`.

I'm not ruling out that _why_ RS-485 not working from the source bundle could be a red herring, but setting the RTS pin to gpio0 **_did_** get things working.

Spark-Concepts, 2021-07-10:
While we do use GRBL_ESP32 core, there are specific changes made throughout to match the specific hardware implemented on the xPRO V5.  Dev files can be downloaded here:  (https://github.com/Spark-Concepts/xPro-V5-Firmware)[https://github.com/Spark-Concepts/xPro-V5-Firmware]

Most options are defined in the CNC_xPRO_V5_Machine_Template.h, simply make changes here, recompile and upload.

closed 2021-08-11 by Spark-Concepts

mathojojo, 2022-08-07:
Hello,

I see that last sources modifications for the xPro V5 (here : [https://github.com/Spark-Concepts/xPro-V5-Firmware](https://github.com/Spark-Concepts/xPro-V5-Firmware)) were done 2021 November.

Your precompiled firmwares had 2 updates April and June : 

April release Note : 
- Minor bug fixes
- TMC5160 Tuning
- Fixes bug requiring double reset for Z motor

June release Note : 
-  minor bug fixes and motor/current tuning 

I successfully compiled the sources, with the your CNC_xPRO_V5_Machine_Template.h file without changing anything, just to see if I ws able to do that : successfully

Does my freshly compiled binary have the bugs you fixed in april and June ? (especially the bug with double reset on Z motor). 

My thought is YES, I have thous bugs in my new binary, because I compiled with an old source. But can you confirm it ?

Thank you.