Spark-Concepts/xPro-V5 #9: Bluetooth pairing

stfarley, 2021-03-26:
Attempting to access with Bluetooth, the device is not found. I tried enabling “Bluetooth radio mode” now I can’t access it all. The Host WIFI network is not found as well as no Bluetooth device.

TRUDIM, 2021-03-30:
You could connect it via USB and change the settings back to WiFi.

stfarley, 2021-03-30:
Interestingly My MacBook and an android phone see the bluetooth device but not my iPhone or iPad. How does one connect via USB? I have a cable connected to it and the MacBook sees something. I imagine a driver and app are required.

stfarley, 2021-03-30:
I managed to get it recovered. Had to do the following.
Connect to a laptop with USB
Load CNCjs
Enter the command $Radio/Mode=AP
After reseting the network is working again in AP mode again.

stfarley, 2021-04-01:
BTW,
Bluetooth doesn't seem to work. 
I can connect to the bluetooth target in CRCjs. The Console opens.
However I can't seem to control anything.

Mboss32, 2021-04-03:
Yep, Bluetooth is available not fully flushed out or advertised.  The plan for the BT connection is an app down the road, but for now it is Wifi or USB only.

philshel1, 2021-04-03:
I'm beginning to believe that most of the "features" are BS and not implemented yet.

Mboss32, 2021-04-03:
Like which ones?  I know there is framework buried for future upgrades but pretty sure all the advertised features are there and functioning.

philshel1, 2021-04-03:
Well, the main one is "Motor drivers can be dynamically assigned to axes, so a 4 motor XYZA controller could be converted to a XYYZ (dual motor Y axis) without any hardware changes."  That is not implemented and when ever someone questions when that will happen, you can hear the crickets.  Plus it appears there are issues with controlling spindles, and now the BT is "buried for future upgrades".  I know that I have not been able to get it to connect to my my network at all, it always reverts back to AP mode no matter how I set it up.  I suspect it' has to do to the radio only wanting to connect to 2.4 GHz like some other ESP32 radios. It may not be ready for primetime I fear.

closed 2021-04-07 by Mboss32