Spark-Concepts/xPro-V5 #10: RS485 controlling a VFD

allistar, 2021-04-01:
The RS485 port has "A" and "B" plugs. How does that relate to the RS- and RS+ connectors on a Huanyang VFD?
I see there's a switch on the XPro V5 to enable the RS485 mode. Is that's all that's needed other than setting $30, $31? What should $Spindle/Type get set to?

stfarley, 2021-04-01:
I am trying to get the same config working. Made the same assumption about the connections. No luck controlling the VFD though.

stfarley, 2021-04-01:
I see there are firmware downloads that appear to have spindle options. I tried one that ended in 485 but still didn't work.

Spark-Concepts, 2021-04-02:
Connect the Huanyang VFD to the xProV5:
[image]
Using the procedure in the Huanyang manual, set the following register values:

Set PD000 to 0 - unlock parameters
Set PD001 to 2 - Command source is RS485
Set PD002 to 2 - Speed source is RS485
Set PD163 to 1 - Communications address 1
Set PD164 to 1 - 9600 b/s
Set PD165 to 3 - 8 Bit No Parity - RTU
Set PD000 to 1 - lock parameters
[Huanyang-VFD-manual.pdf](https://github.com/Spark-Concepts/xPro-V5/files/6247012/Huanyang-VFD-manual.pdf)

allistar, 2021-04-02:
Thank for the VFD config, I've made those changes. Can you confirm these settings:

- in the VFD do we leave the J1 jumper on the right hand two pins, or move it to the left hand two pins?
- on the XPro we move the little switch to the "RS485 A/B" setting?
- What do we set "$spindle/type" to? I notice that this is valid "$spindle/type=huanyang". Is that the correct one to use?
- What other "$spindle" settings do we need to change to get this to work?

When the VFD is powered on he display is flashing with F0000. I assume that's because it's waiting for commands from the RS485 input?

Thanks,
Allistar.

stfarley, 2021-04-02:
OK,
I tried this config with CNC_xPRO_V5_XYYZ_485_NC.bin
Still no luck with the VFD

If I send
S100
M23

I get and Invalid Gcode error.
The VFD just flashed 00000.

I also got an Alarm 10 at one point. Not sure what that is about.

stfarley, 2021-04-02:
BTW I took a look at the signalling with a scope. I see what appears to be some continuous messaging coming across the interface. Strangely though it appears that the signals on both the A and B lines are in the same phase. Should these not be differential? I also see a lot of common mode noise. 
RS 485 protocol decoding seems to make sense although I don't know what data I should be seeing.

allistar, 2021-04-02:
I finally have this working. Here's what I did:
- follow the wiring instructions and VFD settings provided previously on the issue.
- download the latest firmware (CNC_xPRO_V5_XYYZ_485_NO.bin). Install that firmware.
- send this command to the XPro: "$spindle/type=huanyang"

Now an "M3" starts the spindle and "M5" stops it. I have my "max spindle speed" set to 3000 which I don't think is right because it doesn't match what the spindle can do. I need to double check the settings. But at least it's working now.

It would be good if the wiki docs mention the "huanyang" spindle/type setting. At the moment there isn't any spindle/type in the docs that seem to be for RS485.

allistar, 2021-04-02:
I believe I now have this working fully. I have no idea how fast the spindle is actually running but I can tell that the "S" Gcode speed commands matches the frequency I would expect. The second number is the frequency shown on the VFD:

S500 = 8.33
S1000 = 16.66
S3000 = 50
S10000 = 166.6
S20000 = 333.3

I didn't push it more than this, I flinch when it's going that fast!

I notice in the ESP3D web page that pressing the "spindle" button doesn't do anything. I would expect this to start and stop the spindle. The only way I can start it is with M3 and M5 Gcode code commands. Is there a trick to getting the spindle button working?

Mboss32, 2021-04-05:
the $spindle/type = " does need to be defined (this should be done by default in the 485 firmware, looking into that now).   Could you send command $I to your board and paste the response here?  

The spindle button in the websender is an override stop - this is only an option when a gcode file has been paused.  You will need to send M3 and the speed number to turn on the spindle, M5 to turn off.  I would recommend defining a macro if this is something you want to do frequently.

allistar, 2021-04-05:
I have the spindle working well now. The $spindle/type is set to "huanyang". I've added buttons to the ESP32 interface to manually start and stop the spindle, this works well. I'm now only using ESP32 for GRBL configuration and am using CNCjs for running programs.

stfarley, 2021-04-06:
What I get is this
$I
[VER:1.3a.20201212:]
[OPT:MPHBSW]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NC]
[MSG:Mode=AP:SSDI=CNC_xPRO_V5:IP=192.168.0.1:MAC=3C-71-BF-DE-A8-6D]
[MSG:No BT]
ok

stfarley, 2021-04-06:
Also tried 
$Spindle/type=h2a
Still get a Gcode error when I try and start the spindle

Mboss32, 2021-04-06:
What is the gcode statement you are sending?  What brand is your spindle - with 485 each VFD manufacturer has a different set of commands (so an H2A setting for an HY spindle will do weird things)

stfarley, 2021-04-06:
It is a Huanyang VFD controller.

Mboss32, 2021-04-06:
Good, you need to set "$spindle/type=huanyang"

How are you trying to start the spindle when you get the gcode error?  Are you using a button, sending an M3 command, etc

stfarley, 2021-04-06:
Ah My mistake. I thought I read somewhere M23 to start. Using the correct command works much better.
BTW I see you recommended 100 for Min speed. (Water cooled) This value is in Hz which corresponds to 6000RPM. Is that the expectation?

Mboss32, 2021-04-06:
you know, I saw M23 above but thought it was a typo for M3... should have caught that!  Glad to hear it is working now :) 

For water-cooled spindles, the 100 min speed (6000RPM) is the general recommendation.  For a 4pole spindle you can go a bit slower. Note: these are generally accepted values/recommendations - each spindle manufacturer will vary, if you can find a datasheet for the spindle you purchased always follow that.

closed 2021-04-07 by Mboss32