Spark-Concepts/xPro-V5 #11: E Stop NC Switch

stfarley, 2021-04-01:
I have the Estop switch working. However it is Backwards.
i.e. you have to press it for run and pull it to stop.
I don't see a polarity setting for this anywhere.

Spark-Concepts, 2021-04-01:
Look at the backside of your estop button - it should have a switch symbol and marked either "NC" or "NO"
[NC_e-stop-hookup&bin.zip](https://github.com/Spark-Concepts/xPro-V5/files/6245872/NC_e-stop-hookup.bin.zip)
[NO_e-stop-hookup&bin.zip](https://github.com/Spark-Concepts/xPro-V5/files/6245873/NO_e-stop-hookup.bin.zip)

If your switch is NC:
[image]

If your switch is NO:
[image]

stfarley, 2021-04-01:
Ah Thanks, That did the trick

closed 2021-04-07 by Mboss32