Spark-Concepts/xPro-V5 #12: How would I best square the X Axis on a dual Y machine

stfarley, 2021-04-01:
With two Y Axis motors I would expect that one would need to regularly square the X Axis.
The only process I can come up with for this is to run it close to one end, Turn off the controller and manually spin one stepper motor until both sides are the same distance from the end plate.

Any better ideas?

TRUDIM, 2021-04-02:
I am struggling squaring the frame as well. Had been following instructions to move the X axis all the way to the front until it hits the front support (on a LEAD 1515) and supposedly it will square it. Once i get the bolts tight on the front am moving the X axis to the back of the machine for it to be square and it all goes to hell. 
Measuring diagonals I get a deviation of about 7-8mm. I did this procwdure several times using all kind of different approaches with no luck. 
Decided on leaving it like that for now so that I could at least start using the cnc...

allistar, 2021-04-02:
I've just build a Lead 1515 and I got the X axis squared with Y using digital calipers. I move both Y gantries near the front of the machine and measured exactly how far away the gantry was from the end. I then turned the lead screws by hand which would move one gantry very slightly in the direction needed. I'd then measure it again. It's within 0.1mm which is good enough for me. The frame itself came out nearly perfectly square (when measuring diagonals).

I found squaring the dual X C-beams was much more difficult (I have the high-Z mod). That's now within 0.1mm across 1500m which is good enough for me.

stfarley, 2021-04-06:
I assume there will still be an issue any time a stepper slips though. This Manual process would need to be done pretty regularly. Is there a method that uses two limit switches that would allow it to re-square everything time you home.

allistar, 2021-04-06:
Yes, if a single Y stepper misses steps then it will put it out of square. I'm not sure with the XYYZ setup if it can control the two Y steppers independently. One approach could be to home it, then unplug the stepper that's in the correct position and then home again with just the other stepper plugged in. Though if the homing sequence moves 5mm that may be too much to force it out of square (5mm for my machine would be 1000 steps which is heaps).

Another approach is to mark on the Y rails alignment marks every 10mm or so so that at any time you can see if the gantry is square by lining it up one one side of the each Y gantry plate. This way squaring by hand becomes a lot easier. If my machine loses steps in the Y direction this is probably what I'll do.

tobyrobb, 2021-04-09:
I thought there was an option for dual homing y axis, but now i cant find it..

nxperera, 2021-04-13:
Here is what I do :  With xPro V5 I set $Homing/Squared=y and fixed additional limit switch on the second Y connect both parallel to Y limit switch input in controller. make sure line of two limit switches is perpendicular to Y axis.  And I use command $HY to home just Y axis first which will do squaring also.
(once you square like this once use 3-4-5 method to manually verify if the CNC is squared properly if you miss any alignments adjust the limit switch to compensate the miss alignment. This is a repetitive, trial and error process until you get it perfect. Obviously the accuracy/repeatability of  limit switch affect the accuracy of squaring, I find the 4mm Inductive Proximity Switch(SN04-N DC NPN)  works OK for me.)

stfarley, 2021-04-13:
In order to square up, wouldn't each switch need it's own input? If you parallel them the controller can only tell that one of them has hit it's limit (If they are NO) or that both have hit their limits (if they are NC). In order to square it the controller would  need to know when that one limit has been hit and keep moving the second one until it trips.

nxperera, 2021-04-15:
> In order to square up, wouldn't each switch need it's own input? If you parallel them the controller can only tell that one of them has hit it's limit (If they are NO) or that both have hit their [...]

Not really actually. The Squaring routine first back off both motors/rails (Y and A) and then it try homing one at a time. First Y motor then A motor. Yes both limit switches Y and A are connected to same Y limit input but it get triggered one at a time in final squaring. Let me explain step by step how it works for me:
1. run $Home/Y 
2.  both Y and A motors rapid toward homing direction one or both ( depending on proximity) limits switches get triggered .
3. Then both Y and X bounce back by the amount given in $Homing/Pulloff  
4. Seek Y motor until it triggers Y Limit, and bounce
5. Seek A motor until it triggers Y Limit, and bounce
6. Y machine zero is set

Mboss32, 2021-04-16:
great summary of the squaring routine nxperera!  Only note I would add is that NO switches must be wired in parallel on the Y axis limit plug, NC switch needs to be wired in series to the Y axis limit plug.  This way the controller can read a single switch trigger based on which motor, Y or Y2 is moving.  This is only as accurate as the placement, and sensing, of your switches though.  If your switches as 2mm off so will be the final squaring.  

nxperera, can you give more detail on what went weird on the new firmware?

nxperera, 2021-04-16:
I'll reload NC_xPRO_V5_XYYZ_PWM_NO.bin tonight and confirm exact details but as I remember now what happened was it tried to seek Y and A at the same time and failed also what I noticed was the steppers were creeking too loud and felt something was so wrong.

allistar, 2021-05-21:
I see there's mention of squaring in the settings wiki page, but the link doesn't yet go anywhere. Is the $Homing/Squaring  functionality developed and just needs documenting, or is this still a work in progress?

allistar, 2021-08-10:
For this auto squaring solution to work I think we need to be able to set difference "bounce off" values for each Y axis. This means the limit switches don't need to be absolutely perfectly aligned. If one if offset from the other by 0.5mm then we could adjust that axis bounce value to be 0.5mm more (or less). Finding what this value should be would be pretty straight forward with some digital calipers.

I'm quite keen to get this functionality working. Do you entertain the possibility of letting other people submit pull requests to the controller code?

TRUDIM, 2021-08-10:
Yes, please! 😁

Spark-Concepts, 2021-08-11:
Absolutely would take a pull request.  sounds like a great addition to the squaring feature

allistar, 2021-08-11:
Should this be a change to the upstream GRBL32 code or is the XPro code base different enough that it should be only in the XPro code?

Spark-Concepts, 2021-08-12:
Good question, ideally push to the upstream GRBL32 code so more folks can use it.  Our changes are strictly for the hardware integration.  This would likely need to include the use of the Y2 limit pin, the current squaring routine checks only the Y and cannot differentiate between the Y1 or the Y2 sides.

allistar, 2021-08-12:
If both limits were connected as NC in series or NO in parallel then they could both be plugged into a single input on the controller. We don't need to differentiate which Y gantry trips the limit switch, only that one has. I'll investigate the upstream repo and discuss this with their maintainers.

allistar, 2021-08-12:
It looks like this feature is already implemented in Grbl_Esp32:
https://github.com/bdring/Grbl_Esp32/wiki/Motor-Ganging-and-Axis-Squaring
The code for this appears to be here:
https://github.com/bdring/Grbl_Esp32/blob/main/Grbl_Esp32/src/MotionControl.cpp#L353

If this is right, what will it take to get this into the XPro firmware source?

allistar, 2021-08-12:
The ability to set independent offsets for each limit switch has been implemented in a beta branch, the details of this are here:

https://github.com/bdring/Grbl_Esp32/issues/953

Spark-Concepts, 2021-08-12:
Nice thank you!  We are tracking the YAML development but wont implement it until it comes out of Beta and we get a chance to torture test it.  

The general squaring routine is implemented, you can turn it on using  $Homing/Squared=Y, then executing command $Hy

allistar, 2021-08-12:
The new fluid homing supports homing the Y axis without forcing it out of square. It's nice that the XPro V5 has two Y limit switch inputs to support this.

AzionHzmilton, 2022-06-05:
Sorry - relatively new to this but I'm not clear base on the comments above if I need to still wire my limit switches in series or with the new fluid homing, I'd be just wiring up the second switch directly.  Is there a doc that outlines the wiring and process for multi switch squaring?

hellboytat, 2022-06-06:
Im also interested in this!

ImagineerNL, 2022-08-17:
@Spark-Concepts , would have been great if the [information given in the closed issue](https://github.com/Spark-Concepts/xPro-V5/issues/93#issuecomment-1003234085) would have been stated here as well as in the wiki. 

> For axis squaring, this can be done with the default firmware using 2 switches on the Y limit port

So for Automatic squaring of your Y Axis:

1. Install a limit switch on both of the Y Axes, making sure both limit switches are positioned **exactly** the same distance from the start of the Axis.
2. Wire and plug both limit switches to the **same** Y limit switch port. ; this means that the Y2 limit switch is **NOT** connected to the A/Y2 port. Currently the A/Y2 limit switch port seems to be out of use.
3. power on the controller.
4. trigger the Y1 limit switch by hand. confirm it triggers.
5. reset and reconnect (if you have hard limits enabled)
6. trigger the Y2 limit switch by hand. confirm it triggers
7. reset and reconnect (if you have hard limits enabled)
8. in console: $Homing/Squared=Y
9. click on the Y home in the jogging panel or type $Hy in the console
10. watch the Y home closely while keeping your hands close to the emergency stop.

What should be happening.

1. Y Homes until it hits any of the 2 switches.
2. System will debounce both Y1 and Y2 motors
3. System will slowly move Y1 motor until it reaches the limit switch again and debounce a bit.
4. System will then slowly move Y2 motor until it reaches the limit switch and debounce a bit.

This should square out the system as long as you are 100% sure both limit switches are in exactly the same spot.