Spark-Concepts/xPro-V5 #15: Probe failure

allistar, 2021-04-04:
I have the probe hooked up but am getting an error when running a probe cycle: _Probe fail. Probe is not in the expected initial state before starting probe cycle._

I know that the wiring is good as I tested it with a multimeter. There is no short. However, when it's plugged into the xPRO the multimeter shows a connection between the probe plate and the little clamp. This connection is coming from inside the xPRO. It doesn't matter what $probe/invert is set to. When it's plugged in the little green probe light on the side of the xPRO turns on. When I touch the clamp to the probe plate the green light gets brighter.

Does anyone have any suggestions on how I diagnose this? Are there any internal jumpers I can change?
I'm running the latest "XYYZ, 485, NC" firmware.
Thanks.

Mboss32, 2021-04-05:
do you have an NO or NC probe?  If you unplug the probe from the xPRO, does the green probe light stay on?

allistar, 2021-04-05:
It's NO. When the probe isn't plugged in the light isn't on. The light on the side of the XPro works as expected - it lights up with the touch probe "touches" and turns off when it doesn't. But it doesn't work when actually running a touch sequence.

There's a possible cause of this issue which I will investigate after work: I'm using shielded 4 core stepper motor wire for both the touch probe and the E stop (both pairs run down the length of the machine together in a single insulated wire). The E-stop is NC which means that while the machine is powered on it has current running down it. I wonder whether this is interfering with the probe wires and giving a false positive reading. I'll try with separate wires to see if it fixes it.

Mboss32, 2021-04-06:
Let me know how the test goes - since the light on the xPRO goes off when you unplug the probe, the issue is somewhere in the plug, probe, or wiring.  One very common issue is wire strands crossing between slots on the plug-in connector - if you see any of these, remove the wires from the plug, and re-insert carefully, or twist and tin the wire before inserting.

allistar, 2021-04-08:
I finally figured it out. The probe worked always, just not when the clip was clamped to the bit in the spindle. The reason is that the wire connected to the spindle needs to go to the ground wire on the probe plug. I assume this is because the spindle is grounded. With it the other way around it wouldn't work. It was a simple problem in the end and the probe is now working well.
Thanks for your help with this.

closed 2021-04-08 by allistar

tobyrobb, 2021-04-08:
My machine spindle is also grounded and so having the red "positive" conductor connected to the clip triggers the probe logic also..

Mboss32, 2021-04-08:
Thanks for letting us know, I'll make sure that an alternate wiring diagram is available for the grounded spindles/machines