Spark-Concepts/xPro-V5 #16: OpenBuilds Interface pendant

allistar, 2021-04-05:
Are UART pins exposed in some way so this can be plugged into the XPro?

https://docs.openbuilds.com/doku.php?id=docs:interface:start

Mboss32, 2021-04-06:
The UART pins are not exposed, however, and fire table (~$49) over WiFi gives many of the same features :)

allistar, 2021-04-06:
Thanks. Can you provide a link to the "Fire Table"? A google search doesn't reveal anything.

Mboss32, 2021-04-07:
Sorry, that was a typo - [Amazon Fire Tablet ](https://www.amazon.com/dp/B07FKR6KXF/ref=fcatpage_shoveler_1?pf_rd_r=KKZXT8V04Y9N82YFE8GK&pf_rd_p=6470d94d-43cd-4a4d-a1a2-5bfe9651c9dd&pf_rd_m=ATVPDKIKX0DER&pf_rd_s=merchandised-search-1&pf_rd_t=&pf_rd_i=6669703011&ref=fcatpage_desktop_f7)

connect to the wifi -> looks like this :) 

[image]

Note: some android versions do not like the captive portal so you may need to open a browser and navigate to 192.168.0.1 (AP mode) or your designated station IP.

allistar, 2021-04-07:
Ah, I see! Thanks. My other machine uses LinuxCNC and I like the "continuous jog" feature (I use an xbox controller for this). Is this possible with a non realtime system like GRBL? It's annoying to have to nudge the axis 100mm at a time. Also, can I control ESP32 on a tablet while I'll running a program using CNCjs? Or does one prevent the use of the other?

Mboss32, 2021-04-08:
The ability for a continuous jogs is built into the firmware but very few of the sender programs have them implemented.  It is on the roadmap for the WebUI refresh but is further down the list, so for now, we are still in discrete jogging mode.  

You can run the WebUI and a CNCjs at the same time.  Machine positions will track between the two as well.  Just start CNCjs first as it spits out a bunch of initialization statements at the start.  Also, don't initialize a USB connection during a cut from the SD card - the USB connection forces a restart of the firmware and you will lose your cut :).

allistar, 2021-04-09:
Is continuous jogging just a case of sending it a particular $ command to turn jogging on and then another command to turn it off?

Is there a way to get something like an XBox controller to allow the machine to be jogged by moving the joystick (up = y, left/right = X)? On Linux CNC it's a case of mapping the XBox inputs to key presses that the Axis UI responds to. I understand though that LinuxCNC is realtime while GRBL isn't.

Or is it the case with GRBL based systems that everyone jogs by repeatedly "nudging" it in fixed increments?

closed 2021-09-16 by Spark-Concepts