Spark-Concepts/xPro-V5 #17: Connecting router

TRUDIM, 2021-04-06:
What am I not connecting right as it seems that I am not able to turn my router on/off using M3/M5 commands? I can hear the relay inside the xPro click but the router does not turn on. 
What I am using to connect the router to the xPro is an old power socket where the router is plugged in (did not wanted to cut the plug on the router) and that goes into the controller via TOOLHEAD's ground and power.

I see that there is an IoT relay used in the illustrations but is that the only way?!?

Mboss32, 2021-04-07:
Could you upload a picture of your setup?

TRUDIM, 2021-04-07:
There it is. It's just a temporaty setup until everything works as it should!

[image]

Mboss32, 2021-04-08:
Thanks - that connection is a 0-5V pwm output.  This is a control signal typically used to either trigger a relay (like the IoT) or to drive a DC speed controller or VFD.  This output will not be able to power a router.  

Instead of the IoT Relay you could also use a Solid State Relay.

TRUDIM, 2021-04-08:
So in my understanding an SSR of this type will do the job?
[image]

Spark-Concepts, 2021-04-08:
[image]

TRUDIM, 2021-04-08:
At possition 1 on the SSR I have to hook it up to a power outlet? SSR controlls through ground or phase (don't think it matters)?
Does the higher current SSR means lower temperature?

Spark-Concepts, 2021-04-08:
Correct,

The SSRs are able to switch the "ON" and "OFF" state of the load very quickly through the electronic circuit, just like the function of traditional mechanical relays. Compared with the Electromechanical relay, there is no movable mechanical part inside the SSR, and there is also no mechanical action during the switching process of the SSR. Therefore, the Solid-State Relay is also called "non-contact switch". The main advantages of solid state relays are as follows:
- The semiconductor component acts as a switch for the relay, which is small in size (compact size) and long in life (long lifetime).

- Better Electro-Magnetic Compatibility than EMR - immunity to Radio Frequency Interference (RFI) and Electro-Magnetic Interference (EMI), low electromagnetic interference, and low electromagnetic radiation.

- No moving parts, no mechanical wear, no action noise, no mechanical failure, and high reliability.

- No spark, no arc, no burning, no contact bounce, and no wear between contacts.

- High sensitivity, low electrical level (3.3V) control signals (SSR can directly drive large current loads through the small current control signals), compatible with logic circuit (TTL, CMOS, DTL, HTL circuits), easy to implement multiple functions.

Your neutral (ground) wire from your power cord are left uncut as is; however, Position 1 and 2 connect in series with your 220VAC Line (or Phase) - when an isolated voltage is applied to position 3 and 4 (3.3V and Ground from the xPro), Position 1 and 2 essentially make contact thus providing power to your router. When the 3.3V is removed, Position 1 and 2 open effectively removing voltage from your router
  
Higher Amperage means lower ON resistance (thus less potential heat) - when the SSR is triggered "ON" - make sure its rated above what router is rated at. Also, make sure the minimum "Input" side is rated at 3.2V...otherwise the SSR wont trigger.

TRUDIM, 2021-04-09:
That's some great info! Thank you!
I'll get two SSR's right away! So in order to automate the dust extraction I have to do the same hook up but at COOL output?

Regards,
Dimitar

Spark-Concepts, 2021-04-10:
TRUDIM,

Yes to answer your question you would do the same hook up, but to the COOL output.
[image]

TRUDIM, 2021-04-13:
Allright!

I got the relays, made a quick setup to see if everything works and got an issue.
When I engage M3 or M7 the relays are both outputting 240V, oddly when I plug in the Roüter11 in any of the two EU sockets the router starts working on impulses. The whole thing just pulsates and is not running smoothly (like I am rapidly switching it on and off). If plugged in a normal wall socket it works as it should. This problem occurs with other tools but is barely noticeable!
[image]

Made a quick video for better illustration of the problem.
https://drive.google.com/file/d/1K5bbKMFTtgrnaZ7H0XmYZfkh5KdnADoF/view?usp=sharing

Spark-Concepts, 2021-04-15:
TRUDIM,

If you don't mind move the "MIST" jumper<sub>1</sub> over to 24V as shown below and see if you're still getting the pulsations when triggering on the COOL output.
[image]

TRUDIM, 2021-04-16:
Installed ine of these: https://www.google.com/amp/s/www.digchip.com/datasheets/parts/datasheet/2135/S14K385_amp.php

Didn't change the router behavior. Will try connecting it to the on board telay. Might also get a more expensive SSR. The one I got was really cheap conpared to others...

closed 2021-04-21 by Spark-Concepts