Spark-Concepts/xPro-V5 #18: Huanyang VFD Control Issue

Genza12, 2021-04-08:
Controller side - I've updated the firmare to the **CNC_xPRO_V5_XYYZ_485_NO.bin** firmware following th instructions in the wiki, but when i try to issue the command **$Spindle/Type=HUANYANG // RS485** or **$Spindle/Type=H2A // RS485** the console spits out **error:2 (Bad number format)**. When trying to start the spindle the console spits out **[MSG:Critical Spindle RS485 Unresponsive]
ALARM:10**.

VFD side - I have connected the A on the controller to RS+ on the VFD and the B to RS-. Pd001 and Pd002 have both been set to 2 and the jumper is attached to VI. The Baud rate setting on the VFD is set to 9600 but when changing it on the CNCjs software to anything but 115200 I get no response from the software to any command what so ever.

Any help would be greatly appreciated.

Genza12, 2021-04-08:
Update, Im now able to turn on the spindle using M3 but this is the error I now get.

**Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400df3cd  PS      : 0x00060431  A0      : 0x800defcf  A1      : 0x3ffbe800
A2      : 0x3ffd851c  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x3ffc7194  A8      : 0x00000001  A9      : 0x3ffbe8c0
A10     : 0x000000ff  A11     : 0x00000000  A12     : 0x3ffd8b68  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x400df3cd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008866c
Backtrace: 0x400df3cd:0x3ffbe800 0x400defcc:0x3ffbe8f0 0x400d54fc:0x3ffbe920 0x40081f7b:0x3ffbe940 0x40082405:0x3ffbe960 0x40085e35:0x3ffbe980 0x4000bfed:0x3ffbcac0 0x40091242:0x3ffbcad0 0x4008fc60:0x3ffbcaf0 0x4008fdd9:0x3ffbcb30 0x40082dd1:0x3ffbcb50 0x4013b362:0x3ffbcb70 0x4013b3fd:0x3ffbcb90 0x4013b7bf:0x3ffbcbb0 0x400ee5f5:0x3ffbcbe0 0x400eb2c1:0x3ffbcc10 0x400eafa4:0x3ffbcc30 0x400daafc:0x3ffbcc50 0x4009030d:0x3ffbcc70
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NO]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,300.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,300.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:Check limits]
[MSG:'$H'|'$X' to unlock]**

Genza12, 2021-04-08:
After Homing and trying to turn on the spindle I get an **ALARM:1 (Hard limit)**.

allistar, 2021-04-08:
Can you confirm you entered this command:
$Spindle/Type=HUANYANG
Without the "// RS485"at the end?

Also, when you type just "$spindle/type" what does it return? On mine it returns "Huanyang".

The baud rate in CNCjs is for the connection to the XPro, not the connection from the XPro and the VFD.
The Hard Limit seems related to the homing, not to the VFD. How are you trying to turn the VFD on? A "M3" should do it.

Genza12, 2021-04-08:
I have defined the spinfle type as huanyang without the // rs485.
When typing $spindle/type it returns huanyang.

when trying to turn the vfd on I issue an M3 command, it begins to start but straight away stops and throws up a hard limit alarm 1.

Genza12, 2021-04-08:
[image]
[image]
[image]
[image]

Genza12, 2021-04-08:
Update - So if I issue an M3 it will throw up all of that above but if I quickly issue a $X then another M3 S24000 then it will work and I will be able to change the speed and much as I want, when I then issue an M5 it will turn off but as soon as it stops its throws up all of those messages again. See Below.

**[MSG:Check limits]
[MSG:'$H'|'$X' to unlock]
> $x
[MSG:Caution: Unlocked]
ok
> m3
ok
> m3 s24000
ok
> m5
ok
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400df3cd  PS      : 0x00060c31  A0      : 0x800defcf  A1      : 0x3ffbe800
A2      : 0x3ffd8524  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x00000000
A6      : 0x00000001  A7      : 0x3ffc7194  A8      : 0x00000001  A9      : 0x3ffbe8c0
A10     : 0x000000ff  A11     : 0x00000000  A12     : 0x3ffd8b70  A13     : 0x00000000
A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0x00000000
Core 1 was running in ISR context:
EPC1    : 0x400df3cd  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008866c
Backtrace: 0x400df3cd:0x3ffbe800 0x400defcc:0x3ffbe8f0 0x400d54fc:0x3ffbe920 0x40081f7b:0x3ffbe940 0x40082405:0x3ffbe960 0x40085e35:0x3ffbe980 0x400ddf76:0x3ffd9350 0x400de04d:0x3ffd9370 0x4009030d:0x3ffd93c0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NO]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,300.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,300.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-300.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]**

tobyrobb, 2021-04-08:
I'm also having trouble with rs485 on my VFD , any hints on debugging rs485? How do I see what's being sent out on the 485 serial? Or can tell tell me if it is connected via 485 ok? Or do l get out my USB to serial and snoop the 485?

Mboss32, 2021-04-08:
@Genza12, sounds like a lot going on here.  It sounds like you got the setup sorted out but could you confirm that the wiring and all the registers have been set as below?

Connect the Huanyang VFD to the xProV5:
 [image]
 Using the procedure in the Huanyang manual, set the following register values:
 
 Set PD000 to 0 - **unlock parameters**
 Set PD001 to 2 - **Command source is RS485**
 Set PD002 to 2 - **Speed source is RS485**
 Set PD163 to 1 - **Communications address 1**
 Set PD164 to 1 - **9600 b/s (3 - 38400)**
 Set PD165 to 3 - **8 Bit No Parity - RTU**
 Set PD000 to 1 - **lock parameters**
 [Huanyang-VFD-manual.pdf](https://github.com/Spark-Concepts/xPro-V5/files/6247012/Huanyang-VFD-manual.pdf)
 
 _Also verify the following settings (based on the values on the motor's nameplate)_
 
 PD004 **Base frequency as listed on spindle (typically 400)**
 PD005 **Maximum frequency Hz (typical value for spindles is 400)**
 PD011 **Min speed (Recommended Air-cooled=120 Water=100)**
 PD014 **Acceleration time** (Test to optimize)
 PD015 **Deceleration time** (Test to optimize)
 PD023 **Reverse run enabled (set to 1)**
 
 PD141 Spindle max rated voltage (Typically 220)
 PD142 **Max rated motor current (0.8kw=3.7, 1.5kw=7.0, 2.2kw=??)**
 PD143 Motor Poles **This parameter is set for the number of the motor’s pole according to the nameplate of the motor - I believe this is only used for the conversion of Hz to RPM (typically 2 or 4)**
 PD144 **Rated Motor Revolution at 50Hz (Typically 3000 @ 50Hz and 24000 @ 400Hz)**

Next Step:

Set **$Spindle/Type=HUANYANG** without the //RS485, that was placed as a comment but definitely see how it is confusing.  

Now for the M3 command causing a Hard Limit trigger, since you are able to successfully home the machine without triggers, this is almost certainly noise caused by the VFD/spindle wire.   For a quick test set **$Limits/Hard=Off** then execute the M3 command.  Options from here are to keeping Homing Enabled but run the machine without the Hard limits (most common as most folks are really only using homing switch, not limit switches anyways), or upload a few pictures and we can start digging into noise isolation.

Mboss32, 2021-04-08:
@tobyrobb, check the setup on the comment directly above.  

For debugging, the best route is to use CNCjs and check the start-up/initialization messaging that comes out over USB.
1) Power on your VFD, power on xPRO
2) Open CNCjs, connect to the Silicon Labs port
3) Hit the physical **Reset** button on the xPRO

This will spit out a bunch of information - you are looking for the spindle information (directly below the trinamic self-tests)
If there is an issue with the communication you will see something like this:
[image]

While a successful start will look like this:
[image]

If still unsuccessful a few things to check:
1) One of the RS485 specific firmwares are loaded (CNC_xPRO_V5_....**_485_**..) 
2) The switch to select the EN/PWM or RS485 mode on the xPRO V5 uses a long actuator and will sometimes not fully seat.  Flip the switch back to the EN/PWM side then flip it to the RS485 side (use a small driver to make sure the base of the actuator has flipped fully)

tobyrobb, 2021-04-09:
@Mboss32  thanks for your help. The VFD is now working successfully.  BTW i am using the opencontrol software through the bluetooth serial and its working great, however, i have to reset the board each time to upload firmware via the webui. then set it back to BT and redo my settings. The GRBL settings page also does not allow me to save settings to the board but serial commands are doing it np.

tobyrobb, 2021-04-09:
@Mboss32 interestingly the Z axis was unresponsive when runinng the new CNC_xPRO_V5_XYYZ_485_NO.bin - working OK on CNC_xPRO_V5_XYYZ_RS485.bin

Genza12, 2021-04-09:
@Mboss32 All the parameters were set according to your instructions, I turned off hard limits and it seems to be working now.
When I turned it on, the Z axis was unresponsive at first as for some reason the stepper was not being powered, It seems to have been fixed by power cycling the controller. Thanks for your help, much appreciated.

closed 2021-04-09 by Genza12

VitaliiSadovshikov, 2022-11-14:
> For a quick test set **$Limits/Hard=Off** then execute the M3 command.

Sorry, can you elucidate a little bit more, what the console should spit out afterwards if the problem is caused by EMI, as you sayed.

In my case I get the same result (i.e. the message "ALARM:1 (Hard limit)" and as result stop my spindle) whether with $Limits/Hard=Off or $Limits/Hard=On. So, I should get such a result to ensure the problem originates from the noise caused by the VFD/spindle wire.

One more observation: if physically unplug all limit switch cables the message "ALARM:1 (Hard limit)" does NOT arise.

naseralariar, 2023-06-29:
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_TXD_PIN]
[MSG:Undefined VFD_RS485_RXD_PIN]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:RS485 VFD spindle errors]

i config evry thing but i dont khow what is problem ??

can anyone help me