Spark-Concepts/xPro-V5 #19: Error:120when updating ESP settings.

Slixxor, 2021-04-09:
We have a full management suite for Barts firmware and when trying to set network settings we're returning error:120 on your latest NO firmware advising SD job runnng.

Mboss32, 2021-04-09:
Hi slixxor, what software are you using to update the settings?  What setting is being updated and which of the NO firmware are you using?

Slixxor, 2021-04-13:
We are using our CNC3D Commander software and running your firmware on an ESP32 dev module (No SD attached which may be the issue). We have already done all of the ground work in our software as we are launching a similar controller to yours very soon.

A few users who have bulkman xpro V5's have reached out to us to see if it is compatible and yes indeed both of our controllers are identical in Command set (bar 2 that we have added separately).

We've set up our software to be a seemless integration with our Nighthawk Controller and automate things like firmware updates and settings.

Here is a screenshot of it connected:

[image]

Here is how the wifi list looks:
 
[image]

Here is where settings are done:

[image]

Here is how SD card files are displayed and accessible: 
[image]

The issue we have is when you hit "Update connection settings" we begin the process of sending the values until we finally issue [ESP400] for reset. You seem to be confirming all the steps along the way but then throw the ERROR:81 (this time) and none of the settings seems to be retained at all despite the fact you 'ok' confirmed each one.

Please see command sequences top being oldest. "<<" indicates sent to chip, ">>" receivied from chip.

`<< [ESP100]Sharp Software
>> ok
<< [ESP101]12345678
>> ok
<< [ESP102]STATIC
>> ok
<< [ESP103] IP=192.168.1.133 MSK=0.0.0.0 GW=192.168.1.1
>> ok
<< [ESP105]CNC_xPRO_V5
>> ok
<< [ESP106]12345678
>> ok
<< [ESP107]192.168.0.1
>> ok
<< [ESP108]1
>> ok
<< [ESP110]AP
>> ok
<< [ESP112]CNC_xPRO_V5
>> ok
<< [ESP120]ON
>> ok
<< [ESP121]80
>> ok
<< [ESP130]ON
>> ok
<< [ESP131]23
>> ok
<< [ESP400]RESTART
>> ok
>> ok
>> ok
>> ok
>> error:81
>> ok
>> ok
>> ok
>> ok
>> ok
>> ok
>> ok
>> ok
>> ok`

We've tried power cycling and reconnecting but it does not retain a single setting at all. Can you advise if there is an extra command you need to "keep" the settings?

Mboss32, 2021-04-13:
Nice looking sender!  I just fired up one of the V5 and made all these edits within the CNCjs interface with no problem.  Settings stick and no errors.  Maybe double-check your hardware setup?

Slixxor, 2021-04-13:
> 
> 
> Nice looking sender! I just fired up one of the V5 and made all these edits within the CNCjs interface with no problem. Settings stick and no errors. Maybe double-check your hardware setup?

Thank you. it's been a full on project :)

We do not have a physical xPRO V5 to test with just an ESP32 dev module with firmware loaded. I suspect this may be the issue but i'm not sure. Do you have the command sequence output for it?

Slixxor, 2021-05-03:
Ok,

I've got the bottom of this. 

Using a Dev board ESP32 and using old firmware caused the issue. Virtual EEPROM was offset. Also, needed to use newest version of esptool to load.

closed 2021-05-03 by Slixxor