Spark-Concepts/xPro-V5 #20: Z limit switch does not work

WWKSeric, 2021-04-11:
I have the xPro V5, using the open builds control;

I am struggling to get the limit switches to work on my machine. Both the X and the Y work perfectly but Z will not. 

Even with all the limit switches disconnected the Z shows on at all times. I tried plugging the Z into the Y port to confirm the switch works and it's fine. Also rings fine with my meter.

When I toggle the Z on the machine the LED light turns on the controller but does not lockout the machine, and nothing happens on the open builds program.

When I go into the trouble shoot tab on open builds the X and Y show "OFF" but the Z shows "ON". The Z will show on with the switch disconnected, connected, and when the switch is closed. It is a 2 wire limit switch wired as instructed. 

[image]

Spark-Concepts, 2021-04-11:
Hmm, if you can please connect to the webui, then take a screen shot of the console stream (updates every 3 seconds). 

Next, unplug all the limit switches and screen shot the console window.

WWKSeric, 2021-04-11:
This is what I get. Regardless of the switch being connected or disconnected

[image]

WWKSeric, 2021-04-11:
Could it be possible my Z pin is shorted on the board ?

Spark-Concepts, 2021-04-11:
It's possible... Is this a new problem, or was it like this from thw get-go?

First, try updating the firmware on your xPro - also
if you wouldn't mind, please post or email (bill@spark-concepts.com and mike@spark-concepts.com) when and who you purchased your xPro through.

Thx

WWKSeric, 2021-04-12:
From the get-go. I ordered the Workbee from Bulkman end of March, received it early April and assembled it last week. I have been able to run gcode successfully. The only thing that will not work is the Z limit.

Spark-Concepts, 2021-04-12:
Are you located in US?

WWKSeric, 2021-04-12:
I am located in Ontario, Canada. 

I tried the update "CNC_xPRO_V5_XYYZ_485_NO.bin" and no change.

What would be my best course of action now ?

philshel1, 2021-04-12:
> From the get-go. I ordered the Workbee from Bulkman end of March, received it early April and assembled it last week. I have been able to run gcode successfully. The only thing that will not work is the Z limit.

Really?  I ordered one on Feb12 and still have not received it in California.

WWKSeric, 2021-04-12:
> > From the get-go. I ordered the Workbee from Bulkman end of March, received it early April and assembled it last week. I have been able to run gcode successfully. The only thing that will not work is the Z limit.
> 
> Really? I ordered one on Feb12 and still have not received it in California.

Yup came pretty quick from Bulkman as a whole cnc kit. Other than this Z probe issue it's great !

philshel1, 2021-04-12:
Hmmm, I ordered the queen bee mechanical kit and as far as I can tell UPS is still waiting for it to hit stateside.

Spark-Concepts, 2021-04-12:
WWKSeric,

I'm sorry to hear you're having issues with your Z axis - that's certainly an odd issue you're having. 

We're happy to assist - we have two options here:
1. We can send you updated firmware and reassign the Z limit to use the A limit pin - this would be an immediate fix (assuming there are no issues with the A axis limit pin). The downside would be, if you want to use 4 axis homing at some point (_note: most 4 axis machines don't home the 4<sup>th</sup> axis_). 
2. Email us your address and we will send you a replacement along with a return label.

For the 1<sup>st</sup> option, we will gladly offer you an RNA# that can be used in the future should you decide to go 4 axis.

Please email mike@spark-concepts.com

Spark-Concepts, 2021-04-12:
> Hmmm, I ordered the queen bee mechanical kit and as far as I can tell UPS is still waiting for it to hit stateside.

Unfortunately the USA import process can take a while, especially lately. I'd suggest you reach out to Bulkman to get the status on your order.

philshel1, 2021-04-12:
> > Hmmm, I ordered the queen bee mechanical kit and as far as I can tell UPS is still waiting for it to hit stateside.
> 
> Unfortunately the USA import process can take a while, especially lately. I'd suggest you reach out to Bulkman to get the status on your order.

Yes, I apologize for semi-hijacking this thread.  I just was a bit taken aback by the OP's time line.

WWKSeric, 2021-04-12:
> mike@spark-concepts.com

I have emailed you, thank you for helping

closed 2021-04-21 by Spark-Concepts

IlValeSocial, 2022-01-15:
Hi,

I have a similar issue, if I try to press limit switch of Z axis doesn't work, If I try to connect it to xpro board, led don't turn on

If I try to test pressing X limit switch and send "?" command I receive message: 
<Idle|MPos:135.955,-12.500,24.410|F5:0,0|Pn:XZ|WCO:0.000,0.000,0.000>

IF i try with Z limit switch I display this:
<Idle|MPos:135.955,-12.500,24.410|F5:0,0|WCO:0.000,0.000,0.000>

I hope is PIN problem, How can I solve?

Thanks a lot for any help