Spark-Concepts/xPro-V5 #21: Updating firmware

Greylingad, 2021-04-16:
When Updating firmware to CNC_xPRO_V5_XYYZ_485_NC.bin, upon reboot, the unit asks for user and password authentication and asks whether the correct firmware was selected, this has been double checked and is correct. Any idea what the username and password would be/ whether Authentication is selected by default on this build?

Plottmania, 2021-04-16:
same problem here....CNC_xPRO_V5_XYYZ_PWM_NC.bin

Mboss32, 2021-04-16:
Hi folks, 

The CNC_xPRO_xxxx_xxx_**NC**.bin firmware requires a normally closed door/estop switch to be attached at the time of boot or it will lock out like you are seeing.  

Make sure you have an NC switch attached to the door input on the board, and that the switch is closed, then restart the controller.

Plottmania, 2021-04-16:
Nice.....works.Gesendet mit der mobilen Mail AppAm 16.04.21 um 16:24 schrieb Spark Concepts
                
            
            
                
                    Von: "Spark Concepts" ***@***.***>Datum: 16. April 2021An: "Spark-Concepts/xPro-V5" ***@***.***>Cc: "Comment" ***@***.***>,"Plottmania" ***@***.***>Betreff: Re: [Spark-Concepts/xPro-V5] Updating firmware (#21)
                
                
Hi folks,
The CNC_xPRO_xxxx_xxx_NC.bin firmware requires a normally closed door/estop switch to be attached at the time of boot or it will lock out like you are seeing.
Make sure you have an NC switch attached to the door input on the board, and that the switch is closed, then restart the controller.

[
{
"target": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821214063",
"url": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821214063",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Plottmania, 2021-04-16:
Hallo,
very easy...thx.
[image]
[image]
[image]

I go and check my laser now : )

Mboss32, 2021-04-16:
Good to hear!  

Make sure you follow the wiring instruction here (https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#spindletypelaser) depending on if you have a laser with Enable or not.

Plottmania, 2021-04-16:
I am using a SainSmart Laser 3wire....and Lightburn.Thx for your fast Replay and sorry for me Bad englisch.Retardation StephanGesendet mit der mobilen Mail AppAm 16.04.21 um 17:03 schrieb Spark Concepts
                
            
            
                
                    Von: "Spark Concepts" ***@***.***>Datum: 16. April 2021An: "Spark-Concepts/xPro-V5" ***@***.***>Cc: "Plottmania" ***@***.***>,"Comment" ***@***.***>Betreff: Re: [Spark-Concepts/xPro-V5] Updating firmware (#21)
                
                
Good to hear!
Make sure you follow the wiring instruction here (https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#spindletypelaser) depending on if you have a laser with Enable or not.

[
{
"target": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821240003",
"url": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821240003",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Mboss32, 2021-04-16:
No worries, it probably does not have an enable - when in doubt it is best to use this wiring: 
https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#wiring-lasers-without-enable-signals

Plottmania, 2021-04-16:
I think this is Not normal...No Connection to the webui and no connect via usb to lightburn or openbuildcontrolRegards StephanGesendet mit der mobilen Mail AppAm 16.04.21 um 17:13 schrieb Spark Concepts
                
            
            
                
                    Von: "Spark Concepts" ***@***.***>Datum: 16. April 2021An: "Spark-Concepts/xPro-V5" ***@***.***>Cc: "Comment" ***@***.***>,"Plottmania" ***@***.***>Betreff: Re: [Spark-Concepts/xPro-V5] Updating firmware (#21)
                
                
No worries, it probably does not have an enable - when in doubt it is best to use this wiring:
https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#wiring-lasers-without-enable-signals

[
{
"target": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821246894",
"url": "https://github.com/Spark-Concepts/xPro-V5/issues/21#issuecomment-821246894",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

Mboss32, 2021-04-16:
Can you give a few more details, I'm not quite sure what's happening or the problem.

Plottmania, 2021-04-17:
Hello,
please excuse my bad english.
I have put some information and pictures online on my website.
It is in German but should be easy to translate.
I hope not to have destroyed the controller because I connected something wrong.
Many thanks to the support, really great and very fast.

Sincerely, Stephan

https://www.signtech.online/meine-controller/xpro-v5/firmware/

Spark-Concepts, 2021-04-17:
Plottmania,

Please try connecting to your xPro-V5 via [CNCjS](https://github.com/Spark-Concepts/xPro-V5/wiki/USB_guide#cncjs) or any ```Serial Terminal``` program. Refresh, select the applicable serial COM Port and press connect.  Once connected, press the reset button on the side of the xProV5 and please post a screenshot or paste the contents of the console window.
 
[image]

**MAXIMIZE** the console window and copy-paste the contents (it should look something like this) 
[image]

Plottmania, 2021-04-18:
Hello,
I first connected the three cables and then as you described.
Under LightBurn I also hear how the relay switches but no reaction.
It is no longer possible to connect the controller.
As soon as I connect USB, LightBurn shows "Laser Ready" but waits for connection.
No matter if I press Restet, restart the port ... all without results.
OpenBiuldsControll shows me "wrong firmware" port closed.
I have some software here to send something to an Arduino, but to no avail.
The WebbUI cannot be started, I do not have a controller in the WiFi or the controller cannot be found.
The blue LED only flashes when I try to access the controller via USB.
Thank you for all the attempts to help but I think I killed the xPro V5 while trying to connect the laser.
Now it lies here dismantled and I think I have to buy a new one ... it's a nice part.
What LightBurn reports in between can be seen on the screen.
Anyway, I buy a new one and a laser from OptLasers.
I tinker the SainSmart Laser again on my WoodPecker 3.4 or DLC 2.0 and it is good.
Until the attempt, the update with the door opener worked flawlessly.
So my fault ..... and something like that happens when you tinker.

LightBurn showed the following in between:
G0, G0, G0, G0 ...... without being able to stop.
This as soon as I have connected to USB, even after RESET.

Many thanks for your help

[image]
[image]

Spark-Concepts, 2021-04-18:
Plottmania,

I'm sorry your having difficulty. We have some options for you; please send an email to mike@spark-concepts.com

Slixxor, 2021-05-02:
This is a common issue when uploading from earlier versions of firmware. You must have the 4 matching BIN files executed by esptool in the right sequence.. Simply having 1 bin file is not good enough. It can cause unpredictable results.

I might add an easy firmware update tool to our Commander software like I've done with our Nighthawk Controller for the xPRO v5.

The newest version I'm about to release supports the xPRO V5s extended features like SD file job running etc..

At present we do firmware updates very easily:
[image]

[image]

[image]

Mboss32, 2021-05-03:
Hi Slixxor, that is correct for a bare ESP module will need to instal the bin, an appropriate bootloader, etc.  The xPROs come with all that installed and the pre-compiled bin file is made to be uploaded via the WebUI following the official directions published [here](https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading).  

The issue initially experienced was due to the NC door switch not being installed at boot.  The latter issue was a result of some damaged hardware during disassembly.

Slixxor, 2021-05-03:
Thanks for clarifying. 

I was always curious, what happens if the user sets up station mode and accidently enters the wrong password for their station router?

How can you recover that event easily?

lookhope, 2021-05-10:
after the firmware update.  
[image]
what's the password? how to fix it?

DanVass1979, 2021-08-06:
Same issue but never got around it this has destroyed my cnc it's just a giant brick cause I apparently have the wrong firmware and spark concepts have found a way to make sure yiu can never upload any firmware to there unit but preventing yiu from ever connecting again

rhdhybrid, 2021-08-06:
So the NC firmware looks for a switch before it allows you to do anything, install any switch or bridge the connection with wire to act as a switch, then upload the NO firmware.

DanVass1979, 2021-08-06:
Good idea , tried it didn't work

Slixxor, 2021-08-06:
> Good idea , tried it didn't work

If interested I've made a standalone version of our firmware updater. It uses USB for reset and is not internally managed by the esp32 itself.

If interested let me know. I'll look at adding it into our Commander software which already supports the extended features of the xPRO.

DanVass1979, 2021-08-06:
So it can cha get the firmware on the xpro v 5 without needing to connect to it ? Cause there Re a lot of people having the same issue as me ?

Slixxor, 2021-08-07:
> So it can cha get the firmware on the xpro v 5 without needing to connect to it ? Cause there Re a lot of people having the same issue as me ?

It will flash the firmware via USB. This will work regardless of state. Likely it is how spark it sending out their units.

It's safe to do as it uses the internal bootloader. Even if it fails you can repeat the process.

Just connect usb and fire it up.

Spark-Concepts, 2021-08-11:
Quick comment before closing out this one - new firmware is available for upload.  This moves the safety lockout from the login screen to a simple DOOR:1 error in the WebUI giving a bit more detailed feedback on what needs to be fixed.  Note that this is still a safety lock and the controller will not proceed until the Door/Estop condition is met, and a reset triggered.  The new firmware adds a few nice features and fixes some minor bugs as well :) 

For those following along, additional troubleshooting with DanVass was done via email, and it turned out there was a hardware error on his unit (bad FET or a short on the pcb) that made the NC door condition unreachable.

closed 2021-08-11 by Spark-Concepts