Spark-Concepts/xPro-V5 #23: PWM 5V on Constant - TTL Laser Control

rhdhybrid, 2021-04-19:
Hello, I hope I'm in the right place for some help, it seems my Toolhead PWM(0-5V) is at 5V constant, the 0-10V seems to be functioning properly. Limits are working properly. However there is a problem, when issuing M3 S3000, the spindle starts then stops with a hard limit alarm. Unplug limits and it works fine, removed spindle wiring from drag chains, routed away from all control cabling, plug in limits no go, starts then stops. unplug and works perfect. So I guess I have a few issues, the PWM has 5 Volta on it constant through it all though. which seems like a board problem. The Toolhead control is set to EN/PWM. Firmware is XYYZ/NC/PWM. Machine is a OX CNC Belt. Config from Makerhardware.net. Tweaked Axis directions but works perfect other than that. Seems like the Gcode does not take the M4 command in laser mode. Thats my list, I have been at it for days and days. 

I'm running windows, cncj's all the latest version. Installed this morning. I didn't get anywhere with my mac, seems that VCP driver and CNCj's does not work, bugs. That I can't figure out.

Thank you, 
Ken

Spark-Concepts, 2021-04-19:
Ken,

The issue you're having with the constant 5V on the 0-5V PWM, is something we've just recently been made aware of (this problem only affects a relatively small number of xPro's). 

Unfortunately our PCB assembler incorrectly placed a pull-up resistor on the 0-5V PWM output... We will be posting more details on a resolution shortly.... 

The fix on this is relatively an easy one. Unfortunately this requires disassembling your xProV5.

**NOTE:** be sure to remove the SD card

rhdhybrid, 2021-04-19:
Thank you very much for the answer, I tried every wire configuration in the manual. I was going insane. Lol. Please keep me informed, I have work lined up for the machine. On the limit hard alarm, I seen in another closed issue that it can be turned off. Will this affect homing? Thank you again. I have my spindle cable pulled from the cabling so I don’t know where it could be getting noise. But I’ll go pull some screen shots of the console data in cncj’s. If that will help.

Spark-Concepts, 2021-04-19:
[image]
[image]

Re-assembly:
1. Insert USB board into the bottom of the case.   This is a tight fit.
   a.	Insert board at an angle with SW3 headed for the circular hole marked RESET on the case
[image]
   b.	Slide SW3 into the RESET hole.  
[image]
   c. Grab the 4 Green connectors in the board, gently pull the board away from the USB connector and align the USB plug and the SW1 switch lever with the board slots marker “USB” and “EN/PWM- - RS485 A/B”.
[image]
   d.	At this point the board will slide down into the case.

rhdhybrid, 2021-04-19:
Thank you, I will remove the resistor tomorrow, I just went out and turned off the hard limits and everything seems to work normally now with the spindle. Thanks so much for the prompt comments. Thanks so much. Saved me.

Spark-Concepts, 2021-04-19:
To answer your initial question, which you've already verified; Yes, homing still works with hard limits disabled

rhdhybrid, 2021-04-20:
Just wanted to let you know the fix went as planned, the pwm pin provides the desired voltages as commanded with different spindle speeds. however, I think the damage is done to the laser controller, it turns on to full power, but after trying a program in light burn it started with the beam on full power, then it fizzled out, and now nothing. Anyway, thats my luck, the xpro is working as it should now. :)

Spark-Concepts, 2021-04-20:
What model laser are you using? 

Double check the modulation frequency required for the laser _ e.g. "Bulkman 15W Laser Modulation frequency: ttl can be modulated, 0v-off,5 open, switching frequency <9 kHz"_ 

```$Spindle/PWM/Frequency (Spindle PWM Freq)```  _This sets the frequency of the PWM signal_
The default PWM modulation frequency setting on the xPro-V5 is ```$Spindle/PWM/Frequency 5000```

Also double check the $30 setting - go to **Edit** > **Device Settings** in LightBurn, **S-Value Max setting** should match the **$30** setting on the xPro (typically ```$30=1000```)

closed 2021-04-21 by Spark-Concepts

rhdhybrid, 2021-04-21:
Thank you very much for the info, I added the following commands to the start gcode in lightburn setting,  $32=1
$30=1000
$Spindle/PWM/Frequency=9

The end gcode is
$32=1
$30=24000
$Spindle/PWM/Frequency=5000

Returning the machine to the cnc configuration, I hope. 

I believe my problem is with VR-1, the other variable resister clicks when turned clockwise, vR-1 does not. I believe this is the power control resistor. Correct me if I'm wrong. No documentation came with the laser, it is from bulkman 3D.

[image]

Thanks so much again.

Spark-Concepts, 2021-04-22:
While we do not fabricate or sell lasers - however, I am familiar with them.

To check the function of the laser itself, it's important to note that lasers require constant current not a set voltage.

Inside the laser housing, the laser diode itself typically consist of a three terminal device: ground, power (to the laser diode), and a photodiode (output for feedback). Since the Laser driver PCB above only has 2 wires to the LD; I would assume the device you have has either a built-in controller or is running in straight open-loop (uncontrolled and have no idea what your actual laser power output level is). This can shorten the life of your laser. Laser diodes are designed to be operated in closed-loop mode where the power supply senses the output level of the laser through that photodiode and adjusts its output current on the fly to keep the power output level where it is designed to be.

To test the Laser itself, if you keep the current constant and in a safe range for the laser diode, you won't hurt anything, If you have an adjustable power supply with a current limit adjustment, set the voltage at 3-4V and current limit adjustment to around 10mA. Slowly raise the current limit until the laser begins to lase (even at low current levels the laser should emit some light - just don't release the magic smoke.

Okay so the picture you sent me appears to be an open-loop power supply with two outputs and two inputs.
1. Fan Power (out)
2. Laser Diode Power fixed current (out) - with a current adjustment trimmer (VR1 or VR2?)
3. PWM 0-5V (input)
4. 12V or 24V? (input)     

Let's say the laser diode checks out, I would look for and at the following components:
1. I would hope/assume there should be an opto-isolator close to or where the 0-5V PWM plugs in.
2. From there I would look at the power transistors (circled below)
3. Can you read the markings from the SO8 package
[image]
4. I would be surprised if you found the trimmer to be at fault - (to test it, you'll want to remove it from the PCB, then ohm it out with a multimeter) _there should be a 3 number marking on the trimmer first two digits followed by a digit representing how many zeros proceed; i.e. 103 =  10,000 Ohms_

Also, would you mind posting a picture of the underside...
[image]

rhdhybrid, 2021-04-22:
Thanks again for helping out. I understand trouble shooting someone else's product is typical. I really appreciate your knowledge. This is my first laser and second cnc machine. I was more excited for the laser then carving. Lol. So I had a look around. And took some pictures that hopefully will provide some info. I have checked the output on the LD wires and it's 12 volts. Changing the VR's does nothing, plugging in the ttl wires also doesn't change the voltage output. 

Seems way to high as you stated to test at 3-4 volts. Nothing seems burned up though, Drivers look fine, nothing stands out.

[image]

[image]

[image]

I hope that's provides abit more info, on the AliExpress add it says that it power to full so I would assume that's 5ish volts on the meter. 

Thanks again, I ran a cnc program last night and the controller worked great.

rhdhybrid, 2021-06-01:
Hello again, sorry to revive this old topic, but I recieved my replacement laser from bulk man 3D. So all is well. I still have a problem. I changed the  Spindle PWM frequency to 9000 as above. Enabled laser mode. But when I turn on the laser power it's again at full power. I probed the wires and did some experimenting, the pwm 5v pins shows 0.31volts at rest with laser power off. Turn laser power on it shows 2.31ish volts. Disconnect ttl wires at laser control box and the voltages drops to 0 so it's not a short and has to come from the laser controller. So I had light burn open so I manually fired the laser at 0.25% and the laser shuts off. XPros relay is latched. Turn off the laser in lightburn and it goes to full power. Relays disengages in the Xpro. When I turn the laser off the voltage returns to 0.31v on the PWM and the relay NO side has 0v. I think the problem lies in the rest voltage, with it wires through the relay it opens the circuit and the laser goes to 100% as Bulkman 3d says should happen. I just went out to double check the voltages, when light burn commands a setting say 10% to he pwm voltage is 0.10 which switches the laser off. I'll do abit more testing today when I get off. But maybe there's something I'm missing.

Mboss32, 2021-06-01:
Good testing, we have one of the Bulkman lasers headed our way for some testing here.  In the meantime, would you mind checking a few voltages?  (please note that "No" is an acceptable answer and we can measure them directly in the near future and provide more troubleshooting when the unit arrives).  

If you disconnect the xPRO and the Laser control box -  power the laser control box and measure the TTL input pin on the laser control box.  Is there a voltage present?  Trying to decide if the laser control module has a floating input voltage when not actively pulled down.

rhdhybrid, 2021-06-01:
Ok so I did abit more testing. I have full function of the laser when I command a power percentage it outputs voltages that ascend and descend as expected. The laser will be off if commanded like this.
[image]

The output voltage is 0.07ish volts on the PWM pin.

[image]
When commanded off, with M5 through lightburn it turns the PWM pin off and it has the 2.31 volts which comes from the laser controller. 

Laser mode is on, PWM Frequency is set to 9000. So not really sure where to go. I think the PWM pin needs to remain at 0.07 volts when off in order for the laser to be in the off state. 

I have tried both directly into the tool head connector and through the relay as per the documentation with the ttl wiring. Neither way works.

rhdhybrid, 2021-06-01:
Sorry I was writing that last post and seen this. When the laser is disconnected from the xpro it floats at 2.33 volts. And if you need anything looked at please ask, I don't mind at all, I want this to work more then anyone.

Spark-Concepts, 2021-06-02:
Based off your test results it appears that the Bulkman laser may have a 2.3V low impedance offset when open. Currently the xPro-V5 has a fairly weak pull-down and may require a bit more assistance - in otherwords, you may need to add a resistor in parallel with the laser ourput signal  (~1K to 4.7K Ohms between PWM to Ground. ( I'll plan to add an illustration soon)

There's no harm in you adding a resistor yourself. Regardless we will perform thorough testing using the Bulkman laser (once received), and report our findings.

rhdhybrid, 2021-06-02:
Thank you for the response, I went out this morning and found a 3.3 Kohm resistor, direct with the ttl wires to the gnd and pwm in the tool head connector. With the 3.3 kohm resister between them. The laser worked better, it was still emitting some light but not enough to burn the waste board. I probed the output at rest and it is 0.10 volts. I ran a small program with light burn and it functioned as expected. I think we need to fine tune the resistor but it seems to work. Not sure if this affects the voltage delivered to the ttl wires because the engraving was at 100% but the burn was light. I'll run a few more tests hopefully this afternoon when I can get back to the shop.

rhdhybrid, 2021-07-10:
Just checking in to see if you guys recieved your bulkman 3D laser yet for testing. Mine is working as expected but I feel like I'm not quite getting full power. Don't get me wrong, it works, but I have to turn my federated way down to achieve a black burn, previous without the pull down resistor it seemed to have more power. Anyway, more so just checking back to see if there is a permanent tested fix. 

Thanks again for your support. I couldn't be happier.

blakamin, 2021-07-13:
I'm having this issue with the exact same laser (came from makerstore.com.au)

Have there been any updates?

Spark-Concepts, 2021-07-13:
Mike

Uuhg

Bill

Mboss32, 2021-07-13:
Kick back to en relay wiring?

Spark-Concepts, 2021-07-13:
I'll reply later today

rhdhybrid, 2021-07-14:
I excellent guys, can't wait for your response.

Spark-Concepts, 2021-07-15:
I think we have a firmware update that will fix this - note this is untested with the specific BM3D laser.    If one of you would be willing to test it out and report, it would be appreciated.  

Requires NO Estop, or no Estop.  This will make it so enable pin is always on, allowing the PWM pin to sink the voltage from the BM3D laser when the controller is powered on.  So, best practice would be to start the controller, then the laser, likely starting the two at the same time will be OK - just don't start the laser first   

[CNCxPROv5_XYYZ_BM3D_NO.zip](https://github.com/Spark-Concepts/xPro-V5/files/6825718/CNCxPROv5_XYYZ_BM3D_NO.zip)

blakamin, 2021-07-15:
Extracted file is 1kb... Is that correct?

Mboss32, 2021-07-15:
it was indeed not.. let's try that again, this one is the compiled bin, uploads through the WebUI

[CNCxPROv5_XYYZ_BM3D_NO - correct.zip](https://github.com/Spark-Concepts/xPro-V5/files/6826338/CNCxPROv5_XYYZ_BM3D_NO.-.correct.zip)

blakamin, 2021-07-15:
No movement on any axis.
Makes a quick move on connection then...
[Lightburn Console.txt](https://github.com/Spark-Concepts/xPro-V5/files/6826405/Lightburn.Console.txt)

Lightburn showing the below txt

Spark-Concepts, 2021-07-15:
Send $RST=*, then power cycle the controller - this will reset all setting to the firmware default. 

Do you have a door switch?

blakamin, 2021-07-15:
I do but it's NC so I disconnected it

blakamin, 2021-07-15:
That''s fixed it, Thanks... Now to try the laser (sorry if it takes awhile to respond, I'm at work)

blakamin, 2021-07-15:
Works! Thank you!

Spark-Concepts, 2021-07-16:
Well, getting closer - is there a specific error you are seeing? Can you provide a little more detail on what is going on?

blakamin, 2021-07-17:
I'll attach a pic of the error

[image]

rhdhybrid, 2021-07-19:
I think your error may be that lightburn is using M4 command, but the firmware doesn't recognize this, you can switch lightburn to use the M3 command instead. Mine did the same thing but once I switched it to the M3 command I have been working. I'm going to try out this firmware for myself today and will report back.

blakamin, 2021-07-19:
> I think your error may be that lightburn is using M4 command, but the firmware doesn't recognize this, you can switch lightburn to use the M3 command instead. Mine did the same thing but once I [...]

Won't that keep the laser on all the time though?

rhdhybrid, 2021-07-19:
No it shouldn't I'll head out know and give the firmware a go. But to use M3, you need to change the setting in lightburn to use this

[image]

It is the profile it uses to make the grbl output code.

blakamin, 2021-07-19:
My laser power is extremely low with GRBL-M3

rhdhybrid, 2021-07-19:
ok so I uploaded the new firmware through the web app, the update went fine, all my setting ported over. Tested with cncj's and homing and movement are normal. Lightburn loaded as normal, homed on startup. I ran couple of tests, seemed to work as intended, tried acouple of different feedrates with differing results but overall I think it has the rated power it should now, noticed some scorching on the line test which didn't happen with the same feedrates with the pull down resistor, I'll need to do some burn tests for couple of different materials but it doesn't stay on or is at lower power when it rest, so far I would call this a win. 👍 Great Job guys on the support. Got me wondering if there is a power difference using the M3 Vs M4 for laser control.

[image]

Spark-Concepts, 2021-07-20:
Thanks for testing out!  We have found the m4 command issue and are currently putting all of this into a firmware update.  Should be officially posted in the next few days.  We greatly appreciate the testing and reporting back :)

blakamin, 2021-07-20:
Awesome news

rhdhybrid, 2021-07-20:
That's awesome guys, thanks for the support. I look forward to testing out the new updated firmware.

Spark-Concepts, 2021-07-23:
Morning folks, 

New firmware is ready if you are willing to test it out.  Zip attached is a BM3D specific version in XYYZ and NO or NC door switch options.  The new firmware will let you flip between different spindle types with out needing to recompile.

With the BM3D version of the firmware, choosing a PWM or Laser spindle type will driver the enable pin to always be ON, allowing the PWM output to sink the voltage offset from the Laser - i.e. no more unpowered firing (hopefully).  That said, make sure you turn on the xPRO and then the Laser, with the offset on the laser anytime it is powered up without a signal on the pwm, it will fire... please be careful.  

To use this with the BM3D laser, load either the CNC_xPRO_V5_XYYZ_BM3D_NC or CNC_xPRO_V5_XYYZ_BM3D_NO .bin file, depending on your door switch.  Set the spindle type to laser and turn on laser mode:

$Spindle/Type=Laser 
$Gcode/LaserMode=on

You should now be able to make M3 and M4 cuts using lightburn.  
Thanks and let us know how the test goes! 
[CNC_xPRO_V5 20210708.zip](https://github.com/Spark-Concepts/xPro-V5/files/6869329/CNC_xPRO_V5.20210708.zip)

rhdhybrid, 2021-07-23:
Thanks so much, I loaded the NC bin file and it loaded fine, all settings ported over, lightburn homing is correct. The laser doesn't power on as it should. I will do more testing on Sunday when I get back to the shop. Thanks for all the hard work.

blakamin, 2021-07-26:
Works great, Thanks guys!!