Spark-Concepts/xPro-V5 #25: Controller not working after limit switch experiment

Jncls, 2021-05-16:
In the limit switch part of the basic hookup section there is the following warning: "Warning: Incorrect wiring can short V+ to GND causing damage to your controller: Double check wiring before powering on"

I'm afraid that exactly this happened to my xPro V5 when experimenting with the limit switches.

When I now turn the controller on only the power on indicator lights up. I can't see the controller's wifi network nor are the steppers energized. Do you have any idea which component could have been damaged and needs to be replaced?

Spark-Concepts, 2021-05-16:
Hello,

First off I'm sorry you're having difficulty with your xPro.

I'm sorry you're having difficulty - unfortunately, on the earlier batches of xPro's one of the chips (U5B) on the underside of the top circuit board tends to release the magic smoke if/when 5V power is in a short circuit condition (despite the [XL1509](http://www.xlsemi.com/datasheet/XL1509%20datasheet.pdf)'s datasheet "Built in current limit function"). Fortunately, albeit unfortunately this issue has since been resolved.

You have a few options:
1. If you're in the US: contact mike@spark-concepts.com or bill@spark-concepts.com for repair options

2. Replace the chip yourself, which requires some SMD soldering skills - OEM p/n: XL1509-5.0E1; however, p/n: [AP1509-50SG-13](https://www.digikey.com/en/products/detail/diodes-incorporated/AP1509-50SG-13/1301328) actually current limits without destroying itself.

3. Quick DIY temporary (or permanent) solution:, Remove 'U5B' (XL1509 chip) and power 5V via USB (USB-C cord plugged into a USB power brick). Follow instructions below.

## Remove U5B
- To remove U5B without a hot air rework tool, I would recommend doing so by (very carefully) using a sharp E-XACTO knife or sharp razor blade.  Placing the blade flush along the sides of the IC body (see below). Applying light to moderate pressure while pulling the knife toward you. (note: you will need to make several passes on both sides... once you've cut through the legs on U5B, it should easily lift or slide right out).

- I recommend you measure resistance as shown below, if you're getting a resistance value above 90-100 Ohms with U5B removed you should be able to safely power your xPro using a USB-C cable and USB power brick. (note: UL certified USB power adaptors have built in Short circuit/Over current protection)

[image]

Jncls, 2021-05-17:
Thank you very much for the detailed explanation. 

On your picture the U5B component looks damaged but in my case I can't see anything like this (image attached). Is there any chance to find out if it's damaged or not before cutting it out - e.g. by measuring some resistance values between certain pins? As I'm not in the US (Germany) I need to repair it by myself.

[image]

Spark-Concepts, 2021-05-17:
-  I've also seen U5B fail intact; U5B typically fails in a low resistance/short circuit condition...
 1.  Measure the resistance across D4B(+5V to GND): if the resistance should be greater than 50 Ohms.
 2.  Connect 12-24V to the top board and switch the board ON: very carefully measure the voltage across D4B (Voltage should read 4.9-5V).

If one or both conditions don't meet the criteria mentioned above, U5B is likely the culprit.

Jncls, 2021-05-17:
I've marked the points on the picture below where I measured resistance and voltage. Resistance is about 7 Ohms, voltage is 0V. Are these the correct spots where I should measure >50 Ohms and 5V? If yes, I will continue by removing the U5B and try to power it up over USB.

[image]

Spark-Concepts, 2021-05-17:
The points you marked work too - so yes, proceed in removing U5B

Jncls, 2021-05-17:
Yay, it's breathing again!

I think that I've found a shop in Germany where I can get the replacement component, but thank you very much for the offer.

After a quick data sheet check I'm a bit confused as it says "Operating Voltage up to 22V". Is it really the same and correct component?

https://www.diodes.com/assets/Datasheets/AP1509.pdf

Spark-Concepts, 2021-05-17:
Yes, you are correct regarding the Diodes  (p/n AP1509), hence why I mentioned "p/n: AP1509-50SG-13 will actually current limit without destroying itself (although it's more sensitive to input voltages above 24V)"... also note, subtract 500-700mV from the input 24, and that's what U5B See's  (~23.3V).

The AP1509 Datasheet specifies absolute max input of 24V... In which your right on that cusp. I can tell you that I've used the AP1509 when prototyping with 24V input applied and haven't had any issues. 

From load testing the AP1509 at the rated 22V input, the output (5V) remained steady with an applied 2A load; however, with 24V applied, the output load was limited to ~1.5A before it begins thermally limit the output voltage. Also note: while in use, the typical current draw on the 5V is approximately a few hundred mA - if you're pulling over 1amp (1000mA) on the 5V output, something is wrong.

Basically, if you go with the AP1509 understand the long term affects are unknown and the current draw on the 5V rail will be limited to 1.5A vs the 2A, also make sure the input voltage doesn't exceed 24V.

That said, the XL1509 works great and is rated at a 40V for it's absolute maximum input ...as long as the 5V and Gnd aren't directly touching in a short circuit configuration.

closed 2021-05-25 by Spark-Concepts