Spark-Concepts/xPro-V5 #26: X and Y motors getting hot

rhdhybrid, 2021-05-17:
Hello, I'm winding what the operating temperature should be for my ox cnc. The X and Y motor are getting hot, to hot to keep my fingers on them after about a hour of carving. I stopped the program as I was scared they would cook. Is there a way to turn down the current going to the motors? The motors are the supplied nema 23's from Bulkman 3D. 

Thanks again for your hard work, my controller is producing some good work. Cncj's doesn't update when it's working but it does work good.

rhdhybrid, 2021-05-17:
Found a closed post with my solution. Going to give it a try this afternoon.

rhdhybrid, 2021-05-17:
I think I may have done something wrong. I changed all the $X through to $C/Current/Hold to 1.1 and the $X through to $C/Current/Run to 1.8. Now there is no Y/A movement, Y axis works but A is holding. And the Z does not work. 

Could the $B/current/run and hold values be wrong now. Also the $144 is now set to 0.250. I have attached some photos of the console. If someone can let me know what the current run and hold values should be I think this is my new problem. [image]

[image]

I hope someone can help me out, I have a piece to do and was hoping this would keep the motors cool so I could complete it.

Mboss32, 2021-05-17:
I think you have entered the command incorrectly to change the settings.  The command should have an "=" to set the value.  Example: "$X/Current/Hold = 1.5"

Try to reset your values back to default by sending the command "$RST=*" and let's see if that gets motion back to the machine.

rhdhybrid, 2021-05-17:
Thank you for the response, I performed the rst function, then entered all the configuration data again. Got everything homing and travelling in the right direction with the below current settings but the motors are still really hot on the x and z axis, the program was a parallel carve so it was heavy on these axis. Maybe I'm off somewhere but the motors were starting to stutter after ten minutes of use. 

Everything seems to turn fine now that the power is off so I think everything is still ok. Just needs a cool down. 

Maybe you have some setting suggestions. 
[image]

[image]

Mboss32, 2021-05-18:
What is the current rating on your motors? The current set with $X/current/run or hold is an RMS current, not peak. Examples for how to set these properly are here:
[https://github.com/Spark-Concepts/xPro-V5/wiki/Changing-settings#xcurrentrun-or-ccurrentrun-140-thru-145--xyzabc--stepper-run-current-extended-settings](https://github.com/Spark-Concepts/xPro-V5/wiki/Changing-settings#xcurrentrun-or-ccurrentrun-140-thru-145--xyzabc--stepper-run-current-extended-settings)

rhdhybrid, 2021-05-18:
Thanks Moss for the comment, please find the attached motor specs taken from Bulkman 3D's add. It says they are 2.8A. I would think my 1.6A setting should be ok. I would think this machine and controller should be able to run for hours and hours as my cnc is 1500x1500 it should be able to conceivably carve a piece the size of the work area. Correct me if I'm wrong but that why I bought all of it. I have had a nice few problems so far, burned up my laser, xpro needed repairs to fix that, now I have a motor burned up. 
Sorry this is breaking my heart, I need to order another motor from someone in Canada today, hopefully have it fixed next week. But please let me know if I can test anything to see if there is another problem with the xpro, maybe there is to much voltage to the motors? 

[image]

rhdhybrid, 2021-05-18:
Even if I get replacements sent I'm going to order another set of motor, which of these three are recommended. 

[image]

Thank you guys for all your help, your responses and customer service are excellent.

Mboss32, 2021-05-18:
You are correct in your assessment, these should run for hours on end with no issues.  The key thing is getting the current settings correct for your motors, the 1.6A (run) and 1.1A (hold) setting should be good for the motors shown.  

A quick test, loosen/remove your flex coupler from the lead screw so that the motor can turn freely.  Loosen/remove all of them and try to move the motors again.  Do the motors turn again?  Can you move the gantries by hand the entire length, and tight spots, etc? 

For replacement motors, all of the listed motors will work, I would probably go with the 267oz-in motors personally.  (**Greatly oversimplifying things** - generally, with higher oz-in ratings, you will see a lower top speed before stalls occur)

rhdhybrid, 2021-05-18:
Thanks Mboss, again your to the rescue, the gantry moves no issue, I unplugged my motors and moved the gantry around and it moves no issue. Doesn't bind or anything. Motors also turn by hand. But when operated they are stuttering and sound terrible. Z and X are rough, the Y axis sounds abit loud. The A motor seems ok. I'm going to order 4 motors right now and get them shipped but I can do some testing this afternoon when I get back to the shop.

rhdhybrid, 2021-05-18:
Motors are ordered. So hopefully we can get to the bottom of this before they arrive. :)

Mboss32, 2021-05-18:
I would recommend checking all of your wiring - at the motor pigtail and the controller.  A loose connection or swapped coil wires can cause the motors to run like that and behave poorly.  

Could you also post your settings?  (send $S to the controller) 

It looks like the X motor was unresponsive for a few clicks, could you repeat and see if you get a warning LED on the side of the case above the X label?

Mboss32, 2021-05-19:
Ok, looks like a few of the settings are not correct.  I *think* this may be the issue as they determine how the motors get driven.  

Please try the following changes and check the motors again:
$0=4
$2=0
$4=0

rhdhybrid, 2021-05-19:
Ok, I'll give it a go when I get back in the shop this afternoon. :) thanks again for the reply.

rhdhybrid, 2021-05-20:
I still haven't had a chance to check out the settings but this is where I got all my configuration settings, I think they may be wrong in this guide.

http://makerhardware.net/wiki/doku.php?id=electronics:xpro_v5_motion_control_system

[Ox Belt 1500x1500 Standard Torque.txt](https://github.com/Spark-Concepts/xPro-V5/files/6518372/Ox.Belt.1500x1500.Standard.Torque.txt)

Halfway down the guide there is a link and discretion on how to set up a macro for the config. Perhaps they are wrong here.

Thanks again, it/'' be monday evening before I can try the settings now, and the new motors will be here next friday. Thank you again for your support.

rhdhybrid, 2021-05-24:
Hello again, I hope you all had a good weekend. :) I just went out and made the changes above, I ran the axis test's and the z and x are definitely shot. I ran the y/a axis back and forth for abit, let it stand then ran it back and forth again for abit. About 20 minutes or so. Checked the motor temps with a digital temp gun, 130-136 F for the y/a axis and 100 F for the z and x axis which were standing still for the entire time besides a quick test to see how they worked. So this seems abit high but correct me if I'm wrong. 

My new motors will be here tomorrow, so I'll swap them and triple check all the wiring.

Thanks again for all your help. :)

rhdhybrid, 2021-05-27:
I installed the new 267 oz motors, everything runs fine, moved it around abit, motor were in the 60 deg range. So I then started the same parallel carve I used the last time, I ran it for about ten minutes, monitoring the motor temps with digital temp gun, then ran in the 80's for the Y axis, the z and x, which are in constant motion for up to 120 deg F. I shut it down. From the google searches I made this is on the upper range for a nema 23 motor. I'll lower the run current again and see if it makes a difference. I verified all the wiring and connections. It seems to function and move properly. 

Is there a way to test the motor outputs to see if everything is outputting properly. 

Thanks again.

Ken

rhdhybrid, 2021-06-01:
This can be closed up, I adjusted the current run down to 0.9 and the current hold to 0.5 and all is well. Motors run warm to touch. But not hot. I'm probably conservative on my numbers but I want this to operate reliably for years to come. Thank you so much for your help.

Mboss32, 2021-06-01:
Thanks Ken!  I would agree that is pretty conservative, we get a little worried around the 150F range.  However, as long as you are not missing steps there is no need to run hot.

closed 2021-06-01 by Mboss32