Spark-Concepts/xPro-V5 #27: Connecting xPro V5 to home wifi

pacman171, 2021-05-17:
Hi

I please need some advice on the settings I should use on the xPro to get it to join my local wifi. I can connect to it via my cell phone. I have changed all the settings I thought I need to get it to connect, but nothing seems to work.

Any ideas, of if someone can screen shot their settings please?

Cheers
Paul

Mboss32, 2021-05-18:
Generally, we recommend using the Access Point just for network security purposes.  If you can accept that risk, or have a local network you can change the behavior of the wifi connection with the below settings.  

[image]

The first two are hopefully obvious.  For the Station IP Mode, choosing a static IP will keep the xPRO from being re-assigned so select a Station Static IP that you know is on your network and free.  Add your gateway address (192.168.1.1 is fairly typical) and add the network mask (255.255.255.0 is also fairly typical).  

Make sure you click the "SET" button for each of these entries before moving on.  

Lastly, change your Radio Mode to Station.  

[image]

Click "Set".  Reboot the controller and connect your computer to your network, open a browser, and type in the assigned static IP.

pacman171, 2021-05-18:
Many thanks for the reply. Is there a way of resetting the network settings? I made changes yesterday before receiving this message and now cannot connect to it at all. I cannot see the CNC_XPRO_V5 network when I switch it on. I am able to connect to it via USB using CNCjs

Mboss32, 2021-05-18:
Send command "$Radio/Mode=AP" through the command terminal in CNCjs and restart the controller to reset the Wifi to Access Point mode

pacman171, 2021-05-18:
Many thanks, I used the $RST=wifi command to restore all wifi settings and this allowed me to connect the old way and then I could set it up as per the instructions you sent.
Many thanks

Mboss32, 2021-05-18:
Great!  good to hear it is all set

closed 2021-05-25 by pacman171