Spark-Concepts/xPro-V5 #29: Power Switch not working?

jiangmf, 2021-06-13:
I'm very new to the world of CNC and wiring up circuits so forgive me if I'm making some kind of newbie mistake.

I have a very simple set up currently, just 4 motors, 24V power and USB connected. The power switch does not seem to be doing anything, even in the OFF position, the 24V indicator light is ON, and I'm able to spin the motors with CNCjs. Everything still works when the switch is in the ON position as well

Is this normal? if so, what do I need to do to get the power switch working?

[image]

Spark-Concepts, 2021-06-15:
Jiangmf,

That's an interesting one... The motors and green LED should not be operational while in O. F. F. mode. Most certainly sounds like a switch issue.

When toggled, does the switch detent feel loose -or- do you get a satisfying click?

jiangmf, 2021-06-17:
No the switch clicks and feels very firm in both positions.

Spark-Concepts, 2021-06-21:
My most sincere apologies  for not replying to your last comment.

If you wouldn't mind please post or email a picture of your current setup - mainly the wiring (motors, limit switches, power supply, etc...)

Post here, or email to bill@spark-concepts.com & mike@spark-concepts.com

Thanks!

Spark-Concepts, 2021-06-24:
Hmm, very odd, that's a first! 

Please contact us at https://www.spark-concepts.com/contact-us-1/ for options

jiangmf, 2021-06-26:
Closing this issue as somehow the switch magically started working as I was putting everything together for a photo. Don't think I've made any changes to any of the connections other than maybe flashing the firmware to CNC_xPRO_V5_XYYZ_485_NC (don't recall if it started working before or after the flash)

closed 2021-06-26 by jiangmf