Spark-Concepts/xPro-V5 #30: Using Yl620-A VFD with RS485?

Emanuelbastos1, 2021-06-13:
Is it possible to use a Yalang YL620-A VFD with RS485?
I believe a new firmware version is needed.
Thanks!

Spark-Concepts, 2021-06-21:
Emanuelbastos1,  

You are correct and we look to include it with the next firmware update (coming soon). We are in the process of validating the stable RS485/ Yalang YL620 VFD Modbus operation.

In the meantime, I would recommend controlling your spindle using the 0-10V output.

Emanuelbastos1, 2021-06-21:
Dear Spark-Concepts,

I'm a little disappointed with the controller. I can't get any voltage from
the 0-10v anolg output.
Even if I type the command in the webui, as example, M3 S12000, when I
measure the voltage output it is 0V. With pwm, the same story.
I just sent an email to return the controller.
can you help me with this?
I would like to not return the controller, just to have it working
properly!

Thank you for your help,
Kind Regards

Emanuel Bastos

rhdhybrid, 2021-06-21:
Don't lose faith, I had a bunch of growing pains with mine as well, but the support here is great. And once I figured out the couple of issues I had, my machine is now working perfect. I found a number of videos on YouTube for configuring my spindle on youtube. If you want pwm you must flash the pwm firmware into the cnc xpro for the 0-10 volts to work properly. Then you must change acouple of settings in the VFD controller to use the 0-10v input. I found this info on youtube. As I run a different vfd I'm not help. But try here.

https://youtu.be/4T_4a2YgCgk

Emanuelbastos1, 2021-06-21:
I belive I did everything correctly. I flashed the firmware to the pwm
version. Even disconnected from the vfd, there is no voltage in the 0-10v
analog output. I send the command to activate the rotation and
nothing happens, 0v!
I will take some pictures and send it to you!

Emanuelbastos1, 2021-06-21:
The video with the problem.
I had all ready to return, so the connection is something for a quick test!
https://we.tl/t-rjGLbyoccD
The video is just 1 or 2 minutes.

Thanks

Spark-Concepts, 2021-06-21:
Emanuel,

I'm more than happy to assist - Please see [https://github.com/Spark-Concepts/xPro-V5/issues/33](https://github.com/Spark-Concepts/xPro-V5/issues/33)

Emanuelbastos1, 2021-06-22:
![image](https://user-images.githubusercontent.com/85850044/122974510-4f0afd80-d38a-11eb-879f-2d7d2f90d464.pngIt 
The switch is set to "EN/PWM"

Spark-Concepts, 2021-06-22:
Emanuel,

It looks like the spindle setting is configured for RS485 VFD... 

That's an image of the startup screen - if you could, please enter `$S` in the command line then copy paste all settings here..
[image]

Emanuelbastos1, 2021-06-23:
That solved half of the problem. I changed the spindle type during my attempt to use the RS485 connection and I thought the firmware update would reset the settings I changed. 

Now I changed the spindle type to PWM and, when I run the test2.nc file, I am getting the same 0 V in the 0-10 V output but the PWM output is working, and I'm getting 2.7 V. 

What else is wrong?

Old settings:
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=H2A
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=100.000
$Y/Acceleration=100.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=770.000
$Y/MaxTravel=790.000
$Z/MaxTravel=90.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off

New settings:

$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=H2A
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=100.000
$Y/Acceleration=100.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=770.000
$Y/MaxTravel=790.000
$Z/MaxTravel=90.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off

Mawg1, 2021-06-25:
Hi, I think what they are saying is you need to change the $Spindle/Type = PWM . 
 Your "new settings" still have it set to H2A.  Hope this helps! 

 Also, I found that using command "$RST=*" after loading new firmware will reset all the settings back to defaults.  Hope this helps!

Emanuelbastos1, 2021-06-25:
@Mawg1 Yes, you are rigth! 
I paste the old settings two times, however I change it to PWM.
I edited the comment to match the correct settings!
Thank you!

Mawg1, 2021-06-25:
@Emanuelbastos1 - well that makes more sense that it was double pasted, but wanted to be sure.  I have been working on getting my YL620 operating and think I have the code correct.  If you want to give it a try, I have attached it here.  It loads through the webui just like the official bins.  

You will need to set these registers on the YL:

 Configuration required for the YL620

    Parameter number        Description                     Value
    -------------------------------------------------------------------------------
    P00.00                  Main frequency                  400.00Hz (match to your spindle)
    P00.01                  Command source                  3
    
    P03.00                  RS485 Baud rate                 3 (9600)
    P03.01                  RS485 address                   1
    P03.02                  RS485 protocol                  2
    P03.08                  Frequency given lower limit     100.0Hz (match to your spindle cooling-type)

* had to wrap the .bin file in a zip, just extract and upload as normal.  

[CNC_xPRO_V5_YL620_test.zip](https://github.com/Spark-Concepts/xPro-V5/files/6718105/CNC_xPRO_V5_YL620_test.zip)

Emanuelbastos1, 2021-07-03:
@Mawg1 I tried your code! My VFD is a YL&620-A and it worked perfectly! 

I also changed the parameter shown below in the VFD, but I think that is not important.
Parameter number        Description                     Value
-------------------------------------------------------------------------------
P07.08                  First frequency source selection                  5

I can´t get the 0-10v output, but RS485 communication and PWM output are working! 
I'm planning to install an emergency NC button. Can I still use your code?
Thanks!

Emanuelbastos1, 2021-07-18:
@Mawg1 and @Spark-Concepts 
I am back again.
I burned a dc-dc regulator and I have a new controller. This time 0-10v output is working great! 
Because I prefer to use the RS485 communication, I tried to use your firmware and this time it is not working properly.
When I connect it to the computer it says: "Undefined VFD rs485 RTS pin" in the CNCjs console.
When I send the command to move, it starts and doesn´t stop, and stays at maximum speed. It goes to alarm mode because it is reading -1 rpm.
Do you have any clue about the problem?

Spark-Concepts, 2021-07-21:
Emanuel,

Just to clarify - the new controllers dc-dc regulator is still good right?

Have you had any luck getting the rs485 working with your YL620?

"Undefined VFD rs485 RTS pin" is normal - the xPro automatically provides the RTS via hardware rather than from the ESP32 module (thus, the rs485 pin is undefined in the firmware)

1. Double check your [YL620 VFD settings](http://vancura.biz/files/Documents/CNC/YL620-A%20Setup%20table.pdf)
2. Also, double check your [VFD](https://github.com/Spark-Concepts/xPro-V5/wiki/Front_Panel#spindletypexx--rs485) wiring
3. Verify PWM/RS485 switch is toggled to RS485 (toggle switch to PWM then back to RS485)
4. If you're still having issues, please post a screenshot of your current settings (command `$S` or `$$`)

Emanuelbastos1, 2021-07-24:
Hi,

Yes, everything is good, including the dc-dc regulator! 
I checked all the settings and verified everything twice.
I'm using a simple two-wire cable, without a shield, that I twisted myself with 30 cm. Can it be a noise problem?
It has the same behaviour, starts,  go to max speed and don´t even turn off.
The error: [MSG:Critical Spindle RS485 did not reach speed 2000. Reported speed is -1 rpm.]
My settings:
> $S
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=200.000
$Homing/Seek=2000.000
$Homing/Debounce=250.000
$Homing/Pulloff=2.500
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=YL620
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=100.000
$Y/Acceleration=100.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000<Idle|MPos:0.000,0.000,0.000|FS:0,0>
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=780.000
$Y/MaxTravel=785.000
$Z/MaxTravel=75.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok
> $i
[VER:1.3a.20210424:]
[OPT:MPHBSW]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_YL485_NO]
[MSG:Mode=AP:SSDI=CNC_xPRO_V5:IP=192.168.0.1:MAC=40-F5-20-A0-40-95]
[MSG:No BT]
ok

Thanks for the help!

Spark-Concepts, 2021-07-26:
Double check the following [YL620 VFD settings](http://vancura.biz/files/Documents/CNC/YL620-A%20Setup%20table.pdf)

    Configuration required for the YL620
    Parameter number        Description                     Value
    -------------------------------------------------------------------------------
    P00.00                  Main frequency                  400.00Hz (match to your spindle)
    P00.01                  Command source                  3
    
    P03.00                  RS485 Baud rate                 3 (9600)
    P03.01                  RS485 address                   1
    P03.02                  RS485 protocol                  2
    P03.08                  Frequency given lower limit     100.0Hz (match to your spindle cooling-type)

Spark-Concepts, 2021-08-05:
Have you had any success getting your YL620 VFD working via the RS485 without receiving `MSG:Critical Spindle RS485 did not reach speed 2000. Reported speed is -1 rpm` ?

Emanuelbastos1, 2021-08-07:
Unfortunately, I haven´t. I also triple-checked the settings and they are correct.
After I send the speed command, the spindle starts rotating, goes up until max speed, and stays at maximum speed until I cut the power of the vfd.

feeder> M3 S6000
feeder> M5
[MSG:Spindle RS485 Unresponsive 7]
[MSG:Critical Spindle RS485 did not reach speed 6000. Reported speed is -1 rpm.]
ok
ALARM:10

Radar224, 2021-08-14:
I also had lots of similar problems with this VFD. I gave up on it and purchased the HuanYang. 
Works on the first try.

closed 2021-09-16 by Spark-Concepts