Spark-Concepts/xPro-V5 #32: Steppers only work after connecting twice

allistar, 2021-06-16:
I use CNCjs to control my XPro. When I start CNCjs I connect it to the XPro (using USB). I then run the homing cycle. The CNCjs display shows the Z axis moving but the stepper isn't doing anything. I then disconnect and reconnect after which it works as expected. My process now when starting CNCjs is to connect, disconnect, then connect before using the machine. This issue happens every single time. In both cases when I connect I hear the stepper power on (a slight "snapping" sound).

What explains the behaviour and is there anything that can be done to fix it? I'm running the latest XYYZ RS485 firmware.

Spark-Concepts, 2021-06-16:
After connecting the COM port in CNCjs, press the reset button on the side of the xPro-V5 to sync to CNCJS

allistar, 2021-06-16:
Thanks. Do I need to do that every time I start the computer and/or the XPro?

pacman171, 2021-06-17:
Hi Allistar, I have to do the same every time. I switch the vfd and xPro on, then about 30 sec later I have to press the reset button. Also whenever I startup, it always shows an alarm.

Spark-Concepts, 2021-06-18:
Before you connect to the serial port in CNCjs check the "Enable hardware flow control" - this should eliminate you having to manually press the reset every time you connect.

[image]

pacman171, 2021-06-18:
Works for me, thanks!

closed 2021-06-18 by Spark-Concepts

allistar, 2021-06-20:
I always have "Enable flow control" set. I think my issue is that I wasn't giving the XPro enough time to start up. Now I turn on the VFD and XPro, then the computer and by the time the PC has started up and I run CNCjs this issue doesn't occur. I'm reclosing this issue, thanks for the help.