Spark-Concepts/xPro-V5 #33: 0-10 V not working

Emanuelbastos1, 2021-06-19:
I can´t get the 0-10 V analog output working!
Firstly when I sent the command M3 S1200 to the CNC Xpro V5 the voltage 0-10 V output didn´t stop until it reached the maximum value. Even when turning it off and on again, if I send the command M3 S500 the voltage kept going until the maximum. Later it didn´t give any value of voltage regarding any M3 SXXXX I type, and it is like that right now.
All the axis and limit switches are working great.
Anyone can help me?

Spark-Concepts, 2021-06-21:
Emmanuel,

Once again, my apologies for my delayed response - we had a slight family emergency over the weekend. 

I see from your other post that you measured voltage 0-10V on the x-pro. Are you verifying this using a multimeter?

If possible, please check the following (disconnect everything except 24V power and USB (if using a serial sender)): 
1. Ensure the switch on the xPro is set to EN/PWM if using PWM
2. Please download [test2.nc](https://github.com/Spark-Concepts/xPro-V5/blob/main/test2.nc) 
3. Run test2.nc: using a multimeter - place the negative lead on the ground terminal on the front of the xPro - as soon as you hit play to run test2.nc - you should see ~2.5V on GND-and-PWM, and ~5V GND-and-0-10V
[image]

Please let me know if you're measuring voltage while test2.nc is running

Emanuelbastos1, 2021-06-22:
Yes, I'm verifying this using a multimeter.
I did exactly as you said and I still measure 0V at the 0-10V output and 5V
In the PWM. (Before I was reading 0V in the PWM output).

Emanuelbastos1, 2021-06-22:
This time I notice something different. After switching on the XPRO V5,
just connect to power, the PWM output is always 5V. (First I was getting 0V)
I runned the code test2.nc, only connected to 24V and usb connection. The
results were the same, 0v in the 0-10V output and 5V in the PWM output.
I did this previously without being connected, so I did it again and the
program was running correctly in the CNCJs desktop app.
Can I do something else?
Thank you for the support!

terça, 22/06/2021 à(s) 18:22:

Spark-Concepts, 2021-06-23:
I believe some (if not all) of your troubles may be software (setting) related - refer to https://github.com/Spark-Concepts/xPro-V5/issues/30#issuecomment-865413416

Once we've verified the correct firmware & spindle settings - once again, disconnect everything except 24V power and USB (if using a serial sender):

### NOTE: 24V must be connected and `ON` for this test...

1. Ensure the switch on the xPro is set to EN/PWM
2. Download, open, and run [`test2.nc`](https://github.com/Spark-Concepts/xPro-V5)
[image]

4. As soon as you hit run using a multimeter and the negative (black) lead to ⏚; as soon as you hit play to run `test2.nc`, measure "PWM" and "0-10V" on "TOOLHEAD" connector, you should measure the folloing:
[image]

Spark-Concepts, 2021-06-23:
If you're seeing ~2.5V on the `PWM`, but still not seeing anything on the `0-10V`

1. Remove the 4x screws on the lid and press firmly on the two area's as seen below to ensure the board-board connectors are positively engaged
2. Using a small flat head screw driver, adjust the potentiometer clockwise then counter clockwise
3. Follow the test procedure above
4. If voltage is present on `0-10V' while running `test2.nc', adjust the voltage to 5V ±.1V
 
[image]

Emanuelbastos1, 2021-06-24:
I'm doing everything as you say and I still don´t have any voltage in the 0-10 V output.
The PWM seems to be working properly.
I push in the areas you said to ensure the pins are connected.
I also made a video that you can download from the link bellow:
[https://we.tl/t-FExh48CNvU](url)
[image]
[image]

closed 2021-07-18 by Emanuelbastos1

Anson90, 2022-03-20:
I have a similar issue where my voltage is stuck at 2.4v (0-10v pin) and the fine adjustment doesn’t do anything.