Spark-Concepts/xPro-V5 #34: USB problem

Radar224, 2021-07-02:
I wonder if there is anyone that could help me with this:

I recently acquired a XPo V5. 
I installed it and after tweaking the settings it ran fine.  However, I did noticed that I had some intermittent issues with my axis.
After some troubleshooting I realized that if I unplug the USB cable to my computer the controller stops sending stepper outputs. Also, the radio AP is not visible without plugging in the usb cable. 
Today I realized when I connect the USB, the blue LED below the reset button comes on briefly and immediately tuyrns off again. Even if the cable is connected and sending commands.
When connected by USB I am able to connect through WiFi and access the miniSD card.

The whole thing runs fine as long as the USB is connected to a computer.

Please help!

Thanks
Marinus

Spark-Concepts, 2021-07-03:
Hey Marinus,

I see you've also submitted this question via [contact us](https://www.spark-concepts.com/contact-us-1/) through the [Spark-Concepts](https://www.spark-concepts.com/) website.  Please continue to coordinate via email directly if problem your persists. - please know this is a rare issue & one I haven't seen before. (note, you can also plug into a USB power brick instead of a PC...)

The Blue LED will illuminate upon plugging the xPro into a computer USB port or USB power brick... However, if you don't Open the COM Port via a GRBL sender or Serial terminal, the Blue LED will turn off after a few seconds... (this is normal)

That said, and given it's the weekend, and you're willing/able, lets try and check a few things.

1. Disconnect the USB cable and power attempt to power using your 24V power supply
2. Remove the lid (removing the four lid screws) to expose the top PCB
3. Turn your 24V power supply ON 
4. Measure for 5 Volts as shown below (also check the remaining voltages as shown on the right)
5. Remove 24V or turn OFF 24V power supply
6. Plug in USB cable, and power via a PC or USB power brick 
7. Re-Measure for 5V and 3.3V (24V should be OFF)
8. Please let me know your test results

[image]

Radar224, 2021-07-05:
Hi again,

I am not sure if you seen my email.

I have tested as per your request and found 0 volt at the 5v point you
indicated. This is with the unit powered up.

All the other voltages are as you described.

I am currently using a power brick to power the unit through USB.

Thanks much for your help!
Marinus Onos

Spark-Concepts, 2021-07-06:
Marinus,

Okay, for the time being I recommend you continue to use the USB power
brick.

We've noted a few issues where the XL1509 DC-DC converter chip has been
damaged.  This is primarily due to the user inadvertently short circuit 5V
to ground via a limit switch and/or z-probe output (not implying or saying
this is the cause in your case).   Ultimately we've noted slight variances
of this part's features where the "built in" over current protection has
failed; this often results in the permanent damage of the DC-DC regulator.

Fortunately, albeit unfortunately, this issue has been mitigated in the
current production of the xPro. As I mentioned, continuing the use of the
USB power brick - a perfectly valid workaround.  However, if this does
become an issue - I recommend contacting the vendor from which you
purchased your xPro to request a replacement (although, wait until the
vendor receives a new shipment of xPros).

V/r
Bill

closed 2021-07-10 by Spark-Concepts