Spark-Concepts/xPro-V5 #36: Updated Firmware and now i cant change the default GRBL settings

timm052, 2021-07-18:
After updating the firmware to the latest version from whatever the controller came with i can no longer change the default GRBL settings after falshing the cnc will work fine allowing movement all axis. But upon setting the correct settings for my cnc the motor will cease to move GRBL still registers the motors as moving but they will not move i have attempted going back to the firmware from the included SD and fashing it multiple times. When applying my setting via the command line the controller gives this error [MSG:Error: out of RMT channels] this error on pops up if i try to set the setting via macro if i do it line by line no error pops up but the motors cease to move anyway. Any help would be much appreciated.

Spark-Concepts, 2021-07-20:
Could send commands $S and $i to the controller and post the responses?

timm052, 2021-07-28:
I reflashed the firmware a few more times and did some resets eventually it seems to have come good.

closed 2021-07-28 by timm052