Spark-Concepts/xPro-V5 #37: CNC Xpro V5 FW problem

DanVass1979, 2021-08-06:
I installed the incorrect firmware I installed the XYAZ NC when I should have installed the XYYZ, the result is my xpro v5 is junk as is my CNC till I replace it . I can not log into the xpro to change the firmware as it asks me for user name and password , I have tried so many so called fixes to no avail so I need help because if I have to spend over £200 because of this so called feature them spark concepts have made the worst product I have ever encountered

blakamin, 2021-08-06:
Have you pressed the reset button and tried again?
Do you have an e-stop or door switch connected?
I had the same issue, and reset let me straight in again.

Spark-Concepts, 2021-08-11:
Reference back to https://github.com/Spark-Concepts/xPro-V5/issues/21 

New firmware is available for upload. This moves the safety lockout from the login screen to a simple DOOR:1 error in the WebUI giving a bit more detailed feedback on what needs to be fixed. Note that this is still a safety lock and the controller will not proceed until the Door/Estop condition is met, and a reset triggered. The new firmware adds a few nice features and fixes some minor bugs as well :)

closed 2021-08-11 by Spark-Concepts