Spark-Concepts/xPro-V5 #38: Question about the relay output

allistar, 2021-08-08:
Hi there,
  When the spindle turns on I'd like to turn on the water coolant pump. This is a fish aquarium style pump which plugs into the mains (230V, 10A). I assume that plugging the 230V into the relay outputs is a bad idea. Do I need to get the relay to switch some type of power board/plug which the pump plugs into?

Thanks,
Allistar.

Spark-Concepts, 2021-08-10:
Hi Allistar, 

That pump will be right on the operating edge of the on board relay.  We would recommend using an external Solid State Relay hooked to the "Cool" output of the controller for this operation.  Hook up as below, where the LOAD is the trigger side of the SSR.

allistar, 2021-08-10:
Thanks, that's exactly what I needed to know. I'll go searching for an external SSR I can use with mains power.
Thanks again.

closed 2021-08-10 by allistar