Spark-Concepts/xPro-V5 #39: xPro-V5: Warning "Laser Output Pin Not Defined"

Untitledduke, 2021-08-12:
Hi,

Using the xPro on my Lead 1515 with the HUANYANG VFD so I set this up using the RS485 switch and applicable firmware. It works.

I am trying to connect a laser now but I get a warning "Laser Output Not Defined" when setting $Spindle/Type=Laser. Laser mode is enabled. I can't control the laser intensity at all and I assume it is because of the warning.

The laser has an On/Off button on it so I assume I have to connect it to the relay. It has three connection Pins namely +12V, GND and PWM.

Can somebody please assist? What am I missing?

Untitledduke, 2021-08-12:
Some additional information.

To note, the relay keeps switching if the jumper is in the position indicated in the documentation i.e. SP_EN. If kept in the default position, this does not happen. Regardless of the jumper position, the mentioned warning is received.

Is it possibly the case that the firmware for RS485 can not be used to control a laser also?

Spark-Concepts, 2021-08-12:
For the older firmware (everything before yesterdays upload) you need to upload RS485 or PWM specific firmware as they share pins.  with yesterdays upload you can change these real-time using the $Spindle/Type= commands and setting the PWM / RS485 switch on the front to the correct option.  The only caveat to this is the BM3D Lasers, these have a resting PWM offset and will require the CNC_xPRO_V5_XYYZ_BM3D_NO firmware to continually drive the laser low - also recommend starting the xPRO and then the laser with these modules.

Untitledduke, 2021-08-12:
Hi,
Thanks for the feedback.

To confirm, if I used CNC_xPRO_V5_XYYZ_485_NC, then I should use the new firmware CNC_xPRO_V5_XYYZ_NC uploaded yesterday?

My wiring I leave as is for using a VFD connected to the RS switch and laser connected to the relay, correct?

Let me know and then I will test and revert back.

Untitledduke, 2021-08-12:
Hi, Just ran the update and it looks fine but the laser keeps on at what seems to be full power.

As soon as i put the laser on using its onboard on switch, it fires up and does not shut off.

Is there a setting I am missing or is this a laser issue in your opinion?

Untitledduke, 2021-08-13:
I decided to send M3 S3 and this dimmed the laser and kept it dimmed until I moved the laser into position. As soon as i sent the command to move in the -X direction, the laser went back to the same intensity as it was as power on.

Not sure if the fault lies with the laser now or the controller?

Untitledduke, 2021-08-13:
Hello @Spark-Concepts 

I decided to contact the xPro supplier here in South Africa (Neotronics) and did some further tests. They asked that I provide the details below to you.

We need your help please, not sure if the laser is faulty or the controller with new firmware is acting up.

Test process:

1. System switched on
2. Laser and controller confirmed to have power
3. Movements made in X and Y plane to position laser
4. M3 S3 command given, can hear relay switch in controller. Voltage measured between GND and PWM going to the laser - low value observed.
5. M5 command given, can hear relay switch in controller. +-5V measured at same point as above.

CNCJ's, Lightburn and LaserGRBL used to test and in all cases the laser does not switch on.

Kindly please confirm if the voltage outputs observed are correct i.e. this is the normal working or expected working of the controller.

Attached some setting details.
[image]
[image]
[image]

pacman171, 2021-08-13:
Hi Untitledduke, please could you contact me? I live in Cape Town, would be great to have a local person to bounce ideas off.
Cheers Paul (conlonfamily@gmail.com)

rhdhybrid, 2021-08-13:
Hello, I see your having a little issue with your laser, we have been experimenting with some firmware in this thread. 

https://github.com/Spark-Concepts/xPro-V5/issues/23

I would give this a read and try out the firmware, worked great for me. I'm still testing as I feel like the power commanded does not quite reflect the power observed, however it still functions as expected.

Spark-Concepts, 2021-08-13:
Hi Untitledduke - could you post the manufacturer of your laser and a picture or two of the wiring between the controller and the laser?

Untitledduke, 2021-08-13:
> Hi Untitledduke - could you post the manufacturer of your laser and a picture or two of the wiring between the controller and the laser?

Thanks for the reply.

I read your reply, went back to the machine and started taking pictures of the connections. Decided to test wire connections (conductivity and voltage output) for the umpteenth time to be sure they go to where they must and then I noticed an odd fluctuation in the voltage reading and flickering on the laser while testing one of my 4 core wires. I decided to disconnect the wire and connected a new one...looks like I had a 4 core wire that was damaged somewhere and for some reason passed each of my previous tests.

At the moment I can do the following:

1. I have to switch on the xPro first and then go to the software (tested in lasergrbl and lightburn and these worked for this).
2. In the software I need to send M3 S0 which switches the relay.
3. Once the above has been done I have to manually switch on the laser with a button on the laser itself. Because of the M3 S0 command, my laser does not burn at this time.
4. At this point, sending an M5 command results in the relay to switch off and then the laser fires at full strength. I can send an M3 S0 and it will switch on the relay and in turn the laser stops firing at full power.

At the moment It is key that I not switch on the laser using its onboard on/off button before sending the M3 S0 command. If I don't send the command, it starts up at full power and burns whatever is close.

I am not sure if this is normal working procedure for the laser I have and still need to report back to the supplier. Laser in question - [40W Laser](https://neotronics.co.za/index.php?route=product/product&path=113_139&product_id=701). **_I am also not sure if this laser falls in the group you mentioned namely BM3D?_**

With this type of laser I run the risk of burning projects because I see most of the engraving/cutting gcode end with an M5 line and an M5 means my laser goes to full power due to the relay switching off as explained earlier. I still need to see what I can do to work around this if at all possible. I assume that the relay switching off is correct?

Thanks again

Untitledduke, 2021-08-13:
> Hello, I see your having a little issue with your laser, we have been experimenting with some firmware in this thread.
>
> #23
>
> I would give this a read and try out the firmware, worked great for me. [...]

Thanks, will read through it.

Untitledduke, 2021-08-14:
> Hi Untitledduke - could you post the manufacturer of your laser and a picture or two of the wiring between the controller and the laser?

Hello @Spark-Concepts , trust you are all well.

Turns out 

1. I had a faulty wire that has no visible damage but does not conduct all the time, 
2. Laser I have is in the category described above i.e. BM3D - it keeps firing at full when switched on using the onboard on/off switch, M3S0 turns the intensity down but as soon as M5 or a cycle ends, it fires at full again. With no documentation on the laser, I had to run a couple of tests of which the last was changing firmware, switching on controller first then laser else it is a very unsafe situation.
3. Firmware for BM3D did the trick and I have not found any issues after the firmware change today.

Thanks for the support.

closed 2021-08-20 by Spark-Concepts