Spark-Concepts/xPro-V5 #42: P alarm and spindle control

dennis954rr01, 2021-08-15:
Hello, I'm very new to cnc and I'm having a terrible time getting my Xpro V5 to control the queenbee pro.  I've followed the RS485 directions to a T but I kept getting a P alarm when running on commander.  I've searched all over and can't figure out what  alarm is, is it EMI alarm?  I have shielded wire on everything and spindle completely seperate from the controller and limit switches but no luck.  Yesterday I cut off Hard limits and was able to get the spindle to fire up but could not control it via software only manually via the knob.  Today I can't even get the M3 and M5 codes to work.  =(  I enter M3 S6000 or M3 alone and nothing, it keeps showing Pn:P>.  I'm unsure what I'm doing wrong, any help would be so greatly appreciated.

Spark-Concepts, 2021-08-20:
Hi Dennis, for the 485 control, make sure you set the appropriate registers on your VFD and flip the PWM/485 switch on the front of the xPRO to the 485 side.  If you have loaded the latest firmware you will need to define your spindle type using the "$Spindle/Type = _option_" command.  If you open the controller in CNCjs, then press the manual resest button, you will see the 485 startup and/or error messages. 

The Pn:P alarm means that your probe is being triggered.  Could you send a list of your $S settings?

dennis954rr01, 2021-08-21:
Thanks, I wound up getting the RS485 going and just unplugged the Probe so the alarm is still there but it's not stopping the operation of it.   So I had everything working perfectly then the next day my X gantry was going haywire when I would move it up and down the y.  Every 2-3 clicks on the Y in either direction and one stepper would spin up in a different direction causing everything to get out of alignment.  After hours of troubleshooting I found that it was not any of the cables I was using but the Y1 terminal of the xpro v5.  I unplugged everything else and no matter the cable or stepper it was connected to if it was on the Y1 it would spin randomly.  One minute in the Y+ direction the next in the Y- direction even though I was moving it in only one direction.  I thought it might be EMI but I have EVERYTHING properly shielded and grounded plus right now all I have connected is the Y, I put it to the x stepper just so I could show the behavior and I'll try and attach that video to this email.  So to clarify, I have everything good except I have zero idea why the Y is going nuts even when it's the only stepper connected and I've ensured both the stepper and cable work perfectly on all the other terminals.  Please help!  lol

Here are the setting.

$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 5    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 24000.000    (Maximum spindle speed, RPM)
$31 = 8000.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$N1=
$N0=
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000
$104 = 100.000
$105 = 100.000
$110 = 4000.000    (X-axis maximum rate, mm/min)
$111 = 4000.000    (Y-axis maximum rate, mm/min)
$112 = 1500.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000
$114 = 1000.000
$115 = 1000.000
$120 = 500.000    (X-axis acceleration, mm/sec^2)
$121 = 500.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000
$124 = 200.000
$125 = 200.000
$130 = 1500.000    (X-axis maximum travel, millimeters)
$131 = 1500.000    (Y-axis maximum travel, millimeters)
$132 = 120.000    (Z-axis maximum travel, millimeters)
$133 = 300.000
$134 = 300.000
$135 = 300.000

Dennis Trisch
404-910-2249 mobile

dennis954rr01, 2021-08-21:
Here is the video link.  I have the X stepper hooked up to the Y terminal only to show how it randomly moves the stepper in different directions.  Video wouldn't attach the the previous email due to size so here is a link to a cloud version.  Thanks for your help.  https://photos.app.goo.gl/B8eZNxRT2dHzSPLCA
[https://lh3.googleusercontent.com/01flB8a_TiDW2dqpjK-lHWEHC9uChYtdOshdtf9ZKHl964IGrMAnXVcbpSKjtZ0xp-Ao3CLKg4xhJV_7T8ztjHRMZMQQzP9gRFTxI6hsli2YLOPhtQfs8b2d8kVM8UQ2fhB1CV7ZzLE=w600-h315-p-k-no]<https://photos.app.goo.gl/B8eZNxRT2dHzSPLCA>
New video by Dennis Trisch<https://photos.app.goo.gl/B8eZNxRT2dHzSPLCA>
photos.app.goo.gl

Dennis Trisch
404-910-2249 mobile

Spark-Concepts, 2021-08-23:
Hi Dennis - thank you for the video, definitely some strange behavior.  Per our emails, we are going to swap this unit out so we can investigate what is causing the directionality issue on the Y port.  We'll update here when we complete troubleshooting.  Thanks!

closed 2021-08-25 by Spark-Concepts

dennis954rr01, 2021-08-29:
Hello, so I received my new Xpro V5 in the mail today but sadly the problem is still exactly the same.  So that made me think my wiring might somehow be messed up but I had purchased a blackbox controller (dirty word I know) but I plugged it in and it's running perfectly.   I've got the same speeds, wiring, steppers, etc and the Xpro continually messes up on the Y2 no matter the configuration but the blackbox works with zero problems.

So I am totally confused as to what is going on with the xpro controller but unfortunately it won't do me much good given it won't work properly so my next question is if there is a way for me to return the Xpro and receive a refund if I bought it from Bulkman?  I'm going to attach a video to this email so that you can see the difference in performance when I swap them and get a visual of it.  I apologize for the length and amateur camera work but I hope it will illustrate the issue and you'll be able to see that it must be an issue with the controller.

Any help you could provide would be greatly appreciated, I'd love to use the controller but unfortunately I can't seem to get it to work.

https://photos.app.goo.gl/oJCCEd4bzNCVmEJh9
[https://lh3.googleusercontent.com/MLgMbNhPSyHz7uc8AX3KM_946dpXykJ0VtueBBNdB6QfCbyh8fQVD0d-ISBo8v_I_Bme75QywNR3Vax3yEB2la8uwqgRl5Z-o0YXVHnykNE6cJCgk9tJaCGPv28TKPm-t87NzZMkQcE=w600-h315-p-k-no]<https://photos.app.goo.gl/oJCCEd4bzNCVmEJh9>
New video by Dennis Trisch<https://photos.app.goo.gl/oJCCEd4bzNCVmEJh9>
photos.app.goo.gl

Thanks

Closed #42<https://github.com/Spark-Concepts/xPro-V5/issues/42>.