Spark-Concepts/xPro-V5 #43: Y axes won't home

cdo1955, 2021-08-22:
I have a OSH xPRO v5and its set up on a Queenbee 1500x1500mm system. At first everything would work, but sporadically Y would not Home. After Z and x home normally Y just jumps 3mm in the positive direction and errors out with the message seen below. Now Y never homes and I don't know what has changed. The error says failed to clear Y limit switch but it never gets close to the limit switch to start with.
I have homing set to the front left corner. And I have tried with soft limits and had limits on and off. Its strange because some times it works. Can a help with this issue?

GRBL settings

>>> $H
[Error] An error was detected while sending '$H': (ALARM:8) Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring. Streaming has been paused.
[Error] Error while processing response <An error was detected while sending '$H': (ALARM:8) Homing fail. Pull off travel failed to clear limit switch. Try increasing pull-off setting or check wiring. Streaming has been paused.>
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.
Grbl 1.3a ['$' for help]
>>> $$
$0 = 4    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 1    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 1000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 4.000    (Homing switch pull-off distance, millimeters)
$30 = 12000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$N1=
$N0=
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000   
$104 = 100.000   
$105 = 100.000   
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 2500.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 100.000    (X-axis acceleration, mm/sec^2)
$121 = 100.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 1250.000    (X-axis maximum travel, millimeters)
$131 = 1250.000    (Y-axis maximum travel, millimeters)
$132 = 120.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000

Spark-Concepts, 2021-08-23:
Hi this happens when you try to home and there is a trigger signal on the limit switch. 

Send command "?" will query the status of the switches - you will see the line ending with Pn: ### if a switch is triggered (the ### will be replaced with XY and/or Z depending on what axis is triggered).  If you see that Y is triggered, check your wiring to make sure it is wired as NO, check that the switch is fully releasing, and check that the switch is not overly tight (for some reason the microlimit switches short out when overly tight).

cdo1955, 2021-08-24:
Thanks for your response, I try all the above.
 Carl

cdo1955, 2021-09-02:
Can I ask you one more question?
I don't see how to land the control wires coming the xpro v5  to the Hungyang VFD for the 10VDC  to control the speed?
 What terminals inside the VFD would the  Ground abd 10 VDC line land on?
In the picture on the istructions page it just has the lines disappear inside a photo the VFD/
Can you enlighten me on this set up?
 Carl[cid:f02882b1-f1fa-44bb-a823-ade34cf1cd50]

closed 2021-09-15 by Spark-Concepts