Spark-Concepts/xPro-V5 #45: No information on macro functionality and TMC Diag

hellboytat, 2021-08-24:
Can you please update the wiki to include relevant information on these two subjects. 

Also is it possible to add more outputs through the GPIO to the board and control them through GCode?

Spark-Concepts, 2021-08-25:
Hi Hellboytat, 

The TMC Diag pins are in place to allow for future development of the embedded Stallguard functionality of the drivers.  The Macro inputs will be used for implementing specific gcode macros with a button push - also in development but if you have a suggested macro you would like to see please let us know.     

> Also is it possible to add more outputs through the GPIO to the board and control them through GCode?
I'm not 100% sure what GPIOs you are referencing, but unlikely as most of the pins on the ESP are in use for dedicated functions already

Spark-Concepts, 2021-09-21:
Macro and Diag info added to the Wiki.  See this section for setup and use https://github.com/Spark-Concepts/xPro-V5/wiki/Side_Panel#macro1--macro2

These are now active in the latest firmware push https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware

The pins indicator are NOT GPIO expansion, they are data/power/signal pins from the lower control unit to the upper motor driver unit.  Do not solder to these pins, all magic smoke will be released.

closed 2021-09-21 by Spark-Concepts