Spark-Concepts/xPro-V5 #46: Motor Y Light Indicator

luisvelez25, 2021-08-24:
Hi 
My problem is when I turn on the device the motor it never start, I checked the led indicator and the motor Y is red. 
I tried with the motor is disconnected
I checked all connections are ok 
I need your help

Thanks

Spark-Concepts, 2021-08-25:
Hi Luisvelez, 

If you disconnect the motor from the Y terminal, and press the manual reset button.  Does the fault go away?  Generally the indicator faults show if there is a short, or the driver is overheating.

luisvelez25, 2021-08-25:
Hi,
Please check the video
https://user-images.githubusercontent.com/7698841/130806455-8ad72ad0-0f27-4d71-986c-fbad97438564.mp4
Even with the motor disconnected the problem it still happens.  Is there a way to unlock the motor?

Thanks

closed 2021-09-14 by Spark-Concepts