Spark-Concepts/xPro-V5 #49: Y-stepper driver not working.

StefanEric, 2021-09-04:
I'm new to CNC routers and I was just test driving my first build.
Testing with a pencil and I was just getting the hang of it, I've had a 3d-printer for a few years.
But suddenly during a drawing of a nice looking Goomba the Y-motor started sounding bad and stopped.
The machine crashed because Y2 continued to move for a while.
However after that I can't get the Y-stepper motor to run.
When I let it rest for a while and try to move the Y-axis the motor only sounds bad for a short while and then I get a red led on Y-motor indicator. If I press reset and try again I get a temp warning on Y-motor test (Axis Trinamic driver test).
I don't know if I measured correctly but it looks like it's giving about double the volt and amps compared to the other drivers.
Is there anything I can do?
I tried updating to latest firmware but didn't help.
Some setting I can do?

For now I'm running both Y-motors on Y2, don't know if that's OK?
But my Goomba turned out fine. :)

Hope someone can tell me what to do?

Spark-Concepts, 2021-09-07:
Hi StefanEric,  could you send a list of your $S settings?  Running both motors on Y2 is ok while we get things sorted out.

StefanEric, 2021-09-07:
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 3    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 100.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 5.000    (Homing switch debounce delay, milliseconds)
$27 = 5.000    (Homing switch pull-off distance, millimeters)
$30 = 24000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000
$104 = 100.000
$105 = 100.000
$110 = 1500.000    (X-axis maximum rate, mm/min)
$111 = 1500.000    (Y-axis maximum rate, mm/min)
$112 = 1000.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000
$114 = 1000.000
$115 = 1000.000
$120 = 20.000    (X-axis acceleration, mm/sec^2)
$121 = 20.000    (Y-axis acceleration, mm/sec^2)
$122 = 20.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000
$124 = 200.000
$125 = 200.000
$130 = 550.000    (X-axis maximum travel, millimeters)
$131 = 1050.000    (Y-axis maximum travel, millimeters)
$132 = 30.000    (Z-axis maximum travel, millimeters)
$133 = 300.000
$134 = 300.000
$135 = 300.000

Ok! Good to know it also works with Y2 only.
Because everything works fine otherwise.

Mawg1, 2021-09-08:
Hi Stefan, thanks for the settings.  Could you send command $S (second one is the letter S not the $ sign).  This will include all the driver specific settings.  Thanks!

StefanEric, 2021-09-08:
Aha! Sorry!
Here they are:
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=xPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=10
$Stepper/IdleTime=25
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=Off
$Probe/Invert=Off
$Report/Status=3
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=Off
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=500.000
$Homing/Debounce=5.000
$Homing/Pulloff=5.000
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=NONE
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=1500.000
$Y/MaxRate=1500.000
$Z/MaxRate=1000.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=20.000
$Y/Acceleration=20.000
$Z/Acceleration=20.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=550.000
$Y/MaxTravel=1050.000
$Z/MaxTravel=30.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16

seaw14, 2021-09-10:
Yeah I've got two sitting on a self with this same thing. Randomly just dropped off. Nothing had changed in-between. Swapping controller fixed issue and it was on two separate machines.

closed 2021-09-17 by Spark-Concepts