Spark-Concepts/xPro-V5 #51: limit switches error issues

Reid-T, 2021-09-11:
I am new and building a workbee with a Xpro-V5 controller. I have issues with my limit switches. 

First, my limits can be hard triggered manually. But sometimes I get an error when I do the manual trigger.
Second, I can jog into the limit to trigger a hard limit. But sometimes I get an error that resets.
Thirdly, homing worked a couple time but now always fails with hard crash.

I have double checked the limit wires and connections. I even made a second set of limit wires. The results are the same.
I am using CNCjs and I have the same issues when I use SourceRabbit.

I am including my settings and the reset error I am getting.
Hopefully someone can spot the problem or explain what I need to look at?

Thanks in advance for any help!

> $$
$0=5 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=3 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=1 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=7 (Homing direction invert, mask)
$24=50.000 (Homing locate feed rate, mm/min)
$25=300.000 (Homing search seek rate, mm/min)
$26=25.000 (Homing switch debounce delay, milliseconds)
$27=8.000 (Homing switch pull-off distance, millimeters)
$30=12000.000 (Maximum spindle speed, RPM)
$31=100.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=2000.000 (X-axis maximum rate, mm/min)
$111=2000.000 (Y-axis maximum rate, mm/min)
$112=70.000 (Z-axis maximum rate, mm/min)
$113=2000.000
$114=1000.000
$115=1000.000
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=150.000 (Z-axis acceleration, mm/sec^2)
$123=150.000
$124=200.000
$125=200.000
$130=1300.000 (X-axis maximum travel, millimeters)
$131=1820.000 (Y-axis maximum travel, millimeters)
$132=43.000 (Z-axis maximum travel, millimeters)
$133=1270.000
$134=300.000
$135=300.000
ok
>
===== ===== =====
Guru Meditation Error: Core  1 panic'ed (Coprocessor exception)
Core 1 register dump:
PC      : 0x400df471  PS      : 0x00060031  A0      : 0x800df073  A1      : 0x3ffbe800
A2      : 0x3ffd8524  A3      : 0x00000001  A4      : 0x000000ff  A5      : 0x00000000
A6      : 0x00000010  A7      : 0x3ffc91d0  A8      : 0x00000001  A9      : 0x3ffbe8c0
A10     : 0x000000ff  A11     : 0x00000000  A12     : 0x800ef859  A13     : 0x3ffd41f0
A14     : 0x00000036  A15     : 0x00000000  SAR     : 0x0000000b  EXCCAUSE: 0x00000004
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffff0
Core 1 was running in ISR context:
EPC1    : 0x400df471  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008866c
Backtrace: 0x400df471:0x3ffbe800 0x400df070:0x3ffbe8f0 0x400d54a0:0x3ffbe920 0x40081f7b:0x3ffbe940 0x40082405:0x3ffbe960 0x40085e35:0x3ffbe980 0x4008fc7b:0x3ffd4190 0x4018940e:0x3ffd41d0 0x401773d4:0x3ffd41f0 0x400ef856:0x3ffd4220 0x400f84d6:0x3ffd4260 0x400eb380:0x3ffd4280 0x400eb048:0x3ffd42a0 0x400daa54:0x3ffd42c0 0x4009030d:0x3ffd42e0
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_PWM_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400d59f0  PS      : 0x00060031  A0      : 0x800820e3  A1      : 0x3ffbe930
A2      : 0x00000000  A3      : 0x00000000  A4      : 0x3ffc5c40  A5      : 0x00000000
A6      : 0x00000000  A7      : 0x00000003  A8      : 0x3ffd6dd0  A9      : 0x0000000c
A10     : 0x00000000  A11     : 0x3ffd6dd0  A12     : 0x80093262  A13     : 0x3ffd30c0
A14     : 0x00000001  A15     : 0x3ffd6dd0  SAR     : 0x00000000  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffb
Core 1 was running in ISR context:
EPC1    : 0x400d59f0  EPC2    : 0x00000000  EPC3    : 0x00000000  EPC4    : 0x4008866c
Backtrace: 0x400d59f0:0x3ffbe930 0x400820e0:0x3ffbe950 0x40085e35:0x3ffbe980 0x4000bfed:0x3ffd3130 0x40091479:0x3ffd3140 0x400ffdf2:0x3ffd3160 0x4018ef7a:0x3ffd31a0 0x40082045:0x3ffd31d0 0x400ded27:0x3ffd3210 0x400d4602:0x3ffd3230 0x400d2d73:0x3ffd3250 0x400fba97:0x3ffd3270 0x4009030d:0x3ffd3290
Rebooting...
ets Jun  8 2016 00:22:57
rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_PWM_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,1820.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,1820.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(0.000,43.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:Check limits]
[MSG:'$H'|'$X' to unlock]
>

Thanks in advance for any help!

Spark-Concepts, 2021-09-14:
Hi Reid-T, 

What version of the firmware are you using?  Could you send commands "$i", "$s". and "?" and post the response? 

Thanks

Reid-T, 2021-09-14:
Here is the output from $i, $s, and ?
> $i
[VER:1.3a.20201212:]
[OPT:MPHBSW]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_PWM_NC]
[MSG:Mode=AP:SSDI=CNC_xPRO_V5:IP=192.168.0.1:MAC=98-CD-AC-6D-66-C1]
[MSG:No BT]
ok
> $s
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=Z
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=On
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=
$Homing/Squared=
$Homing/Feed=50.000
$Homing/Seek=50.000
$Homing/Debounce=25.000
$Homing/Pulloff=10.000
$GCode/MaxS=12000.000
$GCode/MinS=100.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=PWM
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2000.000
$Y/MaxRate=2000.000
$Z/MaxRate=2000.000
$A/MaxRate=2000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=1220.000
$Y/MaxTravel=1210.000
$Z/MaxTravel=30.000
$A/MaxTravel=1270.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=Off
ok
> ?
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Pn:PXYZ>
ok

Spark-Concepts, 2021-09-14:
OK, so it looks like a few things could be happening here.  

1) Let's get you to the latest firmware:  https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NC.bin (upload with following these directions: https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading#ota-over-the-air-firmware-updates)
*Take note of any specialized settings like steps/mm, direction invert, etc).  Push the default firmware settings using command $RST=*, then reapply specialized settings.  Reboot

2) It looks like your limit switches are NC.  You will need to set $Limits/Invert=0;

Reid-T, 2021-09-14:
I have download the firmware and I have connected to the Xpro-V5. When I attempt the update, I get a file write error. This is shown in the video I have attached.

https://user-images.githubusercontent.com/21341854/133323877-09132d1f-7e2b-4bc9-8a52-69cae617917f.mp4

Spark-Concepts, 2021-09-16:
Ok, we will need to revert to the original firmware over USB, and then push the upload of the newest firmware.  

Attached are instructions for uploading the firmware over USB.  Unzip the programming folder, then unzip the .bin file.  

Once reconnecting to the wifi, you will need to upload the index.html.gz (WebUI) again - also attached.  

Alternatively, we can get this back to the shop and the techs will take care of it.  If you want to go this route, please submit a [contact us form](https://www.spark-concepts.com/contact-us-1/) with your name, order number, and address.  

[Programming USB No Arduino.zip](https://github.com/Spark-Concepts/xPro-V5/files/7177513/Programming.USB.No.Arduino.zip)

[index.html.gz](https://github.com/Spark-Concepts/xPro-V5/files/7177508/index.html.gz)

Reid-T, 2021-09-16:
I'm following your instructions and in the pdf file named "Programming ESP32 chip.pdf" step 1. instructs to download this file: 
https://www.espressif.com/sites/default/files/tools/flash_download_tool_v3.8.5.zip 
but the hyperlink results in a browser 404 error file not found.

Can you help me locate the file I need to download?
Thank you.

Spark-Concepts, 2021-09-16:
https://www.espressif.com/en/support/download/other-tools
[image]

Reid-T, 2021-09-16:
Sorry, but the instructions say to use xPV5xyyz.bin file. That is not included in the download.
Can you help me locate that file?
Thank you

Spark-Concepts, 2021-09-16:
The xPV5xyyz.bin is wrapped in the xPV5xyyz.zip folder.  You will need to extract the folder to get to the .bin.  

xPV5xyyz.zip included below as well.  (sorry, most systems dont play well with .bin files so we have to wrap them) 

Note: this bin will only work with the USB flash tool.  Using it with the Over The Air update will cause issues.  

[xPV5xyyz.zip](https://github.com/Spark-Concepts/xPro-V5/files/7179472/xPV5xyyz.zip)

Reid-T, 2021-09-16:
ok, I have flashed the xPro now.
Next step is : reconnecting to the wifi, you will need to upload the index.html.gz (WebUI) again - also attached.
When I try the  upload I get a file write error
![up
date-index-error](https://user-images.githubusercontent.com/21341854/133664776-620c5177-29c4-4f7b-82d1-cd44e49bca4a.PNG)

Spark-Concepts, 2021-09-16:
Rather stubborn unit... 

Connect to AP, navigate to: http://192.168.0.1/?forcefallback=yes

Hit the trash can button next to the index.html.gz 

Hit Choose Files, select new index.html.gz file, hit upload.  

Once uploaded, press manual reset on the controller.  Reconnect to AP.

Reid-T, 2021-09-17:
Thank you so very much for your help. My switches operate as expected and my homing works without crashing. Once Again Thank You. 
- Reid

Spark-Concepts, 2021-09-17:
Great!

closed 2021-09-17 by Spark-Concepts