Spark-Concepts/xPro-V5 #52: Feed hold, resume, reset buttons

wildfang, 2021-09-13:
Hey there,
I just got the xPro v5 and I wonder if and where I can connect physical feed control buttons.

IIRC the V4 had a connector for these. But now on the v5, I can't find anything in the wiki.

I have feed hold/resume buttons on my other grbl controlled CNCs and it's pretty convenient. Any hints regarding implementation are very welcome.

Thanks!

Spark-Concepts, 2021-09-14:
Hi Wildfang, 

Unfortunately, the V5 does not support physical Feed hold/Resume buttons directly.  These are something that can probably be added to the Macro1 and Macro2 button inputs, but that is untested at the moment.

wildfang, 2021-09-14:
Hi,
thanks for your answer.

How do I go about the Macro input buttons? There is nothing in the wiki.

After I connected them, what happens next? Do I have to enable them in GRBL or CNCjs or how do I access these? I guess I need to map the pins to a custom macro, but I don't see any info where and how.

Spark-Concepts, 2021-09-14:
Basic macros are implemented using settings:
`$User/Macro0=` *assigns a macro to the "Macro 1" input-(Normally Open Switch)*
`$User/Macro1=` *assigns a macro to the "Macro 2" input-(Normally Open Switch)*

1. **Single Command:** This can be any command or gcode. Example: `$User/Macro0=$H`

2. **Multiple commands:** Place and ampersand "&" between commands (limit 250 chars). Example: `$User/Macro0=G90&G53G0Z-1&G0X0Y0`

3. **File from SD Card:** Example: Use the $SD/Run command and a filename Example: `$User/Macro0=$SD/Run=foo.nc`

A file can be gcode, most Grbl_ESP32 commands and settings or a mixture of both.

*You must be in the idle state to execute button macros*

Spark-Concepts, 2021-09-14:
Hmm.. just tested this with feedhold ! and cyclestart ~ commands and this will not work.  

Attached is a version of the firmware with the Macro1 and Macro2 buttons replaced with FeedHold (Macro1) and CycleStart (macro2) 

[CNCxPROv5_XYYZ_NO_Feedhold.zip](https://github.com/Spark-Concepts/xPro-V5/files/7163777/CNCxPROv5_XYYZ_NO_Feedhold.zip)

wildfang, 2021-09-14:
**Thank you very much,** I will take a look tomorrow. That sounds really useful.

closed 2021-09-14 by Spark-Concepts