Spark-Concepts/xPro-V5 #53: Steppers don't work

Massoels, 2021-09-13:
Hi,

I can't get any of the stepper working. I'm using NEMA23 steppers and candle for control via usb. What could be wrong?
In candle, I get no errors and I can control the spindle.

Thanks,
Maarten

Spark-Concepts, 2021-09-14:
Hi Maarten, 

A couple things to check. 

1) Confirm you are sending 24V to the controller, wired for correctly polarity, and that the switch on the xPRO is in on position (...i know sorry, but it is obligatory to make sure we have power).  
2) Confirm that you have the motors wired correctly.  *with the motor unplugged from the controller, remove the first two wires from the connector and press the stripped ends together (short them).  While shorted, try to turn the motor shaft, if it turns the coils are not wires correctly.  If it locks, the coils are correct*
3)  press the manual reset on the side of the controller to re-initialize the motor drivers
4) check for any red fault indicators on the side of the controller

Massoels, 2021-09-14:
Hi, 

Thanks for the quick feedback. 

The reset button seems to work, but when I press it, the motors start making a nioise (like they are turning), but there is no movement. When I move X,Y or Z in candle, they rotate, but they keep making a noise. 
I also have to reset every time I start the controller. 

Does this sound familiar to you, or is this a know issue (or fault from my side)? 

The only red light on the controller is the one next to the STOP/DOOR, but I assume that's normal when there is no emergency button connected. 

Thanks, 
Maarten

Spark-Concepts, 2021-09-14:
There will be an audible noise from the chopping frequency when the motors are powered.  You can set $1=250 to remove power when they are idle.  

In Candle, you will need to make sure you have hardware flow control turned on, or look for an option to reset the controller on port connection.  If Candle does not have this option, you will need to manual press the reset button once you connect to the port. (sorry, not familiar enough with Candle to point you to a specific setting)

closed 2021-09-15 by Spark-Concepts

Massoels, 2021-09-15:
Hi, 

Thanks. I'm almost there. I changed $1 to 250 (was 255) and I could make the steppers spin. Although the sound is very odd. They make a sound before they start and after they stop. 
I have a small cnc milling machine and those steppers sound 'normal'. 

The other strange thing is that when I activate one axis, the other steppers make the same noise... (and don't spin of course) 

I included a small video, so you can hear it. 

Thanks in advance for your valuabale input. 
Maarten