Spark-Concepts/xPro-V5 #54: RS485 to Huanyang spindle not working

cdo1955, 2021-09-14:
I also can't get the spindle to responded to the RS485 signals. Pressing M3 or spindle buttons only gets me "0x9e."
I have read all the others with the same issue and have reset the switch, verified all my setting's in the VFD even reloaded the Firmware, but I still have the same issue.
Seems like a lot of folks have this issue, is it a glitch in the system?
Please give me any help you can in figuring this out
 The VFD values are below followed by the GRBl read out.

PD000=0
PD001=2
PD163=1
PD164=1
PD165=3
PD003=400
PD004=400
PD005=400
PD011=100
PD014=5
PD015=3
PD023=0
PD141=110
PD142=8.5
PD143=4
PD-144=3000

* Connected to COM3 @ 115200 baud ****
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1250.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,1250.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,1250.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-120.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
>>> $$
$0 = 4    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 3    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 1    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 1000.000    (Homing search seek rate, mm/min)
$26 = 250.000    (Homing switch debounce delay, milliseconds)
$27 = 4.000    (Homing switch pull-off distance, millimeters)
$30 = 12000.000    (Maximum spindle speed, RPM)
$31 = 0.000    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$N1=
$N0=
$100 = 200.000    (X-axis travel resolution, step/mm)
$101 = 200.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$103 = 200.000   
$104 = 100.000   
$105 = 100.000   
$110 = 2500.000    (X-axis maximum rate, mm/min)
$111 = 2500.000    (Y-axis maximum rate, mm/min)
$112 = 2500.000    (Z-axis maximum rate, mm/min)
$113 = 1000.000   
$114 = 1000.000   
$115 = 1000.000   
$120 = 100.000    (X-axis acceleration, mm/sec^2)
$121 = 100.000    (Y-axis acceleration, mm/sec^2)
$122 = 100.000    (Z-axis acceleration, mm/sec^2)
$123 = 200.000   
$124 = 200.000   
$125 = 200.000   
$130 = 1250.000    (X-axis maximum travel, millimeters)
$131 = 1250.000    (Y-axis maximum travel, millimeters)
$132 = 120.000    (Z-axis maximum travel, millimeters)
$133 = 300.000   
$134 = 300.000   
$135 = 300.000   
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
>>> $H
>>> $J=G21G91Y10F2000
>>> $J=G21G91Y10F2000
>>> $J=G21G91Y10F2000
ok
ok
ok
ok
>>> $J=G21G91X10Y10F2000
ok
>>> $J=G21G91X10Y10F2000
ok
>>> 0xa1
>>> 0x9e
>>> 0x9e

Spark-Concepts, 2021-09-14:
what software are you using to interface with the controller?  Could you send the full list of your settings ($S)? 

I would recommend updating to the latest firmware: https://github.com/Spark-Concepts/xPro-V5/blob/main/Firmware/CNC_xPRO_V5_XYYZ_NC.bin

You will need to define $Spindle/Type = "whatever your VFD is" (HuanYang, H2A, YL620)

The VFDs are a special pain as most are poorly documented in terms of changes and register settings.  The 0-10V control is pretty simple but for the 485 to work the VFD settings need to be correct, the command set correct, and any induced noise limited/removed.

cdo1955, 2021-09-14:
Thanks for the reply.
The soft ware is  CNC_xPRO-V5_XYYZ_RS485_NO.
I have defined the spindle both ways first as $Spindle/Type =HUANGYANG then as $Spindle/Type =H2A.
The spindle settings and grbl are:

The VFD values are below followed by the GRBl read out.

PD000=0
PD001=2
PD163=1
PD164=1
PD165=3
PD003=400
PD004=400
PD005=400
PD011=100
PD014=5
PD015=3
PD023=0
PD141=110
PD142=8.5
PD143=4
PD-144=3000
Connected to COM3 @ 115200 baud ****
[MSG:Grbl_ESP32 Ver 1.3a Date 20201212]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_RS485_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:X Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1250.000)]
[MSG:X Axis Trinamic driver test passed]
[MSG:Y Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(0.000,1250.000)]
[MSG:Y Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(0.000,1250.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-120.000,0.000)]
[MSG:Z Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485 Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
[MSG:Local access point CNC_xPRO_V5 started, 192.168.0.1]
[MSG:Captive Portal Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Flood coolant on pin GPIO(21)]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X Axis limit switch on pin GPIO(35)]
[MSG:Y Axis limit switch on pin GPIO(34)]
[MSG:Z Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]

$$
$0 = 4 (Step pulse time, microseconds)
$1 = 255 (Step idle delay, milliseconds)
$2 = 0 (Step pulse invert, mask)
$3 = 3 (Step direction invert, mask)
$4 = 0 (Invert step enable pin, boolean)
$5 = 1 (Invert limit pins, boolean)
$6 = 1 (Invert probe pin, boolean)
$10 = 1 (Status report options, mask)
$11 = 0.010 (Junction deviation, millimeters)
$12 = 0.002 (Arc tolerance, millimeters)
$13 = 0 (Report in inches, boolean)
$20 = 1 (Soft limits enable, boolean)
$21 = 1 (Hard limits enable, boolean)
$22 = 1 (Homing cycle enable, boolean)
$23 = 3 (Homing direction invert, mask)
$24 = 25.000 (Homing locate feed rate, mm/min)
$25 = 1000.000 (Homing search seek rate, mm/min)
$26 = 250.000 (Homing switch debounce delay, milliseconds)
$27 = 4.000 (Homing switch pull-off distance, millimeters)
$30 = 12000.000 (Maximum spindle speed, RPM)
$31 = 0.000 (Minimum spindle speed, RPM)
$32 = 0 (Laser-mode enable, boolean)
$N1=
$N0=
$100 = 200.000 (X-axis travel resolution, step/mm)
$101 = 200.000 (Y-axis travel resolution, step/mm)
$102 = 200.000 (Z-axis travel resolution, step/mm)
$103 = 200.000
$104 = 100.000
$105 = 100.000
$110 = 2500.000 (X-axis maximum rate, mm/min)
$111 = 2500.000 (Y-axis maximum rate, mm/min)
$112 = 2500.000 (Z-axis maximum rate, mm/min)
$113 = 1000.000
$114 = 1000.000
$115 = 1000.000
$120 = 100.000 (X-axis acceleration, mm/sec^2)
$121 = 100.000 (Y-axis acceleration, mm/sec^2)
$122 = 100.000 (Z-axis acceleration, mm/sec^2)
$123 = 200.000
$124 = 200.000
$125 = 200.000
$130 = 1250.000 (X-axis maximum travel, millimeters)
$131 = 1250.000 (Y-axis maximum travel, millimeters)
$132 = 120.000 (Z-axis maximum travel, millimeters)
$133 = 300.000
$134 = 300.000
$135 = 300.000
ok
$G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
$H
$J=G21G91Y10F2000
$J=G21G91Y10F2000
$J=G21G91Y10F2000
ok
ok
ok
ok
$J=G21G91X10Y10F2000
ok
$J=G21G91X10Y10F2000
ok
0xa1
0x9e
0x9e

Spark-Concepts, 2021-09-14:
This CNC_xPRO-V5_XYYZ_RS485_NO is older firmware, the one linked is the new firmware.  We also need to see the full xPRO settings using $S (note the second character is the letter S not a second $).  Also need to know what software you are using to interface with the controller (CNCjs, UGS, WebUI, etc).

cdo1955, 2021-09-14:
Ok gime me 1o minutes to get it booted up

cdo1955, 2021-09-14:
Here are the $S settings
>>> $s
$Sta/SSID=CNC_xPRO_V5
$Sta/Password=******
$Sta/IPMode=DHCP
$Sta/IP=0.0.0.0
$Sta/Gateway=0.0.0.0
$Sta/Netmask=0.0.0.0
$AP/SSID=CNC_xPRO_V5
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=AP
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Pulse=4
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=XY
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=On
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=On
$Limits/Hard=On
$Homing/Enable=On
$Homing/DirInvert=XY
$Homing/Squared=
$Homing/Feed=25.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=4.000
$GCode/MaxS=12000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=2500.000
$Y/MaxRate=2500.000
$Z/MaxRate=2500.000
$A/MaxRate=1000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=100.000
$Y/Acceleration=100.000
$Z/Acceleration=100.000
$A/Acceleration=200.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=1250.000
$Y/MaxTravel=1250.000
$Z/MaxTravel=120.000
$A/MaxTravel=300.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.800
$Y/Current/Run=1.800
$Z/Current/Run=1.800
$A/Current/Run=1.800
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=16
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
[Error] An error was detected while sending '$s': $Errors/Verbose=Off. Streaming has been paused.
[Error] Error while processing response <An error was detected while sending '$s': $Errors/Verbose=Off. Streaming has been paused.>
[Error] Error while processing response <ok>: An unexpected command was completed by the controller.

Spark-Concepts, 2021-09-14:
what program are you using to communicate with the controller?

cdo1955, 2021-09-14:
I am using UGS

cdo1955, 2021-09-14:
UGS

Spark-Concepts, 2021-09-14:
Try using CNCjs Desktop - https://cnc.js.org/docs/desktop-app/

cdo1955, 2021-09-14:
Do you want me to upgrade to the new firm ware now?

Spark-Concepts, 2021-09-14:
Yes. please. 

Note when replying to the github generated email, the response will post to the public issue/forum

cdo1955, 2021-09-14:
I think I screwed that up. Before I had the normally closed version but I downloaded the normally open version. Now I’m getting the log in request?
How do I get out of that?

Spark-Concepts, 2021-09-14:
close your switch, then power cycle the board.  

load the NC firmware, and open the switch

cdo1955, 2021-09-14:
So now the spindle button lights up and stays n but the spindle does not come on, or with M3?

cdo1955, 2021-09-14:
So with the new firmware its nt working.
When I type in Spindle/Type=  I get:
>>> 0x9e
>>> $Spindle/Type=
[Error] An error was detected while sending '$Spindle/Type=': (error:2) Missing the expected G-code word value or numeric value format is not valid. Streaming has been paused.
[Error] Error while processing response <An error was detected while sending '$Spindle/Type=': (err

cdo1955, 2021-09-14:
When I define the $Spindle/Type =H2A I get Spindle not responsive With HUANYANG I get RS485 not responsive10 repeating over and over again

Spark-Concepts, 2021-09-14:
Try using CNCjs Desktop - https://cnc.js.org/docs/desktop-app/
It looks like UGS is not appreciating the non-GRBL standard commands/settings

Reset the settings to firmware default - $RST=* 
then you need to define the spindle $Spindle/Type=HuanYang

Spark-Concepts, 2021-09-14:
RS485 switch set to the 485 side? 

Under the switch what color lights do you see?

cdo1955, 2021-09-14:
Done message
>>> $RST=*
[MSG:WiFi reset done]
[MSG:BT reset done]
[MSG:Settings reset done]
[MSG:Position offsets reset done]
ok
>>> $Spindle/Type=HuanYang
[MSG:Initializing RS485 VFD spindle]
[MSG:Undefined VFD_RS485_RTS_PIN]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:GPIO(0)]
Ok

cdo1955, 2021-09-14:
The spindle button lights up with m3 and goes off with m5 but the spindle does not spin

cdo1955, 2021-09-14:
Should I try CNCjs

cdo1955, 2021-09-14:
OK it works in CNCjs but my movements are back wards, must have been the reset

cdo1955, 2021-09-14:
When I reloaded my olds grbl settings I had the Y axes start to make bad noises when I tried to move it. The spindle also started to squeak.
Any ideas?

cdo1955, 2021-09-14:
I did a reboot and things seem normal now. Any other Ideas concerning getting the Spindle working in UGS?

Spark-Concepts, 2021-09-14:
No that it is setup, you should be able to send standard M3 Ssss commands through UGS (untested).  

Try M3 S12000, then follow with an M5

closed 2021-09-15 by Spark-Concepts