Spark-Concepts/xPro-V5 #55: Motor indicator light stays on.

IdahoCreator, 2021-09-19:
I noticed lately that the motor issue indicator lights are on even after cold start after disconnected over night. I also have issues with cutting files stop cutting in the middle of the project for no known issue. Even after cutting one completely and without issue. I even modified file and deleted prior cut order and still had problem in the same location. CNC works when I move stuff manually. controller is out in the open and runs on a 300 watt 24 volt power supply. Nema 23 steppers amperage set to 1.5 amps.

Spark-Concepts, 2021-09-20:
Hi IdahoCreator, 

Could you plug in USB and pull the startup message using CNCjs or similar?  

Start message will look like [this]([image]).  We want to see the motor self test results. 

If you unplug the motors from the xpro, then press the manual reset button, do the fault indicators stay active? 

Thanks

IdahoCreator, 2021-09-22:
CNCjs didnt give anything but yat did. Yat info listed below. Red lights are always on on all 4 motor locations I am running a 1x 2 y 1 z configuration. I did try new firmware but had to go back to original as the system would always lockup no matter which version of software I used. I had one great cut then the second one failed a little over half way through cut. Wifi constatnly drops from network so I have to use it in AP all the time.

From YAT
:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8

[MSG:Grbl_ESP32 Ver 1.3a Date 20200929]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Init Motors]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 Limits(0.000,0.000)]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 Limits(0.000,0.000)]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 Limits(0.000,0.000)]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 Limits(0.000,0.000)]
[MSG:X  Trinamic driver test passed]
[MSG:Y  Trinamic driver test passed]
[MSG:Y2 Trinamic driver test passed]
[MSG:Z  Trinamic driver test passed]
[MSG:TMCStepper Library Ver. 0x000701]
[MSG:PWM spindle Output:GPIO(25), Enbl:GPIO(4), Dir:None, Freq:5000Hz, Res:13bits]
<LF>[MSG:Client Started]
[MSG:Connecting notyours]
[MSG:Connecting.]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connected with 192.168.1.89]
[MSG:Start mDNS with hostname:http://CNC_xPRO_V5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]

Grbl 1.3a ['$' for help]

Spark-Concepts, 2021-09-22:
Thanks!  

That's very strange behavior.  Would you be interested in doing an exchange so we can troubleshoot here?  If so, just send me shipping info here: https://www.spark-concepts.com/contact-us-1/

How far away is your controller from your home wifi - might need to move it closer or use a wifi extending to boost the signal (or run in AP mode).

closed 2021-09-24 by Spark-Concepts