Spark-Concepts/xPro-V5 #56: Which software to use for 0-10V firmware

cdo1955, 2021-09-19:
I am trying to go to the 0-10 volt PWD firmware for control of my VFD since we could never get the RS485 firmware to work.
But the names of the newest firmware don't have the letters PWM in it. So I can't tell which  to chose. What's the the correct firmware for a NC  XYYZ system PWM system?

cdo1955, 2021-09-19:
could I also see a drawing of exactly how to land the 0-10 v wires on the VFD?

Spark-Concepts, 2021-09-20:
Hi cdo1955 

With the new firmware each spindle build is included.  You will need to define $Spindle/Type = PWM  (this activates PWM and the 0-10V outputs).

Wire like this for the xPRO side.  How they land on the VFD will depend on the manufacturer of the VFD

[image]

closed 2021-10-01 by Spark-Concepts