Spark-Concepts/xPro-V5 #57: Two resets required to clear Z axis driver

bobd-danlaw, 2021-09-20:
I have firmware VER:1.3a.20210816 on my xPRO V5 and have found that I need to reset twice to clear a driver test failure on the Z axis.  The message is:  MSG:Z  Axis Trinamic driver test failed. Check motor power]
After the second reset, all works well.  Subsequent resets are also successful.  This only happens on the Z axis and does the same thing with different motors.

Spark-Concepts, 2021-09-20:
Hi bobd, would you be willing to trade out for a new controller?  I have one other report of the Z powering up slowly but can't replicate in our shop.  I would love to get a hold of one to see whats going on.  If this is ok, please submit a form [here](https://www.spark-concepts.com/cnc-xpro-v5/) with your shipping info.  If not, I can push you some firmware to upload that *should* correct the issue.  Thanks

bobd-danlaw, 2021-09-21:
> CAUTION: This email originated from outside of the organization. Do not click links or open attachments unless you recognize the sender and know the content is safe.

Yes, I’d be willing to trade but that link to the form didn't work – it just went to the normal xPRO-V5 page.  So here is my shipping info:

Bob D’Aoust
Danlaw Inc.
41131 Vincenti Ct.
Novi, MI 48375

My phone number is 313 598 1535

If you need anything else, please let me know.  I am also sending this information on the Spark Concepts contact webpage.

Bob

Robert D’Aoust
Sr. Hardware Engineer
mobile: (313) 598-1535
phone: (248) 987-7345
e-mail: ***@***.******@***.***>
web: www.danlawinc.com<http://www.danlawinc.com/>
***@***.***
Danlaw Inc., 41131 Vincenti Court, Novi, MI 48375