Spark-Concepts/xPro-V5 #58: Cooling the M8 does not work

R-M-Y, 2021-09-22:
Hi, 
I have a little problem, to activate the cooling ports I have to make an M7 instead of an M8.
Is it possible to use the M8 instead of the M7? and how to?
Thank you

Spark-Concepts, 2021-09-22:
Hi, 

M7 is the default command to activate cooling but switching to M8 is a compile time option.  

The source code is available here: https://github.com/Spark-Concepts/xPro-V5-Firmware 

In the CNC_xPRO_V5_Machine_Template.h file you will need to comment out line 85, 

#define COOLANT_MIST_PIN        GPIO_NUM_21

and uncomment line 88

//#define COOLANT_FLOOD_PIN		GPIO_NUM_21

Then upload to the controller.  You can export a compiled binary from Arduino and load through the WebUI, or load directly using USB.  If using USB you must press and hold the programming button on the controller until the firmware begins loading.

R-M-Y, 2021-09-23:
> Salut,
> 
> M7 est la commande par défaut pour activer le refroidissement, mais le passage à M8 est une option de compilation.
> 
> Le code source est disponible ici : https://github.com/Spark-Concepts/xPro-V5-Firmware
> 
> Dans le fichier CNC_xPRO_V5_Machine_Template.h vous devrez commenter la ligne 85,
> 
> #define COOLANT_MIST_PIN GPIO_NUM_21
> 
> et décommentez la ligne 88
> 
> //#define COOLANT_FLOOD_PIN GPIO_NUM_21
> 
> Ensuite, téléchargez sur le contrôleur. Vous pouvez exporter un binaire compilé à partir d'Arduino et le charger via l'interface Web, ou le charger directement via USB. Si vous utilisez l'USB, vous devez appuyer sur le bouton de programmation du contrôleur et le maintenir enfoncé jusqu'à ce que le micrologiciel commence à se charger.

Hello, thank you for your quick response.
Will this operation make me lose my settings?
Thank you

Spark-Concepts, 2021-09-23:
Yes.  Make note of your current settings, upload the new firmware, then issue command $RST=* (this will set all settings to firmware default)

Spark-Concepts, 2021-09-23:
Are you using an NO or NC door switch?

R-M-Y, 2021-09-24:
> Utilisez-vous un interrupteur de porte NO ou NC ?

No, I do not use a door switch, only an NC emergency stop

Spark-Concepts, 2021-09-24:
Ok, give this a try.  Unzip and upload through the WebUI, expect the door/estop port to be closed on startup 

[CNCxPROv5_XYYZ_NC_M8.zip](https://github.com/Spark-Concepts/xPro-V5/files/7227858/CNCxPROv5_XYYZ_NC_M8.zip)

closed 2021-10-01 by Spark-Concepts