Spark-Concepts/xPro-V5 #59: Connect Tools with IoT Relay not working

Reid-T, 2021-09-23:
question regarding my iot relay on xPro-V5:
Prior to the help you gave me on #51 https://github.com/Spark-Concepts/xPro-V5/issues/51 my iot relay was working fine. Now it is not.
I have the xPro-V5 flashed with CNC_xPRO_V5_XYYZ_NC.bin
M3 commands don't turn on the router (or anything plugged into the iot's normally off plugs)
I have attached images which show the wiring connection.
What do I have wrong?
Thanks
[image]
[image]

Spark-Concepts, 2021-09-23:
Have you defined the spindle type?  

Looks like you will want to send command $Spindle/type = pwm

Reid-T, 2021-09-23:
Yes spindle type is pwm. I hear the controller when M3 is issued - but nothing is happening.
Here's my setup file

$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Spindle/type = pwm
$0=4
$1=250
$2=0
$3=4
$4=0
$5=1
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=1
$21=1
$22=1
$23=7
$24=50.000
$25=600.000
$26=25.000
$27=8.000
$30=24000.000
$31=0.000
$32=0
$100=200.000
$101=200.000
$102=200.000
$103=200.000
$104=100.000
$105=100.000
$110=2000.000
$111=2000.000
$112=500.000
$113=1000.000
$114=1000.000
$115=1000.000
$120=50.000
$121=50.000
$122=50.000
$123=200.000
$124=200.000
$125=200.000
$130=1240.000
$131=1200.000
$132=28.000
$133=300.000
$134=300.000
$135=300.000

Reid-T, 2021-09-23:
I changed the wiring to pin EN and changed $Spindle/Type = RELAY
works now

closed 2021-09-24 by Reid-T