Spark-Concepts/xPro-V5 #62: Questions on Stepper Motor Wiring - Different Schematics Shown

mschreib-il, 2021-09-26:
- there are 2 different wiring configurations shown in the Spark-Concepts documentation for the xPro V5 controller: https://github.com/Spark-Concepts/xPro-V5/wiki/Basic-hookup#connect-motors shows the wiring configuration for Y-Axis and Y2-Axis to be reversed while https://github.com/Spark-Concepts/xPro-V5/wiki/Rear_Panel#motor-connectors shows consistent wiring connections across all 4 stepper motors. Which is correct?  Incidentally, I'm not seeing any reason why Y-Axis and Y2-Axis should be reversed since the 2 axes should have identical operation. 

- additionally, https://github.com/Spark-Concepts/xPro-V5/wiki/Basic-hookup#connect-motors  asks the user to identify separate wire coils from the stepper motors, but does not explain why this is necessary. I'm assuming that 1 coil is wired to "A" ports and the 2nd coil is wired to "B" ports: is this correct? If so, perhaps the documentation can be clarified? Thanks

Spark-Concepts, 2021-09-27:
Hi mschreib-il. 

1) the different wiring is for reversing one of the Y motors as you noted.  This is for use on machines like the Workbee (belt) shapeokos etc. 

2) Correct, the coils are wired across the A+ A- terminal, this is a little more clear is you look at the silkscreen on the controller.  

We will clarify on the wiki as well

mschreib-il, 2021-09-28:
- OK: so for a thread-screw router where Y and Y2 are operating  in parallel, the same wiring is appropriate; for a belt-driven CNC where Y and Y2 are operating in  complementary motion, then the wirings need to be reversed. Please confirm that I'm understanding this correctly and I'll close this issue. 

- probably a good idea to document this more clearly as well; for beginners such as myself these concepts are not intuitively obvious. 

Thanks for the quick response.

Spark-Concepts, 2021-09-29:
1) You are correct, basically if the motors face in the same direction on a bound/slaved axis they get wired identically.  If the face opposite directions, one must turn counter to the other.  

2) agreed, thank you for mentioning!

closed 2021-09-29 by mschreib-il