Spark-Concepts/xPro-V5 #64: Mid Program stoppage

281Er, 2021-10-01:
I am running the cnc xpro v5 and I am having issues with long programs. Usually when I run a program longer that 30 minutes it ends up stopping mid program. I'm running over webui on grbl, and it just goes to idle on the top right where you are able to pause program if needed. From my understanding once I send the program to the controller it goes to the sd card, so it shouldn't be an issue of the laptop. The program is perfect no issues there, I'm using aspire. If anyone has any insight I would appreciate it, I will also post this on the grbl column in case it's an issue with grbl.

Spark-Concepts, 2021-10-01:
Here is a troubleshooting guide for the WebUI.  I think this should get you going again.  
[Troubleshooting CNC xPRO V5 WebUI 1OCT21.docx](https://github.com/Spark-Concepts/xPro-V5/files/7269170/Troubleshooting.CNC.xPRO.V5.WebUI.1OCT21.docx)

281Er, 2021-10-01:
It seems formatting the sd card might have done it. If the problem persists I'll upgrade the sd card.

closed 2021-10-05 by Spark-Concepts