Spark-Concepts/xPro-V5 #65: Unable to Control VFD

sflom, 2021-10-02:
So I have been struggling getting my QueenBee 1515 with the CNC Xpro V5 to communicate with my HuanYang VFD - 2.2kw water cooled spindle via USB.  I can issue the M3 S6000 command and the spindle starts and the relay kicks on my water pump.  After a little while, I get the [MSG:Spindle RS485 Unresponsive 6]
[MSG:Critical Spindle RS485 Unresponsive]
ALARM:10
[MSG:Critical Spindle RS485 did not reach speed 6000. Reported speed is -1 rpm.]
ok
ALARM:10

I purchased an inline power filter, star grounding and ferrite chokes to help with interference.  I added these products and was able to get the spindle to start and run error free for one evening on the bench.  The next evening, I mocked up my electronics cabinet and added the parts where I thought they might fit and decided to verify it was still working.  But I started getting unresponsive messages and I can not turn off the spindle (again).  

I seen the latest guide that Spark-Concepts wrote up for the RS485 spindle control but that hasn't helped me.  I've reset my VFD and reinstalled my firmware but still can't get past this.  I'm hoping it just a setting that I'm missing.

I'm also using CNCjs to communicate via USB. The $H works fine.

Any suggestions are great appreciated.

 ets Jun  8 2016 00:22:57
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[MSG:Grbl_ESP32 Ver 1.3a Date 20210816]
[MSG:Compiled with ESP32 SDK:v3.2.3-14-gd3e562907]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Axis count 3]
[MSG:RMT Steps]
[MSG:Door switch on pin GPIO(16)]
[MSG:Macro Pin 1 GPIO(13)]
[MSG:Macro Pin 2 GPIO(0)]
[MSG:Init Motors]
[MSG:TMCStepper Library Ver. 0x000703]
[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,1300.000)]
[MSG:X  Axis Trinamic driver test passed]
[MSG:Y  Axis Trinamic TMC5160 Step:GPIO(27) Dir:GPIO(26) CS:GPIO(17) Disable:None Index:2 R:0.050 Limits(-1270.000,0.000)]
[MSG:Y  Axis Trinamic driver test passed]
[MSG:Y2 Axis Trinamic TMC5160 Step:GPIO(33) Dir:GPIO(32) CS:GPIO(17) Disable:None Index:3 R:0.050 Limits(-1270.000,0.000)]
[MSG:Y2 Axis Trinamic driver test passed]
[MSG:Z  Axis Trinamic TMC5160 Step:GPIO(15) Dir:GPIO(2) CS:GPIO(17) Disable:None Index:4 R:0.050 Limits(-106.000,0.000)]
[MSG:Z  Axis Trinamic driver test passed]
[MSG:Initializing RS485 VFD spindle]
[MSG:Auto-RTS in use]
[MSG:VFD RS485  Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:Client Started]
[MSG:Connecting WIFI]
[MSG:Connecting.]
[MSG:VFD: VFD settings read: RPM Range(6000, 24000)]]
[MSG:Connecting..]
[MSG:Connecting...]
[MSG:Connecting....]
[MSG:Connected with 10.0.1.199]
[MSG:Connected with 10.0.1.199]
[MSG:Start mDNS with hostname:http://CNC_xPRO_V5.local/]
[MSG:SSDP Started]
[MSG:HTTP Started]
[MSG:TELNET Started 23]
[MSG:Mist coolant on pin GPIO(21)]
[MSG:X  Axis limit switch on pin GPIO(35)]
[MSG:Y  Axis limit switch on pin GPIO(34)]
[MSG:Z  Axis limit switch on pin GPIO(39)]
[MSG:Probe on pin GPIO(22)]
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]
> $S
$Sta/SSID=WIFI
$Sta/Password=******
$Sta/IPMode=Static
$Sta/IP=10.0.1.199
$Sta/Gateway=10.0.1.1
$Sta/Netmask=255.255.255.0
$AP/SSID=CNC_xPRO_V5<Alarm|MPos:0.000,0.000,0.000|FS:0,0|Pn:P>
$AP/Password=******
$AP/IP=192.168.0.1
$AP/Channel=1
$System/Hostname=CNC_xPRO_V5
$Http/Enable=ON
$Http/Port=80
$Telnet/Enable=ON
$Telnet/Port=23
$Radio/Mode=STA
$Bluetooth/Name=XPRO_V5_BT
$Notification/Type=NONE
$Notification/T1=
$Notification/T2=
$Notification/TS=
$Message/Level=Info
$User/Macro0=
$User/Macro1=
$User/Macro2=
$User/Macro3=
$Homing/Cycle0=Z
$Homing/Cycle1=XY
$Homing/Cycle2=
$Homing/Cycle3=
$Homing/Cycle4=
$Homing/Cycle5=
$Report/StallGuard=
$Stepper/Enable/Delay=0
$Stepper/Direction/Delay=0
$Stepper/Pulse=5
$Stepper/IdleTime=255
$Stepper/StepInvert=
$Stepper/DirInvert=
$Stepper/EnableInvert=Off
$Limits/Invert=On
$Probe/Invert=Off
$Report/Status=1
$GCode/JunctionDeviation=0.010
$GCode/ArcTolerance=0.002
$Report/Inches=Off
$Firmware/Build=
$Limits/Soft=On
$Limits/Hard=Off
$Homing/Enable=On
$Homing/DirInvert=X
$Homing/Squared=
$Homing/Feed=100.000
$Homing/Seek=1000.000
$Homing/Debounce=250.000
$Homing/Pulloff=10.000
$GCode/MaxS=24000.000
$GCode/MinS=0.000
$Laser/FullPower=1000
$GCode/LaserMode=Off
$GCode/Line1=
$GCode/Line0=
$Spindle/Enable/Invert=Off
$Spindle/Enable/OffWithSpeed=Off
$Coolant/Delay/TurnOn=1.000
$Spindle/Delay/SpinDown=0.000
$Spindle/Delay/SpinUp=0.000
$Spindle/PWM/Invert=Off
$Spindle/PWM/Frequency=5000.000
$Spindle/PWM/Off=0.000
$Spindle/PWM/Min=0.000
$Spindle/PWM/Max=100.000
$Spindle/Type=HUANYANG
$X/StepsPerMm=200.000
$Y/StepsPerMm=200.000
$Z/StepsPerMm=200.000
$A/StepsPerMm=200.000
$B/StepsPerMm=100.000
$C/StepsPerMm=100.000
$X/MaxRate=3000.000
$Y/MaxRate=3000.000
$Z/MaxRate=2500.000
$A/MaxRate=3000.000
$B/MaxRate=1000.000
$C/MaxRate=1000.000
$X/Acceleration=150.000
$Y/Acceleration=150.000
$Z/Acceleration=150.000
$A/Acceleration=150.000
$B/Acceleration=200.000
$C/Acceleration=200.000
$X/Home/Mpos=0.000
$Y/Home/Mpos=0.000
$Z/Home/Mpos=0.000
$A/Home/Mpos=0.000
$B/Home/Mpos=0.000
$C/Home/Mpos=0.000
$X/MaxTravel=1300.000
$Y/MaxTravel=1270.000
$Z/MaxTravel=106.000
$A/MaxTravel=1270.000
$B/MaxTravel=300.000
$C/MaxTravel=300.000
$X/Current/Run=1.950
$Y/Current/Run=1.950
$Z/Current/Run=1.950
$A/Current/Run=1.950
$B/Current/Run=0.250
$C/Current/Run=0.250
$X/Current/Hold=1.250
$Y/Current/Hold=1.250
$Z/Current/Hold=1.250
$A/Current/Hold=1.250
$B/Current/Hold=0.125
$C/Current/Hold=0.125
$X/Microsteps=8
$Y/Microsteps=8
$Z/Microsteps=8
$A/Microsteps=8
$B/Microsteps=16
$C/Microsteps=16
$X/StallGuard=16
$Y/StallGuard=16
$Z/StallGuard=16
$A/StallGuard=16
$B/StallGuard=16
$C/StallGuard=16
$Errors/Verbose=On
ok
> $+
$0=5 (Step pulse time, microseconds)
$1=255 (Step idle delay, milliseconds)
$2=0 (Step pulse invert, mask)
$3=0 (Step direction invert, mask)
$4=0 (Invert step enable pin, boolean)
$5=1 (Invert limit pins, boolean)
$6=0 (Invert probe pin, boolean)
$10=1 (Status report options, mask)
$11=0.010 (Junction deviation, millimeters)
$12=0.002 (Arc tolerance, millimeters)
$13=0 (Report in inches, boolean)
$20=1 (Soft limits enable, boolean)
$21=0 (Hard limits enable, boolean)
$22=1 (Homing cycle enable, boolean)
$23=1 (Homing direction invert, mask)
$24=100.000 (Homing locate feed rate, mm/min)
$25=1000.000 (Homing search seek rate, mm/min)
$26=250.000 (Homing switch debounce delay, milliseconds)
$27=10.000 (Homing switch pull-off distance, millimeters)
$30=24000.000 (Maximum spindle speed, RPM)
$31=0.000 (Minimum spindle speed, RPM)
$32=0 (Laser-mode enable, boolean)
$N1=
$N0=
$33=5000.000
$34=0.000
$35=0.000
$36=100.000
$100=200.000 (X-axis travel resolution, step/mm)
$101=200.000 (Y-axis travel resolution, step/mm)
$102=200.000 (Z-axis travel resolution, step/mm)
$103=200.000
$104=100.000
$105=100.000
$110=3000.000 (X-axis maximum rate, mm/min)
$111=3000.000 (Y-axis maximum rate, mm/min)
$112=2500.000 (Z-axis maximum rate, mm/min)
$113=3000.000
$114=1000.000
$115=1000.000
$120=150.000 (X-axis acceleration, mm/sec^2)
$121=150.000 (Y-axis acceleration, mm/sec^2)
$122=150.000 (Z-axis acceleration, mm/sec^2)
$123=150.000
$124=200.000
$125=200.000
$130=1300.000 (X-axis maximum travel, millimeters)
$131=1270.000 (Y-axis maximum travel, millimeters)
$132=106.000 (Z-axis maximum travel, millimeters)
$133=1270.000
$134=300.000
$135=300.000
$140=1.950
$141=1.950
$142=1.950
$143=1.950
$144=0.250
$145=0.250
$150=1.250
$151=1.250
$152=1.250
$153=1.250
$154=0.125
$155=0.125
$160=8
$161=8
$162=8
$163=8
$164=16
$165=16
$170=16
$171=16
$172=16
$173=16
$174=16
$175=16
ok
> $I
[VER:1.3a.20210816:]
[OPT:MPHBSW]
[MSG:Using machine:CNC_xPRO_V5_XYYZ_NC]
[MSG:Mode=STA:SSID=WIFI:Status=Connected:IP=10.0.1.199:MAC=98-CD-AC-6D-66-28]
[MSG:No BT]
ok
> M3 S6000
error: GCode cannot be executed in lock or alarm state
> $X
[MSG:Caution: Unlocked]
ok
> M3 S6000
[MSG:Spindle RS485 Unresponsive 6]
[MSG:Critical Spindle RS485 Unresponsive]
ALARM:10
[MSG:Critical Spindle RS485 did not reach speed 6000. Reported speed is -1 rpm.]
ok
ALARM:10
Grbl 1.3a ['$' for help]
[MSG:'$H'|'$X' to unlock]

Spark-Concepts, 2021-10-05:
Hi sflom, 

Could you add some pictures of your controller inside your cabinet so we can see the wire routing etc?  

What command are you using to turn on the relay to activate the coolant? 

Form these messages:

[MSG:Initializing RS485 VFD spindle]
[MSG:Auto-RTS in use]
[MSG:VFD RS485 Tx:GPIO(4) Rx:GPIO(25) RTS:None]
[MSG:Client Started]
[MSG:Connecting WIFI]
[MSG:Connecting.]
[MSG:VFD: VFD settings read: RPM Range(6000, 24000)]]

You have bidirectional communication before powering the Spindle, so we are likely chasing a noise issue or a spin-up time issue.

sflom, 2021-10-10:
I went ahead and verified all my connections, set proper star grounding, routed the power lines away from motors and limit lines and installed everything back into my enclosure.  Now I can control the spindle as it should. 

All is working as it should. 

This case can be closed. 

Thanks. 

Scott

PS - I have an IOT power relay that the VFD switches on when the spindle starts. It starts my water pump for the spindle and shuts it off when the spindle turns off. Works pretty slick. 

FB and 24w are jumped together and a line from FC and DCM runs down to the relay.

closed 2021-10-12 by Spark-Concepts