Spark-Concepts/xPro-V5 #66: Unresponsive Z Motor Output

APG-77, 2021-10-05:
Hello! I'm running into some trouble with my CNC xPRO V5. I am unable to get the Z-Axis working. I have narrowed the problem down to the xPRO not outputting a signal to the Z motor. I know this because I switched the wiring between X and Z motors, and in both cases, the motors work when connected to the X axis pins, but not when connected to the Z axis pins. In fact, it appears the Z motor output is not supplying any power at all. I am using the default firmware and have reviewed the GRBL settings, but none seem relevant. Is there a setting or firmware version I should be using? Sorry if this is something obvious. I checked the Wiki but don't see anything either. Cheers and thanks!

Spark-Concepts, 2021-10-05:
This is definitely not normal operation.  Could you send command $i and $S for the controller and paste the responses here.

Spark-Concepts, 2021-10-06:
Thanks!  The A/Y2 limit switch port is disabled by default so no values should be showing up.  

It looks like both of you have some older firmware, could you update to the latest push here:  https://github.com/Spark-Concepts/xPro-V5/tree/main/Firmware

Following these directions: https://github.com/Spark-Concepts/xPro-V5/wiki/Checking_firmware_and_upgrading#ota-over-the-air-firmware-updates

Take note of any special settings that you have changed (direction inversions, homing inversions, steps/mm etc) prior to loading new firmware.  Once loaded send command $RST=* to default to the new firmware settings and then apply the custom settings you recorded. 

If the issue persist, lets run a motor self-test:  Connect to the controller using USB, then open CNCjs and connect to the controller (make sure to check the "Enable hardware flow control" box in the com port area).  You will see a long output of diagnostics and settings -  look for these lines for each motor and paste them back here:

[MSG:X  Axis Trinamic TMC5160 Step:GPIO(12) Dir:GPIO(14) CS:GPIO(17) Disable:None Index:1 R:0.050 Limits(0.000,300.000)]

[MSG:X  Axis Trinamic driver test passed]

APG-77, 2021-10-08:
The update worked for me as well, problem solved. Have to say I'm impressed with the overall quality and UX of this project. You guys have done great. Thanks again!

closed 2021-10-08 by APG-77