Spark-Concepts/xPro-V5 #80: FluidNC

ahedderich, 2021-11-02:
Hi, i would like to try running FluidNC as firmware on the xPro V5.
to create a config file i would need to have a schematic of the gpio pins used.
alternatively could you provide a sample config yaml for the xpro like this:
https://github.com/bdring/FluidNC/blob/main/example_configs/6_pack_TMC5160.yaml

ahedderich, 2021-11-02:
i just found the pinout picture...
https://github.com/Spark-Concepts/xPro-V5/blob/main/images/xProV5-GPIO_PinMap.jpg

closed 2021-11-02 by ahedderich

Spark-Concepts, 2021-11-02:
quick find!  If you would like help generating the config file feel free to post and let us know.  

We are tracking the FluidNC development, if you have thoughts and experiences you would like to share please do!

ahedderich, 2021-11-03:
i will try it with this config on the weekend.
not super sure though about the 10V section.
since i have to output 10V to control spindle speed, thats my best guess going by the schematics on how the toolhead pinout work.
maybe for xPro V6 you could use the i2s protocol to add more stepper drivers. like bartons latest controller board.
it seems to have some benefits for connecting more things and synchronising with the stepping engine
i will give feedback if this config file works for me.
```
board: xPro V5
name: xPro V5 XYZ 3 motors
stepping:
  engine: RMT
  idle_ms: 250
  dir_delay_us: 1
  pulse_us: 2
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 1.75
        hold_amps: 0.50
        microsteps: 256
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  y:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 1.75
        hold_amps: 0.50
        microsteps: 256
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 5000.000
    acceleration_mm_per_sec2: 100.000
    max_travel_mm: 300.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      debounce_ms: 500
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: true
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 1.750
        hold_amps: 0.50
        microsteps: 16256
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: StealthChop
        homing_mode: StealthChop
        use_enable: true
        step_pin: gpio.15
        direction_pin: gpio.2
        disable_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.21
  delay_ms: 0

i2so:
  bck_pin: NO_PIN
  data_pin: NO_PIN
  ws_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.16
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.13
  macro1_pin: gpio.0
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.22
  check_mode_start: false

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%```

ahedderich, 2021-11-08:
so after a bit of testing i can say it works...
one issue so far is tmc driver initiation does not work properly when reseting while connected via usb. i wrote with bart on discord and he will investigate that behavior. if i find the time i will also have a look at the source why this happens but other than that everything works.
my config so far is as follows:
```
board: xPro V5
name: xPro V5 XYZA
stepping:
  engine: RMT
  idle_ms: 255
  pulse_us: 2
  dir_delay_us: 1
  disable_delay_us: 0

axes:
  shared_stepper_disable_pin: NO_PIN
  x:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 900.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.35:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: gpio.17
        spi_index: 1
        r_sense_ohms: 0.075
        run_amps: 2.8
        hold_amps: 0.60
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.12
        direction_pin: gpio.14
        disable_pin: NO_PIN

  y:
    steps_per_mm: 660.000
    max_rate_mm_per_min: 2500.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 900.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: false
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.34:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        spi_index: 2
        r_sense_ohms: 0.075
        run_amps: 3.0
        hold_amps: 0.60
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.27
        direction_pin: gpio.26
        disable_pin: NO_PIN

  a:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 9000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 900.000
    soft_limits: false
    homing:
      cycle: 2
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 200.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.36:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 3
        r_sense_ohms: 0.075
        run_amps: 2.8
        hold_amps: 0
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.33
        direction_pin: gpio.32
        disable_pin: NO_PIN

  z:
    steps_per_mm: 800.000
    max_rate_mm_per_min: 4000.000
    acceleration_mm_per_sec2: 250.000
    max_travel_mm: 400.000
    soft_limits: false
    homing:
      cycle: 1
      positive_direction: true
      mpos_mm: 150.000
      feed_mm_per_min: 100.000
      seek_mm_per_min: 800.000
      seek_scaler: 1.100
      feed_scaler: 1.100

    motor0:
      limit_neg_pin: gpio.39:low
      limit_pos_pin: NO_PIN
      limit_all_pin: NO_PIN
      hard_limits: false
      pulloff_mm: 1.000
      tmc_5160:
        cs_pin: NO_PIN
        spi_index: 4
        r_sense_ohms: 0.075
        run_amps: 2.8
        hold_amps: 0.60
        microsteps: 16
        stallguard: 0
        stallguard_debug: false
        toff_disable: 0
        toff_stealthchop: 5
        toff_coolstep: 3
        run_mode: CoolStep
        homing_mode: CoolStep
        use_enable: true
        step_pin: gpio.15
        direction_pin: gpio.2
        disable_pin: NO_PIN

coolant:
  flood_pin: NO_PIN
  mist_pin: gpio.21
  delay_ms: 0

i2so:
  bck_pin: NO_PIN
  data_pin: NO_PIN
  ws_pin: NO_PIN

spi:
  miso_pin: gpio.19
  mosi_pin: gpio.23
  sck_pin: gpio.18

sdcard:
  card_detect_pin: NO_PIN
  cs_pin: gpio.5

control:
  safety_door_pin: gpio.16:low
  reset_pin: NO_PIN
  feed_hold_pin: NO_PIN
  cycle_start_pin: NO_PIN
  macro0_pin: gpio.13:low
  macro1_pin: gpio.0:low
  macro2_pin: NO_PIN
  macro3_pin: NO_PIN

probe:
  pin: gpio.22:low
  check_mode_start: false

macros:
  startup_line0:
  startup_line1:
  macro0:
  macro1:
  macro2:
  macro3:

user_outputs:
  analog0_pin: NO_PIN
  analog1_pin: NO_PIN
  analog2_pin: NO_PIN
  analog3_pin: NO_PIN
  analog0_hz: 5000
  analog1_hz: 5000
  analog2_hz: 5000
  analog3_hz: 5000
  digital0_pin: NO_PIN
  digital1_pin: NO_PIN
  digital2_pin: NO_PIN
  digital3_pin: NO_PIN

start:
  must_home: false

10V:
  forward_pin: NO_PIN
  reverse_pin: NO_PIN
  pwm_hz: 5000
  output_pin: gpio.25
  enable_pin: gpio.4
  direction_pin: NO_PIN
  disable_with_s0: false
  s0_with_disable: true
  spinup_ms: 0
  spindown_ms: 0
  tool_num: 0
  speed_map: 0=0.000% 1000=0.000% 24000=100.000%

```