Spark-Concepts/xPro-V5 #81: 48V for tmc5160

ahedderich, 2021-11-08:
Hi,
is in any way possible with the current hardware and maybe some "hacking" to supply the trinamic drivers with a 48V input power, to drive higher impedance motors with higher speeds?
when opening up the top panel i don't see an easy way to get to the bottom part of the board.
is the bottom panel glued in place?

Spark-Concepts, 2021-11-13:
Hi Ahedderich, currently there is no easy way to split the motor voltage and the operating voltage.  It could be done but would require extensive modification to the controller.  The top board is held in place with 3 locking standoffs, you can see the white locking tips protruding through the top board.   Pinch the tops of these and pull the top board up out of the case (takes some wiggling to the the power switch out - also make sure ALL of the pluggable headers and the SD card are removed).  The bottom board is then held in with screws.

ahedderich, 2021-11-15:
i think that is a non starter. that seems to be too much hacking to make it work.

closed 2021-11-15 by ahedderich