Spark-Concepts/xPro-V5 #82: Y Axis Trouble

Accufactor, 2021-11-10:
I seem to have a serious problem with the y axis, I have 1 wired into the Y port and 1 wired into the A/Y2. When I started a small job to test the machine for the first time, it ran for about 30 minutes just fine, then whenever the Y's had to move it made a horrible noise, I stopped the job with the estop, and tried to investigate the problem. Turns out the y axis is no longer moving in unison, one side seems to move before the other and it causes it to skew, I thought it might have had something to do with the machine not being square as it lifted at 1 point as I was trying to move the axis. So 1 rebuild later and the problem still persists, we even swapped out the stepper motor on that 1 side that is moving delayed to the other 1 and it still no change, I thought it might have been a setting so I updated with fresh firmware and used the $RST=* command still no luck.

Mawg1, 2021-11-11:
Hi Accufactor, Sorry to hear there is trouble.  Could you confirm that both motors are spinning but one is lagging the other?  Is the delay following the motor or the xPRO port when you swapped sides?

Accufactor, 2021-11-11:
it turns out that it is the Y port, not the A/Y2 port on the xpro v5, I did a simple test of disconnecting the leadscrew from the coupler to the stepper motor and attached some tape to the coupler to watch which direction, turns out the A/Y2 works as intended, however, the Y port attempts to work, then sort of freezes and tries to start again for the duration of the steps, I tested it multiple times with a single stepper motor to remove any uncertainty. For example if i decided i wanted to step the A/Y2 by 10 it would rotate smoothly, however the Y port would make a horrible noise like it is fighting itself and not moving.

Spark-Concepts, 2021-11-13:
Thanks for testing!  Let's get this fixed up, could you submit a contact form with order and shipping info  - [https://www.spark-concepts.com/contact-us-1/](https://www.spark-concepts.com/contact-us-1/)

Accufactor, 2021-11-13:
I have submitted the form and await your reply

closed 2021-11-27 by Accufactor